espressif/esp-idf #5020: PSA Request:  spi_flash: failed to get chip info... Please Post This In Easy To GOOG Location

jwestmoreland, 2020-03-29:
Hello All,

I spent more time than I can to admit figuring this out - I even looked at projects that were fine vs. those that weren't - and just missing one click box made all of the difference.

I was getting this:

[image]

until I realized in some of my projects - this WAS NOT CLICKED:

[image]

Please post in big read letters on your home pages - it'll save some a lot of time I think.

Thanks In Advance,
John W.

igrr, 2020-03-29:
Which hardware are you using? Is this a custom board or module?

The flash chip detected is Macronix MX25R1635F. Indeed this chip does not seem to be supported in IDF v4.0, which might be a regression from v3.3... We will look into this, in the meantime you may use the CONFIG_SPI_FLASH_USE_LEGACY_IMPL=y as a workaround, as you have already found.

jwestmoreland, 2020-03-29:
Hello @igrr ,

I'm using a custom board but I put the uBlox NINA-W102 module on it for now - so I'm using that.
I have all available I/O's pinned out on this board - so I'm able to use JTAG as well as some of the other features.

Thanks for looking into this.

Regards,
John W.

ginkgm, 2020-04-08:
Hi @jwestmoreland ,

61ce037fc1f3c5747ee35328e3cd9cdfd2b8b86f The latest master may have solved this, could you help have a try? (Though the fix was made for MX25L12872)

jwestmoreland, 2020-04-08:
Hello Michael,

OK - I'll give it a try later today - I'll wait for the sync.

Thanks,
John

jwestmoreland, 2020-04-09:
Hello Again Michael,

The latest I can see right now from 4/8 is origin/release/v3.1 - I'm up to v4.0 - otherwise I see origin/head from 4/2 - so it would appear the update I need isn't in the rev. I need (yet).

I should have mentioned I'm running the AT stuff in 2MB (currently on the NINA-W102) - I tried to compile tip of tree in both the AT and IDF and errored out here:

C:/esp-tot/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal.c:42:23: error: field 'sha256' has incomplete type
   struct sha256_state sha256;
                       ^~~~~~
C:/esp-tot/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal.c: In function 'crypto_hash_init':
C:/esp-tot/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal.c:78:3: error: implicit declaration of function 'sha256_init'; did you mean 'sha256_vector'? [-Werror=implicit-function-declaration]
   sha256_init(&ctx->u.sha256);
   ^~~~~~~~~~~
   sha256_vector
C:/esp-tot/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal.c:131:4: error: implicit declaration of function 'sha256_process'; did you mean 'sha256_vector'? [-Werror=implicit-function-declaration]
    sha256_process(&ctx->u.sha256, key, key_len);
    ^~~~~~~~~~~~~~
    sha256_vector
C:/esp-tot/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal.c:132:4: error: implicit declaration of function 'sha256_done'; did you mean 'sha256_vector'? [-Werror=implicit-function-declaration]
    sha256_done(&ctx->u.sha256, tk);
    ^~~~~~~~~~~
    sha256_vector

cc1.exe: some warnings being treated as errors

Since enabling the legacy SPI init works for me - I probably won't bother trying to get this to build until I hear the AT part of this is synched up to whatever build v4.x will exist at that time.

Thanks,
John

ESP-Marius, 2020-04-09:
@jwestmoreland Hi!

The last errors you posted here seems to be unrelated to the issue with the flash chip.  Did you disable " Use MbedTLS crypto API's" ( `CONFIG_WPA_MBEDTLS_CRYPTO`) for the wpa supplicant? This seems to cause issues on the v4.x now.

We will investigate it, thanks for reporting this!

jwestmoreland, 2020-04-09:
Hello @ESP-Marius ,

Getting this now:
C:/esp-tot/esp-idf/components/driver/spi_bus_lock.c:820:41: error: implicit declaration of function 'xSemaphoreCreateBinaryStatic'; did you mean 'xSemaphoreCreateBinary'? [-Werror=implicit-function-declaration]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                         xSemaphoreCreateBinary
C:/esp-tot/esp-idf/components/driver/spi_bus_lock.c:820:39: warning: assignment to 'SemaphoreHandle_t' {aka 'void *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                       ^
cc1.exe: some warnings being treated as errors

I had the MbedTLS crypto stuff disabled.

Thanks,
John

ESP-Marius, 2020-04-09:
@jwestmoreland `FREERTOS_SUPPORT_STATIC_ALLOCATION` needs to be selected now to compile the latest version of v4.2. It is set to `y` by default, but you might have a problem compiling it if you are using an old sdkconfig file. 

Setting it to `y` in menuconfig (under FreeRTOS) should fix this. Sorry for your inconvenience.

jwestmoreland, 2020-04-09:
@ESP-Marius ,

OK - good to know - getting this now - 
cmd.exe /C "cd . && C:\esptools\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\blink.elf.rsp  -o blink.elf  && cd ."
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x0): undefined reference to `_Unwind_GetTextRelBase'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x4): undefined reference to `_Unwind_GetDataRelBase'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x8): undefined reference to `_Unwind_GetRegionStart'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x8): undefined reference to `_Unwind_GetLanguageSpecificData'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0xc): undefined reference to `_Unwind_GetIPInfo'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x10): undefined reference to `_Unwind_SetGR'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x14): undefined reference to `_Unwind_SetIP'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__cxa_call_unexpected+0x0): undefined reference to `_Unwind_Resume'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `base_of_encoded_value(unsigned char, _Unwind_Context*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:114: undefined reference to `_Unwind_GetTextRelBase'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:116: undefined reference to `_Unwind_GetDataRelBase'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:118: undefined reference to `_Unwind_GetRegionStart'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `parse_lsda_header(_Unwind_Context*, unsigned char const*, lsda_header_info*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:55: undefined reference to `_Unwind_GetRegionStart'
[... cut ...]
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:83: undefined reference to `_Unwind_DeleteException'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Build complete (27 errors, 9 warnings):
 
Note those build tools are actually esp-2020r1-8.2.0 - I had to do it that way since I got tired of fighting the build system in eclipse...  I thought all of the errors (__Unwind...) were due to not using the new build tools.

Thanks,
John

jwestmoreland, 2020-04-09:
@ESP-Marius ,

Got the code built but it's resetting - need to look at that - could take a little while.

Thanks,
John

ginkgm, 2020-04-10:
@jwestmoreland Could you help provide the log when it's resetting?

jwestmoreland, 2020-04-10:
Hello @ginkgm ,

Log file attached - it's txt format.

Thanks,
John
[YAT-Log-20200409-224645.log](https://github.com/espressif/esp-idf/files/4460076/YAT-Log-20200409-224645.log)

jwestmoreland, 2020-04-15:
Hello @ginkgm and Espressif,

So, no comments on the log posted 5 days ago?

Regards,
John

ginkgm, 2020-04-16:
Hi @jwestmoreland ,

Sorry, but I forgot to tell you that what we need is the traceback (i.e. the call stack when your program failes.). From the log above I cannot locate where the issue happens..

Could you please post the log with traceback (using idf_monitor) or even your elf with symbol table (optional, this will expose the details of your application), to help us at least locate where the issue is?

jwestmoreland, 2020-04-16:
If traceback and backtrace are the same, I know that is in the posted log
file.

xcguang, 2020-04-16:
Hello,
At present, the IDF used by the latest AT project is v4.0, which may not be updated in the next two months.

jwestmoreland, 2020-04-16:
From the log file:

V (6620) esp_netif_objects: esp_netif_next_unsafe 0x3ffcbb78<ESC>[0m<CR><LF>
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.<CR><LF>
<CR><LF>
Core  0 register dump:<CR><LF>
PC      : 0x400ef007  PS      : 0x00060430  A0      : 0x800ed4ef  A1      : 0x3ffbd610  <CR><LF>
A2      : 0x00000000  A3      : 0x3f408878  A4      : 0x00000000  A5      : 0x3ffbd670  <CR><LF>
A6      : 0x3f41bffc  A7      : 0x3ffbd620  A8      : 0x800ede4e  A9      : 0x3ffbd620  <CR><LF>
A10     : 0x00000000  A11     : 0x3ffcb614  A12     : 0x3ffb645c  A13     : 0x00000000  <CR><LF>
A14     : 0x3f408d9c  A15     : 0x3ffcd4ac  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  <CR><LF>
EXCVADDR: 0x00000024  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  <CR><LF>
<CR><LF>
Backtrace:0x400ef004:0x3ffbd610 0x400ed4ec:0x3ffbd660 0x400e1749:0x3ffbd680 0x400e3ab2:0x3ffbd6a0 0x400e03ea:0x3ffbd710 0x400d6c4a:0x3ffbd740 0x400d4776:0x3ffbd770 0x400944cd:0x3ffbd7b0<CR><LF>
<CR><LF>
<CR><LF>

jwestmoreland, 2020-04-16:
Hello @xcguang ,

Can you please post the build commit ID of the IDF that the AT lib is compatible with?

I guess that's in the lib folder but would be nice to know if you guys are trying different commits of the IDF.

Thanks In Advance,
John

jwestmoreland, 2020-04-16:
Hello @ginkgm and Espressif,

Ran and built the idf monitor:

ELF file SHA256: b6a5d08c233d9aa4

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7052
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x40080674
I (31) boot: ESP-IDF v4.2-dev-1097-g2e14149bf-dirty 2nd stage bootloader
I (31) boot: compile time 00:37:38
I (32) boot: chip revision: 1
I (36) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 2MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (81) boot:  2 nvs              WiFi data        01 02 00012000 0000dc00
I (88) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (96) boot:  4 blink            factory app      00 00 00100000 00100000
I (103) boot: End of partition table
I (107) boot: Defaulting to factory image
I (112) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x21a00 (137728) map
I (174) esp_image: segment 1: paddr=0x00121a28 vaddr=0x3ffb0000 size=0x03b9c ( 15260) load
I (180) esp_image: segment 2: paddr=0x001255cc vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/esp-tot/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (181) esp_image: segment 3: paddr=0x001259d8 vaddr=0x40080404 size=0x0a640 ( 42560) load
I (208) esp_image: segment 4: paddr=0x00130020 vaddr=0x400d0020 size=0xa47c4 (673732) map
0x400d0020: _stext at ??:?

I (464) esp_image: segment 5: paddr=0x001d47ec vaddr=0x4008aa44 size=0x0ccb0 ( 52400) load
0x4008aa44: lmacTxFrame at ??:?

I (488) esp_image: segment 6: paddr=0x001e14a4 vaddr=0x400c0000 size=0x00064 (   100) load
I (500) boot: Loaded app from partition at offset 0x100000
I (501) boot: Disabling RNG early entropy source...
Hello From ESP32! uxTopUsedPriority =24
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 2MB external flash
Starting app in 3 seconds...
Starting app in 2 seconds...
Starting app in 1 seconds...
Starting app in 0 seconds...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400ed9bf  PS      : 0x00060230  A0      : 0x800ec4a7  A1      : 0x3ffbd8e0
0x400ed9bf: esp_netif_dhcpc_stop at C:/esp-tot/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:890

A2      : 0x00000000  A3      : 0x3f406abc  A4      : 0x00000000  A5      : 0x3ffbd930
A6      : 0x3ffb6a9c  A7      : 0x3ffbd8e0  A8      : 0x800ecc92  A9      : 0x3ffbd8e0
A10     : 0x00000000  A11     : 0x3ffcb3d8  A12     : 0x3ffb675c  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x3ffbd910  SAR     : 0x00000013  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000024  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x400ed9bc:0x3ffbd8e0 0x400ec4a4:0x3ffbd920 0x400e1171:0x3ffbd940 0x400e34da:0x3ffbd960 0x400dfe12:0x3ffbd9d0 0x400d66d3:0x3ffbda00 0x400d433e:0x3ffbda30 0x400943ed:0x3ffbda70
0x400ed9bc: esp_netif_dhcpc_stop at C:/esp-tot/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:890

0x400ec4a4: tcpip_adapter_dhcpc_stop at C:/esp-tot/esp-idf/components/tcpip_adapter/tcpip_adapter_compat.c:241

0x400e1171: at_set_dhcp_status at ??:?

0x400e34da: at_nvm_init at ??:?

0x400dfe12: esp_at_module_init at ??:?

0x400d66d3: app_mainb at c:\users\john\foo\f4\build/../main/app_mainb.c:266

0x400d433e: main_task at C:/esp-tot/esp-idf/components/esp32/cpu_start.c:581

0x400943ed: vPortTaskWrapper at C:/esp-tot/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: b6a5d08c233d9aa4

Regards,
John
[idf_monitor-log-file.txt](https://github.com/espressif/esp-idf/files/4485606/idf_monitor-log-file.txt)

xcguang, 2020-04-16:
You cannot change IDF version, because AT is a lib based on IDF v4.0.

jwestmoreland, 2020-04-16:
OK - can you give the complete SHA-1 signature of the IDF commit that correlates to please?
Thanks.

xcguang, 2020-04-16:
It is defined in https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/IDF_VERSION, and the detailed description is in https://github.com/espressif/esp-at/blob/master/docs/How_To_Update_IDF.md

jwestmoreland, 2020-04-16:
Hello @xcguang ,

Ha!  I was looking here for correlation:

ESP32 lib       : 518c1ce7
ESP32S2 lib     : 518c1ce7
ESP8266 lib     : 518c1ce7

I thought that was part of the commit sig when I saw it; since it's built against the IDF....

OK - Thanks,
John

xcguang, 2020-04-16:
The commit SHA is f9cb434ee in the response of AT+GMR, which is same as https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/IDF_VERSION

jwestmoreland, 2020-04-16:
Yes, I see that now - but I have this build running too (previous post...)

The current build I have running is:
AT+GMR

   AT version:2.1.0.0-dev(9a63a02 - Apr  2 2020 13:05:33)                
SDK version:v4.0-280-ga3f3c7bdc-dirty
compile time(0x1234-ESP-AT-MANUALLY-SET-JCW):Apr 9 2020 05:24:43

xcguang, 2020-04-27:
By default, it should be `compile time( AT project commit SHA):Apr 16 2020 02:54:24`, it seems you delete project SHA info in `app_main()`. The AT version you got is different because you compile it based on different AT project SHA.

AQIBRIAZ, 2020-05-22:
Hi,

I have the same issue with idf v4.0.0. I am using ninaW102 module on platform io, below is my logs
[image]

can you help me solve the issue?

tcpipchip, 2020-05-31:
solved to NINA W102
thanks!

xcguang, 2021-05-10:
Hi,
The latest [ESP32 AT](https://github.com/espressif/esp-at/) project is based on IDF v4.2, if possible, please try it again.

Alvin1Zhang, 2021-08-18:
Thanks for sharing updates, will close due to short of feedback, feel free to reopen with more updates.

closed 2021-08-18 by Alvin1Zhang