espressif/esp-idf #5022: bcn_timout,ap_probe_send_start on HTTP POST request

galah92, 2020-03-29:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version v4.0
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: VSCode (without the extension)
- Power Supply: USB

## Problem Description

```cpp
I (47352) wifi: bcn_timout,ap_probe_send_start
I (49852) wifi: ap_probe_send over, resett wifi status to disassoc
I (49852) wifi: state: run -> init (c800)
I (49852) wifi: pm stop, total sleep time: 25660789 us / 48198940 us

I (49852) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (49862) WiFi: WIFI_EVENT_STA_DISCONNECTED - 200
E (49872) esp-tls: read error :-76:
I (49872) WiFi: retry to connect to the AP
```

This is the log I'm getting right after doing `esp_http_client_perform(client)` for a POST request. I've got over 30KB available in the heap right before the call.

Alvin1Zhang, 2020-03-30:
@galah92 Thanks for reporting, we will look into.

mahavirj, 2020-04-01:
@galah92 Is it possible to free up more dynamic memory? For optimized TLS connection, it will need roughly 30-32KB memory and per default WIFI buffer configuration (dynamic TX buffer), it will need additional ~10-15KB free memory.

galah92, 2020-04-01:
@mahavirj Thanks. I'll try free more heap space.
Is it possible for the TLS code to allocate buffers statically instead of dynamically?
Can all WiFi buffers be allocated statically?
I'm using both `esp_http_client` & `esp_mqtt_client` instances in my code. Is it possible for the TLS code to share memory between these two?

mahavirj, 2020-04-01:
@galah92 For WiFi buffers, yes you may configure using options provided at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp32-wifi-static-rx-buffer-num. For TLS buffers, static allocation is not possible.

galah92, 2020-04-01:
> @galah92 For WiFi buffers, yes you may configure using options provided at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp32-wifi-static-rx-buffer-num. For TLS buffers, static allocation is not possible.

Thanks. What about sharing TLS memory between HTTP & MQTT?

mahavirj, 2020-04-01:
@galah92 No. However, if you are okay to operate in single core mode, then you can turn on config option https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L36, which will redirect TLS buffers to IRAM. So that shall allow you to use DRAM for application purpose. Please note this feature is available with `master` branch (soon to reflect in `v4.2`) only.

mahavirj, 2020-04-06:
@galah92 Did you get chance to try out suggestions? Can we close the issue?

galah92, 2020-04-06:
I was able to free some more memory on heap and it's working now. Using static WiFi buffers didn't help much. I'm working on the stable branch (`v4.0`) so can't still check the latest suggestion regarding single core mode. I I'm looking forward to try it as soon as possible. We can close the issue, thank you very much.

closed 2020-04-06 by mahavirj