espressif/esp-idf #5023: WiFi reconnect fails

sarasini, 2020-03-29:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3.1
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:  Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Once connected to WiFi, after some time (minutes or hours), connection is lost and can't be re-established. 

Every subsequent connection attempts obtains errors like: 
WIFI_REASON_ASSOC_FAIL
WIFI_REASON_MIC_FAILURE
WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT

Powering off and on the module do not fix the problem. Once program restart, the module cant connect to Wifi.

Sometimes, after a lot of retries the module manage to re-connect to the wifi network.

Other peripherals (laptop, smartphone,...) are connected to the same WiFi network and works normally.

## Debug Logs

```
I (2134421) wifi: Start wifi connect
I (2134421) wifi: connect status 4 -> 0
I (2134431) wifi: connect chan=0
I (2134431) wifi: first chan=1
I (2134431) wifi: connect status 0 -> 1
I (2134441) wifi: filter: set rx policy=3
I (2134441) wifi: clear scan ap list
I (2134441) wifi: start scan: type=0x50f, priority=2, cb=0x400ec6a0, arg=0x0, ss_state=0x1, time=2134012014, index=0
0x400ec6a0: cnx_start_handoff_cb at ??:?
I (2134451) wifi: perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
I (2134461) wifi: rsn valid: gcipher=1 ucipher=3 akm=5
I (2134471) wifi: profile match: ss_state=0x7
I (2134481) wifi: rsn valid: gcipher=1 ucipher=3 akm=5
I (2134481) wifi: profile match: ss_state=0x7
I (2134581) wifi: rsn valid: gcipher=1 ucipher=3 akm=5
I (2134581) wifi: profile match: ss_state=0x7
I (2134581) wifi: scan end: arg=0x0, status=0, ss_state=0x7
I (2134581) wifi: find first mathched ssid, scan done
I (2134591) wifi: filter: set rx policy=4
I (2134591) wifi: first chan=1
I (2134591) wifi: handoff_cb: status=0
I (2134601) wifi: ap found, mac=e0:b9:e5:5d:9f:25
I (2134601) wifi: new_bss=0x3ffc6340, cur_bss=0x0, new_chan=<1,0>, cur_chan=1
I (2134611) wifi: filter: set rx policy=5
I (2134611) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2134621) wifi: connect_op: status=0, auth=5, cipher=3 
I (2134621) wifi: auth mode is not none
I (2134631) wifi: connect_bss: auth=1, reconnect=0
I (2134631) wifi: state: init -> auth (b0)
I (2134641) wifi: start 1s AUTH timer
I (2134641) wifi: clear scan ap list
I (2134651) wifi: recv auth: seq=2, status=0
I (2134651) wifi: state: auth -> assoc (0)
I (2134651) wifi: restart connect 1s timer for assoc
I (2134661) wifi: recv assoc: type=0x10
E (2134661) wifi: Set status to INIT
I (2134661) wifi: state: assoc -> init (c00)
I (2134671) wifi: connect status 1 -> 4
I (2134671) wifi: stop beacon/connect timer
I (2134671) wifi: add bssid e0:b9:e5:5d:9f:25 to blacklist, cnt=0
I (2134681) wifi: stop CSA timer
[... 146 lines cut ...]
I (2173321) wifi: Send sta connected event
I (2173321) wifi: connect status 1 -> 5
I (2173331) wifi: obss scan is disabled
I (2173331) wifi: start obss scan: obss scan is stopped
I (2173341) WifiManager: ESP32 station connected to AP
D (2173341) WifiManager: connected
I (2173391) wifi: AP's beacon interval = 102400 us, DTIM period = 3
I (2173391) wifi: rsn valid: gcipher=1 ucipher=3 akm=5
I (2173491) wifi: rsn valid: gcipher=1 ucipher=3 akm=5
I (2173591) wifi: rsn valid: gcipher=1 ucipher=3 akm=5
I (2173621) wifi: recv deauth, reason=0xe
I (2173621) wifi: state: run -> init (ec0)
I (2173621) wifi: recv deauth/disassoc, stop beacon/connect timer
I (2173621) wifi: connect status 5 -> 4
I (2173621) wifi: sta leave
I (2173621) wifi: pm stop, total sleep time: 168496 us / 304620 us
I (2173631) wifi: stop CSA timer
I (2173631) wifi: remove e0:b9:e5:5d:9f:25 from rc list
I (2173641) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2173641) wifi: filter: set rx policy=8
I (2173651) wifi: Send disconnect event, reason=15, AP number=0
I (2173661) WifiManager: ESP32 station disconnected from AP
D (2173661) WifiManager: disconnected
I (2173661) WifiManager: WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT
```

Alvin1Zhang, 2020-03-30:
@sarasini Thanks for reporting, we will look into.

kifantidis, 2020-04-08:
Hello,
is your device near router ? This behavior might have to do with low wifi signal in the spot that you have put your esp device.

liuzfesp, 2020-07-10:
HI @sarasini, have you solved your problem?  The `WIFI_REASON_MIC_FAILURE` looks very strange, could you help to provide following info:

- Router brand
- The frequency of the issue

HarveyRong-Esp, 2020-11-09:
@sarasini , Thanks for reporting, will close due to short of feedback, feel free to reopen.

closed 2020-11-09 by Alvin1Zhang