espressif/esp-idf #5024: HFP disconnects incoming call when disconnected

shimomu2, 2020-03-30:
## Environment
- Development Kit: EVK-NINA-W102-00
- Module or chip used:  NINA-W102
- IDF version:  4.0
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc 8.2.0
- Operating System: Linux
- Using an IDE:No
- Power Supply:USB

## Problem Description

If you connect to the smartphone with HFP and disconnect the incoming call from esp32, the connection will be disconnected.

### Steps to reproduce

Follow BTstack instructions for ESP32: https://github.com/bluekitchen/btstack/blob/master/port/esp32/README.md

Switch to btstack/port/esp32/example/hfp_hf_demo

start demo.
After connecting to the device, disconnect the call to the device

## Debug Logs
[snooplog.zip](https://github.com/espressif/esp-idf/files/4401740/snooplog.zip)

AbnerFederer, 2020-04-08:
My test is good, so may you can provide more info like which phone.

AbnerFederer, 2020-04-09:
Hi, @shimomu2 

I have reviewed the air log you provided.
But I haven't seen any LMP log, can you support the complete log?
Or I would think the two devices are not really connected.

Thanks
Abner

shimomu2, 2020-04-15:
Here, we are checking with Galaxy S10,
Even with the same Galaxy S10, there were some cases where the phenomenon occurred and some did not occur.

When confirmed on different smartphones, the phenomenon did not occur.

Please check the attached airlog when the phenomenon occurs.
[airlog.zip](https://github.com/espressif/esp-idf/files/4479613/airlog.zip)

AbnerFederer, 2020-04-27:
Hi, @shimomu2 

Have you ever compare the Bluedroid host in IDF?
I cannot reproduce your issue with Bluedroid.

Thanks
Abner

AbnerFederer, 2020-04-27:
Hi, @shimomu2 

From the air log, I see that the initiating process of the ACL link is okay but the initiating process of the SCO link has some issues. **I don't see that SCO link has already been established.** And because of "AT+CHUP" is a string sent to the controller, so please check the host string handler. The detach may be caused by the timeout of the SCO parameter negotiation.

Thanks
Abner

AbnerFederer, 2020-04-27:
Hi, @shimomu2 

The air log shows that the SCO connection has not been established and ESP32 initiates **LMP Response Timeout**, in the end, to disconnect with the peer device.

The disconnection comes from the LMP timeout (30 sec) rather than the error of AT+CHUP.

I will check the controller and please try Bluedroid in IDF first.

Thanks
Abner

Wth-Esp, 2020-05-07:
Hi, @shimomu2

Any update for this issue?

Thanks
Weitianhua

Wth-Esp, 2020-05-08:
Hi, @shimomu2 

There is a lib file for you, please try it and better scratch another air log.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/4596783/libbtdm_app.zip)

Thanks
Weitianhua

Wth-Esp, 2020-05-12:
Hi @shimomu2 

Any updates about this issue?

Thanks
Weitianhua

Alvin1Zhang, 2020-05-13:
Will close due to short of feedback.

closed 2020-05-13 by Alvin1Zhang