espressif/esp-idf #5025: Tools install in eclipse just dies.

umfundi, 2020-03-30:
win 10 64, 32g memory

latest copy of esp-idf cloned into local drive.

latest eclipse environment, updated.

running 'Tools Install' 

Nothing, no popups, no error report no feedback.

Running 'list installed tools'  same thing.

umfundi, 2020-03-30:
esp p[lugin is installed

kolipakakondal, 2020-03-30:
Hi @umfundi Could you please check this thread https://github.com/espressif/idf-eclipse-plugin/issues/63#issuecomment-605502100 You might be having the same issue.

To check further on this, could you please provide the error log https://github.com/espressif/idf-eclipse-plugin#error-log

Also, provide the Eclipse configuration and plugin details that you're using. Please attach the file.
`Help > About Eclipse >Installation Details > Configuration
`

kolipakakondal, 2020-03-30:
You can also raise the issues directly here https://github.com/espressif/idf-eclipse-plugin/issues to get a quicker response.

umfundi, 2020-03-30:
hmmm. Ok that Git path caught me. but it has moved on to a slightly different error now. 

Complaining about something not being a valid win32 executable. Trying to run idf_tools.py itself instead of via the Python interpreter.

Then tried again and switched to Python 3.6 ..... strangely that appears to have worked but still no tools available .... no tools listed when list installed tools is selected - Python path appears to be null.

This was a 'by the book' install, frustrating that it's so broken

kolipakakondal, 2020-03-31:
Any python versions listed in the install tools dialog? It reads the python versions from the windows registry. Please check.
```
HKEY_CURRENT_USER\SOFTWARE\Python\PythonCore\
HKEY_LOCAL_MACHINE\SOFTWARE\Python\PythonCore\
```

umfundi, 2020-03-31:
3.8 and 3.6 listed in the dialog
3.8 is in HKEY_CURRENT_USER\SOFTWARE\Python\PythonCore\ along with 2.4,2.6 and a few others
3.6 is in HKEY_LOCAL_MACHINE\SOFTWARE\Python\PythonCore\ and its path leads to the VS location.

exported the 3.8 keys and modified them to HKEY_LOCAL_MACHINE then imported them back to registry. 3.6 deleted.

list tools in eclipse now runs and displays the following on the console.

[image]

the log file however still says its failing in some way 

[image]

and the sdkconfig tool is not working.

umfundi, 2020-03-31:
A project compile however now seems to work.

umfundi, 2020-03-31:
it looks like the steps I took have got me a working install.

closed 2020-03-31 by umfundi