espressif/esp-idf #5026: BLE_MESH transport.c buffer overflow

pokerfacett, 2020-03-30:
INSTRUCTIONS
============
When I review the code of BLE MESH , I found a buffer overflow in esp-idf/components/bt/esp_ble_mesh/mesh_core/transport.c master branch 1470 lines:
   
`  /* Location in buffer can be calculated based on seg_o & rx->ctl */
    memcpy(rx->buf.data + (seg_o * seg_len(rx->ctl)), buf->data, buf->len);`

the buf could controlled by attacker through proxy Pattern，and seg_o is parsed from buf as follows:
[image]
so an attack could control the offset of rx->buf.data  and write more data in rx->buf.data which buf->data is controlled by attack and parsed from ble data.

FIX SUGGESTION
============
`if(rx->buf.data + (seg_o * seg_len(rx->ctl)) + buf->len< rx->buf.data.len){
    memcpy(rx->buf.data + (seg_o * seg_len(rx->ctl)), buf->data, buf->len);
}`

Campou, 2020-03-30:
Hi @pokerfacett,

Thanks for the feedback.
1. First `rx->buf.data.len` should be `rx->buf.len`, right? And `rx->buf.len` is only updated when `seg_o` equals to `seg_n`. So if the suggested fix is added, the following `memcpy` will never be reached.
2. `rx->buf.data` is a pointer, and `rx->buf.len` is an integer, i don't think we should compare a pointer with a integer.
3. Before the low transport layer tries to receive a segmented message, `seg_o` will be checked firstly, and it will also check if the total length of the message is too large to be received, which is `sdu_len_is_ok` in LINE 1531. So in this case, the buffer should not be overflowed.

Thanks.

pokerfacett, 2020-03-31:
> Hi @pokerfacett,
>
> Thanks for the feedback.
>
> 1. First `rx->buf.data.len` should be `rx->buf.len`, right? And `rx->buf.len` is only updated when `seg_o` equals to `seg_n`. So if the suggested fix is [...]

Thanks， I review the code it is true and I learned a lot

closed 2020-03-31 by pokerfacett