espressif/esp-idf #5031: Modbus Master/Slave example isn't working in RTU and ASCII mode

waterfox207, 2020-03-30:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it):
    // v4.1-beta1-63-g9f024df9e
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [|ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB|]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
Master and Slave should exchange data like shown in the Readme of the example. 

### Actual Behavior
I have tried RTU and ASCII mode. Both are not working correct. 
In my Setup I have 2 ESP Dev kits connected with two MAX485 together like shown in the readme. The communication in UART works just fine. Also the echo RS485 example works with my setup. 

For the ASCII mode I've compiled Master and slave with default config. 
On my Master monitor i get this Errors. I don't get any read successful message. 

> E (4559) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x107) (ESP_ERR_TIMEOUT).
E (4559) MASTER_TEST: Characteristic #3 (Humidity_2) read fail, err = 263 (ESP_ERR_TIMEOUT).
E (4579) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (4579) MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure. <

My Slave monitor looks like this: 
>I (85844) SLAVE_TEST: HOLDING READ (74030907 us), ADDR:1, TYPE:2, INST_ADDR:0x3ffb28b0, SIZE:2
I (87134) SLAVE_TEST: COILS READ (74031360 us), ADDR:3, TYPE:8, INST_ADDR:0x3ffb28a4, SIZE:2
I (87134) SLAVE_TEST: INPUT READ (75448608 us), ADDR:1, TYPE:8, INST_ADDR:0x3ffb28a0, SIZE:2
I (88414) SLAVE_TEST: COILS READ (75449061 us), ADDR:1, TYPE:2, INST_ADDR:0x3ffb28b0, SIZE:2
I (88414) SLAVE_TEST: COILS READ (76737395 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb289d, SIZE:8
E (88434) MB_PORT_COMMON: xMBPortEventPost(103): xMBPortEventPost: Post message failure.<

If I know change the settings on Both devices to RTU mode, I don't get any message on the Slave monitor. The Master Monitor looks the same as in ASCII mode. 

I've also tried to increase  CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND (as mentioned in the Readme) but this doesn't change anything. 

### Steps to reproduce

1. Setup up to ESP 32 with the Master and Slave example
2. Connect them with 2 RS485 to ttl Converter (MAX485)

// If possible, attach a picture of your setup/wiring here.
[image]

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder) --> use default settings
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-03-31:
@waterfox207 Thanks for reporting, we will look into.

alisitsyn, 2020-03-31:
@waterfox207,
Thank you for sufficient information and clear explanation of your issue. I tried to reproduce your issue with release v4.1-beta1 (could not find your commit 9f024df9e). 

1. I think your connection looks good and communication should be reliable.
2. I was able to start communication between master and slave board with master and slave examples on my side and they are able to communicate without any issues in RTU and ASCII mode (release v4.1-beta1).
3. According to your log, slave board is not able to receive all requests from master board reliably.

See my connections and logs below for your reference:
[image]

[comm_logs_master_slave_examples.txt](https://github.com/espressif/esp-idf/files/4407043/comm_logs_master_slave_examples.txt)

For further investigation of your issue I would suggest you to set logging verbosity to debug as showed below then start master and slave monitor and save logs into text file.
[image]

Please send me the information described above. Thanks.

waterfox207, 2020-03-31:
Hey @alisitsyn,

thanks for your fast testing of my problem. I've done the RTU and ASCII Debug log:

- [master-ascii.log](https://github.com/espressif/esp-idf/files/4407985/master-ascii.log)
- [slave-ascii.log](https://github.com/espressif/esp-idf/files/4407987/slave-ascii.log)
- [master-RTU.log](https://github.com/espressif/esp-idf/files/4408001/master-RTU.log)
- [slave-RTU.log](https://github.com/espressif/esp-idf/files/4408002/slave-RTU.log)

alisitsyn, 2020-03-31:
Hi @waterfox207,

Thank you for your logs. It was verified. Your boards can communicate correctly only one direction. Master can only send the data and slave is able to receive data. Slave sends response but master is not able to receive it. This usually related to incorrect connection of your ~RE/DE pins on bus driver to your board GPIO18. Please check that the RTS pin (GPIO18) is connected correctly on master and slave board. I can guess that on your master board on ~RE/DE pin is always logic 1 or on your slave board ~RE/DE pin is constant logic 0. Also please check the pin configuration in menuconfig and actual connections to RS485 bus driver. Check signal connections from your pins to RS485 line driver.  

Let me know once you get result. Do you have logic analyser or oscilloscope. On next stage it can be useful.

Thanks.

waterfox207, 2020-03-31:
I've checked all my connections already. I also tried to rebuild all the jumper wires again. But every time I run into the same Problem. 
 
Sadly I have either an oscilloscope nor an logic analyzer. I only have an Multimeter. 

Do you have an Idea how i can fix my issue? 

Maybe it will help if I order something like this: https://www.ebay.de/itm/8-Kanal-USB-Logik-Analyser-24-MHz-Logic-Analyzer-Arduino-Logikanalysator-0-5-5V/252756113817?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20170511121231%26meid%3D26c59c3ff4b345deb5472fe6ac0c0e58%26pid%3D100675%26rk%3D1%26rkt%3D15%26mehot%3Dpp%26sd%3D252756113817%26itm%3D252756113817%26pmt%3D0%26noa%3D1%26pg%3D2380057&_trksid=p2380057.c100675.m4236&_trkparms=pageci%3Aa3d9837a-733a-11ea-8eaa-f2405f66637e%7Cparentrq%3A3024765e1710ad31c394dda4fff62a26%7Ciid%3A1

alisitsyn, 2020-03-31:
This simple analyser on the link will work just fine to check UART traffic. What you need is just check signals on your master board: RXD, TXD, RTS. Then the same on your slave board or even all signals together during communication can give the clear picture of the root reason of issue. It is 98% probability hardware issue not related to SW of examples or Modbus driver itself.

waterfox207, 2020-03-31:
I also think now that this isn't a software problem. 
I've done some research on the max485 but were not able to find out wich version I have.
Maybe it's the 5V version. than it could be a solution to connect them with a level shifter? Do you think this will help?
Otherwise i will wait until my Logic Analyzer will arrive and give you then an update.

alisitsyn, 2020-03-31:
If you use two the same versions of RS485 5V drivers and use 3.3V power supply it can work just fine without level shifters (at least do not cause communication issues) . Logic levels should be within the scope in this case. I think it is better to use your oscilloscope and record RXD, TXD, RTS signals with the trigger set to start bit of byte on RXD.
Another reason: The MAX485 transceiver could be damaged after connection of boards and connection of power over USB. They are connected together and when you connect boards to USB their pins can have spikes with active logic output. In this case usually the power consumption of driver exceeds limit. Good luck!

waterfox207, 2020-04-01:
So my Analyzer arrived today and I have done some measurements. The logs on the monitor are the same as yesterday. 
--> ASCII - Mode:
[image]

--> RTU - Mode:
[image]

- [Modbus-log.zip](https://github.com/espressif/esp-idf/files/4415358/Modbus-log.zip)

I hope you can help me with those files.

waterfox207, 2020-04-01:
Small update:
I've Done some Testing over the Day.
If i have the analyzer connected I'm able to exchange some data in RTU mode. If I disconnect it from USB I only get timeouts on the Master side. 

Maybe i need some resistors in my setup for proper work?

alisitsyn, 2020-04-02:
@waterfox207,

Hmm. Your data is really surprise for me. 
1. On first picture (RXD master/slave, TXD master/slave, RTS master/slave) I see ASCII communication between master and slave board and it is correct and shows that software forms data correctly and boards can communicate each other just fine. 
2. First picture (D+, D- signals) they are completely incorrect and I don't see communication between boards. Signals are wrong and it looks like your RS485 bus tranceivers do not work at all. Bus communication trafic is incorrect. I don't know how you can get the signals on your RXD and TXD lines correct when bus communication is completely incorrect. The only one way i can explain this is that your RXD, TXD pins connected directly between boards or you connected to wrong pins.
3. RTU communication of master is correct but slave probably is still compiled in ASCII mode or even is not flashed correctly.

According to information above can state that your bus drivers are damaged and can not work correctly. The RS485 board showed on your picture shall include all required elements including pull up resistors. Below is the correct communication trafic between Modbus master and slave boards based on latest master:
[Master_slave_comm_ASCII_1.zip](https://github.com/espressif/esp-idf/files/4419097/Master_slave_comm_ASCII_1.zip)

I can advice to change your hardware. I don't see any issues related to SW for now.

waterfox207, 2020-04-03:
It looks like I had an bad MAX485 I bought a new one wich works correct.

closed 2020-04-03 by waterfox207

alisitsyn, 2020-04-05:
I glad to hear it was helpfull. Good luck with your project!