espressif/esp-idf #5032: ESP-IDF doc :  mcpwm_cache_enable documentation misleading

vortex314, 2020-03-30:
ESP-IDF documentation on mcpwm_capture_enable seems to suggest that the parmeter num_of_pulse is a 32 bit value. However the underlying register PWM_CAP_CH0_CFG_REG has only 8 bit's for the prescaling. So any value above 255 will give unexpected results. 
Also there is an offset by 1 to take into account. 

esp_err_t mcpwm_capture_enable(mcpwm_unit_tmcpwm_num, mcpwm_capture_signal_tcap_sig, mcpwm_capture_on_edge_tcap_edge, uint32_t num_of_pulse)

Hope this helps somebody with the same issues.

Alvin1Zhang, 2020-03-31:
@vortex314 Thanks for reporting, we will look into and provide feedback.

Alvin1Zhang, 2020-03-31:
@vortex314 Thanks for reporting, we will fix the issue. Thanks. @koobest

closed 2020-04-01 by vortex314