espressif/esp-idf #5033: 文档jtag-debugging章节出现错误

lgnativs, 2020-03-31:
> 上传待调试的应用程序
> 您可以像往常一样构建并上传 ESP32 应用程序，具体请参阅 第八步：编译工程 章节。
> 
> 除此以外，还支持使用 OpenOCD 通过 JTAG 接口将应用程序镜像烧写到闪存中，命令如下:
> `openocd -f board/esp32-wrover-kit-3.3v.cfg -c "program_esp filename.bin 0x10000 verify exit"`
> 
> 其中 OpenOCD 的烧写命令 program_esp 具有以下格式：
> 
> program_esp <image_file> <offset> [verify] [reset] [exit]
> 
> image_file - 程序镜像文件存放的路径
> 
> offset - 镜像烧写到闪存中的偏移地址
> 
> verify - 烧写完成后校验闪存中的内容（可选）
> 
> reset - 烧写完成后重启目标（可选）
> 
> exit - 烧写完成后退出 OpenOCD（可选）
> 
> 现在可以进行应用程序的调试了，请按照以下章节中讲解的步骤进行操作。

[上传待调试的应用程序](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/jtag-debugging/index.html#jtag-upload-app-debug)

此处的program_esp应该为program_esp32

suda-morris, 2020-03-31:
Hi @lgnativs probably you're using the old openOCD, right? In the latest openocd, we have changed the command name to `program_esp` to support multiple targets (i.e. esp32, esp32-s2). Please consult https://github.com/espressif/openocd-esp32/releases and download the latest verison.

igrr, 2020-03-31:
Alternatively, you may be using the correct OpenOCD version, but looking at latest documentation instead of the one matching your IDF version. Please mention which IDF version you are working with.

lgnativs, 2020-03-31:
我用的IDF是V4.0-beta1,可能是这个原因吧.谢谢 @igrr

closed 2020-03-31 by lgnativs