espressif/esp-idf #5035: BLE_MESH proxy_client.c stack buffer overflow

pokerfacett, 2020-03-31:
INSTRUCTIONS
============
bt_mesh_proxy_client_adv_ind_recv function has buffer overflow：
[image]

When I post packet as follow when bt_mesh scan:
[image]

It could cause 8 bytes buffer overflow

Fix Suggestion
============
```
if(buf->len <= sizeof(ctx.net_id.net_id)){
    memcpy(ctx.net_id.net_id, buf->data, buf->len);
}
```

Alvin1Zhang, 2020-03-31:
@pokerfacett Thanks for reporting and really appreciate you bring fix suggestion, we will look into . Thanks.

pokerfacett, 2020-04-01:
> @pokerfacett Thanks for reporting and really appreciate you bring fix suggestion, we will look into . Thanks.

Any progress？

Campou, 2020-04-01:
Hi @pokerfacett 

Thanks, i will add a length check here.

pokerfacett, 2020-05-21:
commit https://github.com/espressif/esp-idf/commit/fab9b944a47067024ac61e48a08945e02b924369

closed 2020-05-21 by pokerfacett

projectgus, 2020-06-24:
Hi @pokerfacett,

Thanks for reporting this issue so clearly. We've analyzed the impact and although this is a buffer overflow onto the stack, it's not a security bug in any of the configurations or versions we've checked.

The reason for this is that the maximum size of the buf->len argument to memcpy that an attacker can inject is 16 bytes. Of this, 8 bytes are copied correctly to the destination buffer, one byte overflows to another field in the destination buffer, and 7 bytes overflow onto the stack.

The stack frame for the bt_mesh_proxy_client_adv_ind_recv() function is 48 bytes long (`entry a1, 48`).  `a1` is the stack pointer register. The destination structure is the only local variable stored on the stack, so the `memcpy()` call evaluates as `memcpy(a1, src buffer, 16)`. This means that although the structure on the stack overflows by up to 7 bytes, these overflow bytes are all unused bytes of this function's stack so they don't corrupt any other memory. These bytes are never read back, they might be written over if this function makes another function call.

In addition, xtensa stack frames are always 16-byte aligned so the smallest possible stack frame for this function is 16 bytes.

The attacker would need `buf->len` to be the length of the stack frame or more in order to trigger a security bug.

Regarding the other issue you've reported, thanks for this report also. We're investigating this currently and will have an update for you soon.

In future, when reporting potential security bugs please consider taking advantage of the [ESP32 Bug Bounty Program](https://esp32.com/viewtopic.php?f=10&p=7266&sid=b1e9e459e291f6aed17185dd21aed1f4).