espressif/esp-idf #5036: Add ext3/ext4 filesystem

mikhail-khalizev, 2020-03-31:
**Is your feature request related to a problem? Please describe.**

At the moment, ESP-IDF supports only one full-fledged file system - FAT. But it is not journaling. Because of this, data loss may occur during a sudden power outage.

In fact, I constantly lose data. Because the write cycle to a flash card consists of two parts - erasing sector and writing sector. And if, for example, we just simply create and delete files in the directory on a regular basis, then when the power is turned off, it is highly likely that the sector describing the directory in the FAT will be erased, but new data will not be written to it yet. And thus, all files in the directory will be lost.

**Describe the solution you'd like**

The problem can be solved by adding support for the journaled file system.

For example project https://github.com/gkostka/lwext4 implements ext2/ext3/ext4 filesystem library for microcontrollers.

igrr, 2020-03-31:
@mikhail-khalizev unfortunately https://github.com/gkostka/lwext4 is licensed as GPL 2.0, which is incompatible with ESP-IDF license.

You can try using littlefs filesystem — it has existing ports to ESP-IDF (components), and seems to work okay w.r.t power outage.

mikhail-khalizev, 2020-03-31:
Wow. I did not know about littlefs filesystem before. It completely solves the problem. Thanks a lot.

closed 2020-03-31 by mikhail-khalizev