espressif/esp-idf #5037: To update flash encryption document for flashing pre-encrypted firmware

AxelLin, 2020-03-31:
In Flash Encryption document:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#using-host-generated-flash-encryption-key

The "Using Host Generated Flash Encryption Key" section mentioned:
It is possible to pregenerate the flash encryption key on the host computer and burn it into the ESP32’s eFuse key block.
This allows data to be pre-encrypted on the host and flashed to the ESP32 without needing a plaintext flash update.

However, the document does not mention how to flash pre-encrypted firmware at all.
I think this needs update.

Alvin1Zhang, 2020-03-31:
@AxelLin Thanks for your suggestion in the documentation,  we will evaluate the suggestion and provide feedback. Thanks.

projectgus, 2020-04-14:
Hi @AxelLin,

Thanks for pointing this out and sorry for the delay in getting back to you.

> However, the document does not mention how to flash pre-encrypted firmware at all.

For simplicity, we try not to descrbe every possible way to work with flash encryption. Instead, we choose some recommended workflows for Development & Production and then document these. The workflow described at the link can be summarised as:

1) Before first boot, pre-load the encryption key to efuse
2) Configure with "Development" mode
3) Flash plaintext binaries, the device will encrypt itself on first boot from the plaintext files and using the pre-loaded key.
4) Use the `encrypted-flash` commands for any additional flash updates.

This is the recommended workflow for using a pre-generated key for development.

ESP-IDF v3.x actually did have documentation about how to encrypt the files on the host and then flash them:
https://docs.espressif.com/projects/esp-idf/en/v3.3.1/security/flash-encryption.html#reflashing-with-pregenerated-key

This was removed in V4.0 and replaced with the "Development mode" workflow. We made this decision for two reasons:

1) It's simpler, the build system can do everything instead of requiring manual steps.
2) We found some users chose to deploy pre-encrypted binaries in production and shared the same key across all devices. We can't prevent this, but it's considered to be a security risk so it's discouraged.

(The old workflows still all work, but as I said we don't want document every possible workflow - it's too complex.)

We still want to make sure that ESP-IDF users have all of the tools they need for their product development. Can you explain why you need to flash a pre-encrypted binary, instead of using Development mode?

AxelLin, 2020-04-17:
Hi @projectgus 

I understand the risk of pre-encrypted with the same key.
However, I'd like to compare the production speed difference at the moment.
IMHO, if it's actually possible to flash pre-encrypted firmware it should
be well documented. Otherwise, people just try to dig old discussion threads
and may find out-of-date information.

Yet another issue might be related to pre-encrypted firmware:
Is it possible to do OTA with pre-encrypted firmware?
From the information on the discussion forum I understand your point is the
firmware should be in a safe place. However, it's not a perfect world.
In some case, the customer asks to allow manually upload firmware for OTA.
We would prefer to use pre-encrypted firmware rather than unencrypted firmware
for customers in such case. (Although I still don't know how to OTA pre-encrypted firmware at this moment. Is it possible?)

tim-nordell-nimbelink, 2020-05-08:
Hi @projectgus and @AxelLin -

Just wanted to chime in that for us in production, we're encrypting with the host PC.  We found that it was much faster to encrypt with the PC to the target than it was for the target to do the in-place encryption.  Our production flow is thus:

1. Generate a random flash key for the unit, and record it on the functional tester
2. Generate a random signing key for the unit, and record it on the functional tester
3. Burn in the following eFuses:
 - BLK1, enabling write protection only
 - BLK2, enabling write protection only
 - CRYPT_CONFIG 15
 - CRYPT_CNT 1, enabling write protection
 - ABS_DONE_0 1, enabling secure boot
4. Generate per-unit encrypted image and write into unit
5. Run functional test in unit
 - When the functional test completes successfully, the firmware in the unit then burns the eFuses for read protection on BLK1 and BLK2

This was a *lot faster* than doing the in-place encryption.  I don't have the numbers in front of me, but just wanted to chime in that it's a valid way of doing it, and IMHO better than doing the in-place encryption that's done with the example since it ends up with a faster functional test in production.

 \- Tim

projectgus, 2020-05-20:
Hi @tim-nordell-nimbelink @AxelLin ,

Thanks Tim for updating with your process and sorry to both of you for the slow turnaround.

You're right that we should document a process for pre-encrypting firmware. We're having some internal discussion about providing better support for this kind of factory workflow (while still discouraging sharing flash encryption keys). It may be some time before something is ready, unfortunately, but this issue will be updated when it is.

> Although I still don't know how to OTA pre-encrypted firmware at this moment. Is it possible?)

Not supported out of the box with the current OTA APIs, unfortunately. We also plan to support a way to do encrypted OTA of this type (probably using a separate key that's protected by flash encryption inside the device), but no ETA for this I'm afraid.

tim-nordell-nimbelink, 2020-05-20:
Hi @projectgus -

For us, we're set for our production workflow.  However, the following may help other folks move into production.

Maybe some kind of tool that could be configured where you can give it a list of partitions, images for those partitions, including NVS encrypted partitions?  Then it could automate the generation and burning actions on the PC.  It should probably optionally save the keys utilized on the local PC, and such.  If a full tool had been available I probably would have used it, but instead I have an external bash shell script that is calling into espefuse.py (for setting keys into the eFuses), espsecure.py (for pre-encrypting the partitions with the PC, and signing the bootloader with the unique key), esptool.py (for flashing), and nvs_partition_gen.py (for encrypting the default NVS partition correctly) to accomplish the PC side manipulation required.

There are a lot of moving pieces within the flow with the current tools inside the ESP-IDF.  If a single tool combined all of those (even if it invokes external tools on your behalf) such that you could provide as parameters:

- (optional) Folder to save encryption keys and digest keys into
- (optional) Ability to defer marking keys as not readable
- List of binaries to program partitions from.
- Flags for each binary indicating how it needs to be mutated for flashing.  E.g. unencrypted, encrypted, and inputs for nvs_partition_gen.py for generating a default NVS partition

Maybe somehow partitions.csv could be incorporated into the tool for sourcing flags for how to mutate each partition as it's flashed?

Of course, some of the above is different with "secure boot v2" (namely the digest key part is effectively an known value to everyone) a similar process could probably be used.

- Tim

tim-nordell-nimbelink, 2020-05-20:
I would also note that deferring marking the eFuse keys as non-readable should be done even with the "normal" flow until it's been programmed successfully (just as the last step within the automated process), just in case it gets interrupted for whatever reason.

- Tim

ahmedwahdan, 2021-03-23:
Hello,
If we are using flash protection with development option with pre-generated key, why we are using "encrypted-flash" option?
I don't provide the key here!

projectgus, 2021-03-23:
Hi @ahmedwahdan,

I saw you posted the same question on the forum, I've replied to you there: https://esp32.com/viewtopic.php?f=2&t=20046&p=73741#p73741

closed 2021-07-16 by espressif-bot

roysG, 2021-09-08:
Hi,
I generated a key and burned it to device.
When i enabled flash encryption, set it to development mode and run the command: idf.py flash monitor

I get errors that i burned regular plaintext(not encrypted).
When i run the command "idf.py encrypted-flash monitor" then it run ok.

Then I tried to run the flash encryption in production,
I changed in the menu to production and run the command "idf.py flash monitor".

But i got errors (running without ending):

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

What do i miss?

I follow this document:
https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/security/flash-encryption.html#flash-encryption-algorithm

projectgus, 2021-09-08:
@roysG I don't work for Espressif any more so I'm afraid I can't help you. As this issue has been closed, you might not get a reply - suggest maybe posting on https://esp32.com or opening a new issue to make sure someone sees it. :handshake: