espressif/esp-idf #5038: Unable to use the init_priority extension with C++, it gets ignored on execution

leofabri, 2020-03-31:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.0
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: No (VScode)
- Power Supply:         USB

## Problem Description

I'm trying to get the `__attribute__(init_priority(int))` extension to work with my code (written in C++), but it looks like it gets ignored with out-of-the-box configuration. 

In particular, I have the necessity to initialize a global object with a specified priority, like so:
```c++
Variable MyAppVariable __attribute__((init_priority(1000)));
```

### Expected Behavior

Init sequence should follow the assigned priority.

### Actual Behavior

When I set multiple variables with different priorities, nothing seems to be affected, and the initialization seems to follow their order of definition.

### Steps to reproduce

Just try to assign different priorities to some objects and you'll probably face the same problem.

## The init_priority extension

I was able to use this rule with esp-idf v3.
After looking for more info about this extension, I noticed that it should be possible to enable it with appending  "-finit-priority" to the **idf_build_set_property** as documented here: https://gcc.gnu.org/onlinedocs/gcc-2.95.3/gcc_2.html.

I tried to add:

```c++
list(APPEND compile_options "-finit-priority")
``` 
to CMakeLists.txt **[here](https://github.com/espressif/esp-idf/blob/4fe04f1151c2faf4b034e9c73592fbd443f96b99/CMakeLists.txt#L82)** but, when I try to compile I get the following erorrs:

```terminal
[1/60] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/gpio_periph.c.obj
FAILED: esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/gpio_periph.c.obj 
C:\Users\my_user\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/development/esp/components/soc/esp32/include -IC:/development/esp/components/soc/include -IC:/development/esp/components/log/include -IC:/development/esp/components/esp_rom/include -IC:/development/esp/components/esp_common/include -IC:/development/esp/components/xtensa/include -IC:/development/esp/components/xtensa/esp32/include -IC:/development/esp/components/esp32/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -finit-priority -fno-stack-protector -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -DBOOTLOADER_BUILD=1 -MD -MT esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/gpio_periph.c.obj -MF esp-idf\soc\CMakeFiles\__idf_soc.dir\esp32\gpio_periph.c.obj.d -o esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/gpio_periph.c.obj   -c C:/development/esp/components/soc/esp32/gpio_periph.c
xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-finit-priority'; did you mean '-finit-derived'?
[2/60] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_clk_init.c.obj
FAILED: esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_clk_init.c.obj 
C:\Users\my_user\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/development/esp/components/soc/esp32/include -IC:/development/esp/components/soc/include -IC:/development/esp/components/log/include -IC:/development/esp/components/esp_rom/include -IC:/development/esp/components/esp_common/include -IC:/development/esp/components/xtensa/include -IC:/development/esp/components/xtensa/esp32/include -IC:/development/esp/components/esp32/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -finit-priority -fno-stack-protector -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -DBOOTLOADER_BUILD=1 -MD -MT esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_clk_init.c.obj -MF esp-idf\soc\CMakeFiles\__idf_soc.dir\esp32\rtc_clk_init.c.obj.d -o esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_clk_init.c.obj   -c C:/development/esp/components/soc/esp32/rtc_clk_init.c
xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-finit-priority'; did you mean '-finit-derived'?
[3/60] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/cpu_util.c.obj
FAILED: esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/cpu_util.c.obj 
C:\Users\my_user\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/development/esp/components/soc/esp32/include -IC:/development/esp/components/soc/include -IC:/development/esp/components/log/include -IC:/development/esp/components/esp_rom/include -IC:/development/esp/components/esp_common/include -IC:/development/esp/components/xtensa/include -IC:/development/esp/components/xtensa/esp32/include -IC:/development/esp/components/esp32/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -finit-priority -fno-stack-protector -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -DBOOTLOADER_BUILD=1 -MD -MT esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/cpu_util.c.obj -MF esp-idf\soc\CMakeFiles\__idf_soc.dir\esp32\cpu_util.c.obj.d -o esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/cpu_util.c.obj   -c C:/development/esp/components/soc/esp32/cpu_util.c
xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-finit-priority'; did you mean '-finit-derived'?
[4/60] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_sleep.c.obj
FAILED: esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_sleep.c.obj 
C:\Users\my_user\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/development/esp/components/soc/esp32/include -IC:/development/esp/components/soc/include -IC:/development/esp/components/log/include -IC:/development/esp/components/esp_rom/include -IC:/development/esp/components/esp_common/include -IC:/development/esp/components/xtensa/include -IC:/development/esp/components/xtensa/esp32/include -IC:/development/esp/components/esp32/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -finit-priority -fno-stack-protector -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -DBOOTLOADER_BUILD=1 -MD -MT esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_sleep.c.obj -MF esp-idf\soc\CMakeFiles\__idf_soc.dir\esp32\rtc_sleep.c.obj.d -o esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_sleep.c.obj   -c C:/development/esp/components/soc/esp32/rtc_sleep.c
xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-finit-priority'; did you mean '-finit-derived'?
[5/60] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_time.c.obj
FAILED: esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_time.c.obj 
C:\Users\my_user\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/development/esp/components/soc/esp32/include -IC:/development/esp/components/soc/include -IC:/development/esp/components/log/include -IC:/development/esp/components/esp_rom/include -IC:/development/esp/components/esp_common/include -IC:/development/esp/components/xtensa/include -IC:/development/esp/components/xtensa/esp32/include -IC:/development/esp/components/esp32/include -mlongcalls -Wno-fram [... cut ...] dir/esp32/rtc_wdt.c.obj   -c C:/development/esp/components/soc/esp32/rtc_wdt.c
xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-finit-priority'; did you mean '-finit-derived'?
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
```

I don't know if this is the right way of doing it or not. Unfortunately, I couldn't find much documentation about this. 

Since this is a gcc-specific feature, I wish I could use it. Any suggestion?

Regards, 
Fabrizio

igrr, 2020-03-31:
Could you please try adding SORT() inside KEEP in this line? 

https://github.com/espressif/esp-idf/blob/4fe04f1151c2faf4b034e9c73592fbd443f96b99/components/esp32/ld/esp32.project.ld.in#L279

leofabri, 2020-03-31:
@igrr Should I make it look like this?
```
    KEEP (*(EXCLUDE_FILE (*crtend.* *crtbegin.*) .ctors SORT(.ctors.*))) 
```

Doing this way, the init priority **is finally working**. Thank you so much! 

Should we add this as a pull request or something, in order to be functional to everyone?

igrr, 2020-03-31:
Thanks for confirming. If you'd like to make a PR, please go ahead (and make sure to apply to similar fix to esp32s2 target as well). Otherwise we will fix commit this fix ourselves.

leofabri, 2020-03-31:
I would like to submit a PR, but since I've been working only on an older version of esp-idf (at least compared to the master branch) I would prefer to leave this commit to you. 

Thank you again @igrr you saved my day.

closed 2020-04-21 by leofabri