espressif/esp-idf #5043: exceed the default bootloader size esp32 flash encryption

SERIDJ, 2020-03-31:
hi,

i have problem,  when i enabled flah encryption development mode, the size of  bootloader (0x7000 = 28672 BYTE) increase , so solve that  i increase  the partition table offset from 0x8000 to 0x10000, that work in the first boot, but when i reset esp32 i have this error in  TERMINAL : 

[image]

projectgus, 2020-04-02:
Hi @SERIDJ,

I need some more information in order to help you debug this:

* What ESP-IDF version do you have?

> that work in the first boot,

* Can you explain exactly what this means, please? What steps did you follow to flash the board, and what output did you see after you did that?

* Can you please run `espefuse.py -p PORT summary` to dump the efuse status for the board you have.

> i cant comme back bcs the development mode is enabled ...
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html

[Development mode](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#flash-enc-development-mode) is a flash encryption mode that is designed to let you reflash (with `make encrypted-flash` or `idf.py encrypted-flash`). Do you mean to say that Secure Boot is also enabled, so you can't reflash the bootloader?

SERIDJ, 2020-04-02:
hi @projectgus  thanks,

so i use ESP IDF V4.0.

to flash esp32 i use ECLIPSE IDE after instaling Espressif IDF Tools Manager (Git Executable Location + ESP IDF Directory ) 

i  can run espefuse.py -p PORT summary in cmd windows 10 command 

i want to use Development mode  (flash encryption ) so i folow those steps :
 
[image]

i cant enable this mode until i change in menuconfig the offset of partition table to 0x10000 and i have in TERMINAL eclipse somthing lik : 
[image]

that mean Development  mode  is enabled 

so when i reset my esp32 i have this error each 1s 

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

SERIDJ, 2020-04-02:
for more  information about my menuconfig ; 

[image]

[image]

[image]

[image]

[image]

igrr, 2020-04-02:
I think the trick is that after the first flash, once flash encryption is enabled, you need to use `idf.py encrypted-flash` command instead of `idf.py flash`. You can edit or create a new launch configuration, changing the flash command. Please see https://github.com/espressif/idf-eclipse-plugin#customizeLaunchConfig for details.

SERIDJ, 2020-04-02:
hi @igrr 
thanks for ur replay, yes thats true i tried  with cmd command using idf.py enccrypted-flash (windows 10) ,  i had  continuously printing the following message 

[image]

igrr, 2020-04-02:
It looks like the partition table has not been loaded or generated correctly.
Could you please add the full logs of: `idf.py encrypted-flash` command, and `idf.py partition_table`?

Also, kind request to paste logs as text, rather than screenshots. The text in screenshots is hard to read and impossible to search.

SERIDJ, 2020-04-02:
yes yes, i know when i change the offset  maybe i need to leave all partition offsets blank in the CSV file but i dont know how i can do that ...

for **idf.py encrypted-flash** work fine : 
[image]

and  **idf.py partition_table**  : 

 C:\Users\mohammed.seridj\Desktop\idfeclipse\flash_encryption>idf.py partition_table
Checking Python dependencies...
Python requirements from C:\Users\mohammed.seridj\esp\esp-idf\requirements.txt are satisfied.
Executing action: partition_table
Running ninja in directory c:\users\mohammed.seridj\desktop\idfeclipse\flash_encryption\build
Executing "ninja partition_table"...
ninja: no work to do.

Partition Table build complete. To flash, run this command:
C:\Users\mohammed.seridj\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe ..\..\..\esp\esp-idf\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after no_reset write_flash 0x10000 build\partition_table\partition-table.bin
or run 'idf.py -p (PORT) partition_table-flash'

SERIDJ, 2020-04-02:
yes i donk know and what i change in this table 

[image]

[image]

[image]

SERIDJ, 2020-04-03:
i try with blink and hello_world work fine 

[image]

[image]

so i have problem with flash_encryption code roboot automatically ...

projectgus, 2020-04-03:
Hi @SERIDJ ,

Here is a modified copy of the `partitions_example.csv` file from the `examples/security/flash_encryption` example of IDF:

[partitions_example.csv.txt](https://github.com/espressif/esp-idf/files/4426651/partitions_example.csv.txt)

You should be able to use this with a partition table that has moved to a different offset, such as what you have.

You will need to rename the file as it has the `.txt` extension appended, to be accepted for upload to GitHub.

Remember to use encrypted-flash target if the flash encryption feature has already been enabled in Development mode.

We'll fix this CSV file properly in the example supplied with IDF, and update this issue once that fix has merged.

SERIDJ, 2020-04-03:
@projectgus  ah ok i understand well thanks, 

so i dont know why i cant run :  espefuse  and esptool : command" is not recognized as an internal or external command, operable program or batch file

i do all that (flash encryption and secure boot ) to hide MQTT keys so waht is the best thing, encrypte all flash esp32 or i can hide only keys (client.crt and client.key)  ? 

and what is the best method to crypte flash memory : 
1-     Using ESP32 Generated Flash Encryption Key (interne) 
2-	Using Host Generated Flash Encryption Key (externe) 

  
[image]

SERIDJ, 2020-04-07:
@igrr hi sir, 
i want to use eclipse but i cant run idf.py encrypted-flash command instead idf.py flash ? 

[image]

igrr, 2020-04-07:
There seem to be two issues:

* idf.py is duplicated in the command (`idf.py idf.py`)
* the error reported: `IDF_PATH environment variable is set to c:\users\mohammed.seridj\esp but C:\Users\mohammed.seridj\esp\esp-idf\tools\idf.py path indicates IDF directory c:\users\mohammed.seridj\esp\esp-idf` — make sure IDF_PATH is set correctly in Eclipse, it should point to `c:\users\mohammed.seridj\esp\esp-idf` instead of `c:\users\mohammed.seridj\esp`.

SERIDJ, 2020-04-07:
@igrr yes thanks, 

I fixed the second issues ( change  IDF_PATH).

The first one ( idf.py is duplicated in the command) , i dont know where i can change or delete this duplicate ?

SERIDJ, 2020-04-07:
@igrr  yes yes that work fine afther fixe that, thanks

i have another question about eclipse  (windows 10 ) : 
i  can debug ESP32 using JTAG  of  **ESP32-Ethernet-Kit_A_V1.1**  ? 

[image]

Alvin1Zhang, 2020-04-09:
@SERIDJ Thanks for reporting and follow up. For **Eclipse** specific issues, you could help raise an issue in our Eclipse repo. at https://github.com/espressif/idf-eclipse-plugin/issues, and for "how" questions or third party issues, you could raise a post on our forum at https://esp32.com. If you have a new issue, suggest you could file a new ticket. Thanks.

projectgus, 2020-06-03:
Closing as the original issue seems to be resolved. For follow-up questions, please make use of the resources linked by my colleague in the above comment .

closed 2020-06-03 by projectgus