espressif/esp-idf #5045: BLE_MESH join mesh network buffer overflow in prov.c gen_prov_start function

pokerfacett, 2020-04-01:
INSTRUCTIONS
============
[image]

And the entry point is bt_mesh_scan_cb:
[image]
 
If the type == BLE_MESH_DATA_MESH_PROV and will goto fhunction  bt_mesh_pb_adv_recv:
[image]
 
Here, the param rx.gpc is parsed from buf which attacker could controlled it equals 0x01. Goto function gen_prov_recv
[image]

rx.gpc == 0x01 will call function gen_prov_start and pass buf to it which could controlled by attacker. Goto gen_prov_start 
[image]

line 1561 could cause buffer overflow which an attacker send mesh packet as follows:

[image]

FIX SUGGESTION
============
```
if(buf->len <= link.rx.buf->len){
     memcpy(link.rx.buf->data, buf->data, buf->len);
}
```

Alvin1Zhang, 2020-04-01:
@pokerfacett Thanks for reporting and really appreciate you bring fix suggestion, we will look into. Thanks.

pokerfacett, 2020-04-01:
> @pokerfacett Thanks for reporting and really appreciate you bring fix suggestion, we will look into. Thanks.

Any progress？

Campou, 2020-04-02:
Hi @pokerfacett 

In these places, the data/segment will be copied into the pre-initialized buffer, which is large enough to hold the whole pdu.

Thanks.

closed 2020-04-08 by pokerfacett