espressif/esp-idf #5046: driver/spi_bus_lock.c:820:41: error: implicit declaration of function 'xSemaphoreCreateBinaryStatic'

keelung-yang, 2020-04-01:
On latest master c9f29e0b59988779bf0792e2e954c3448fd37e3b

```shell
[1/486] cmd.exe /C "cd /D C:\dev\esp\bootloader\esp32\build\esp-idf\partition_table && C:\Users\jyang37\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E echo "Partition table binary generated. Contents:" && C:\Users\jyang37\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E echo ******************************************************************************* && C:\Users\jyang37\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:/dev/esp/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB C:/dev/esp/bootloader/esp32/build/partition_table/partition-table.bin && C:\Users\jyang37\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E echo *******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/486] Performing build step for 'bootloader'
ninja: no work to do.
[3/484] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj 
ccache C:\Users\jyang37\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/dev/esp/esp-idf/components/driver/include -IC:/dev/esp/esp-idf/components/driver/esp32/include -IC:/dev/esp/esp-idf/components/driver/include/driver -IC:/dev/esp/esp-idf/components/newlib/platform_include -IC:/dev/esp/esp-idf/components/freertos/include -IC:/dev/esp/esp-idf/components/freertos/xtensa/include -IC:/dev/esp/esp-idf/components/heap/include -IC:/dev/esp/esp-idf/components/log/include -IC:/dev/esp/esp-idf/components/lwip/include/apps -IC:/dev/esp/esp-idf/components/lwip/include/apps/sntp -IC:/dev/esp/esp-idf/components/lwip/lwip/src/include -IC:/dev/esp/esp-idf/components/lwip/port/esp32/include -IC:/dev/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/dev/esp/esp-idf/components/soc/src/esp32/. -IC:/dev/esp/esp-idf/components/soc/src/esp32/include -IC:/dev/esp/esp-idf/components/soc/include -IC:/dev/esp/esp-idf/components/esp_rom/include -IC:/dev/esp/esp-idf/components/esp_common/include -IC:/dev/esp/esp-idf/components/esp_system/include -IC:/dev/esp/esp-idf/components/xtensa/include -IC:/dev/esp/esp-idf/components/xtensa/esp32/include -IC:/dev/esp/esp-idf/components/esp32/include -IC:/dev/esp/esp-idf/components/esp_ringbuf/include -IC:/dev/esp/esp-idf/components/efuse/include -IC:/dev/esp/esp-idf/components/efuse/esp32/include -IC:/dev/esp/esp-idf/components/espcoredump/include -IC:/dev/esp/esp-idf/components/esp_timer/include -IC:/dev/esp/esp-idf/components/soc/soc/esp32/. -IC:/dev/esp/esp-idf/components/soc/soc/esp32/include -IC:/dev/esp/esp-idf/components/soc/soc/esp32/../include -IC:/dev/esp/esp-idf/components/vfs/include -IC:/dev/esp/esp-idf/components/esp_wifi/include -IC:/dev/esp/esp-idf/components/esp_wifi/esp32/include -IC:/dev/esp/esp-idf/components/esp_event/include -IC:/dev/esp/esp-idf/components/esp_netif/include -IC:/dev/esp/esp-idf/components/esp_eth/include -IC:/dev/esp/esp-idf/components/tcpip_adapter/include -IC:/dev/esp/esp-idf/components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-994-gc9f29e0b5\" -DESP_PLATFORM -MD -MT esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj -MF esp-idf\driver\CMakeFiles\__idf_driver.dir\spi_bus_lock.c.obj.d -o esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj   -c C:/dev/esp/esp-idf/components/driver/spi_bus_lock.c
C:/dev/esp/esp-idf/components/driver/spi_bus_lock.c: In function 'spi_bus_lock_init_main_dev':
C:/dev/esp/esp-idf/components/driver/spi_bus_lock.c:820:41: error: implicit declaration of function 'xSemaphoreCreateBinaryStatic'; did you mean 'xSemaphoreCreateBinary'? [-Werror=implicit-function-declaration]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                         xSemaphoreCreateBinary
C:/dev/esp/esp-idf/components/driver/spi_bus_lock.c:820:39: warning: assignment to 'SemaphoreHandle_t' {aka 'void *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                       ^
cc1.exe: some warnings being treated as errors
[4/484] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/esp32/touch_sensor.c.obj
[5/484] Building C object esp-idf/pthread/CMakeFiles/__idf_pthread.dir/pthread.c.obj
[6/484] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/timer.c.obj
[7/484] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/uart.c.obj
[8/484] Building C object esp-idf/pthread/CMakeFiles/__idf_pthread.dir/pthread_local_storage.c.obj
ninja: build stopped: subcommand failed.
Executing action: flash
Running ninja in directory c:\dev\esp\bootloader\esp32\build
Executing "ninja flash"...
ninja failed with exit code 1

```

suda-morris, 2020-04-01:
cleanup your build (especially remove the old sdkconfig) and rebuild again.

keelung-yang, 2020-04-01:
@suda-morris 
I've deleted build directory and sdkconfig, and then use _idf menuconfig_ to generate sdkconfig (without any modifying, got  same file context reported by BeyondComparer).

but still error:
```shell
[454/936] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj
```
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4412518/sdkconfig.txt)

keelung-yang, 2020-04-01:
It's a bug:
In components\freertos\include\freertos\semphr.h
```c
#if( configSUPPORT_STATIC_ALLOCATION == 1 )
	#define xSemaphoreCreateBinaryStatic( pxStaticSemaphore ) xQueueGenericCreateStatic( ( UBaseType_t ) 1, semSEMAPHORE_QUEUE_ITEM_LENGTH, NULL, pxStaticSemaphore, queueQUEUE_TYPE_BINARY_SEMAPHORE )
#endif /* configSUPPORT_STATIC_ALLOCATION */
```

In components\driver\spi_bus_lock.c
```c

esp_err_t spi_bus_lock_init_main_dev(void)
{
    spi_bus_main_set_lock(g_main_spi_bus_lock);
    g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
    if (g_spi_lock_main_flash_dev->semphr == NULL) {
        return ESP_ERR_NO_MEM;
    }

    return ESP_OK;
}
```

projectgus, 2020-04-01:
Hi @keelung-yang ,

Yes, this is a bug. Sorry for the inconvenience.

The workaround is to enable "Enable FreeRTOS static allocation API" in menuconfig:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-support-static-allocation

This increases the binary size, but only by a small amount (+48 bytes in most cases).

A proper fix will be merged soon.

AxelLin, 2020-04-06:
> Hi @keelung-yang ,
> 
> Yes, this is a bug. Sorry for the inconvenience.
Any chance to push an urgent fix ASAP?
Several examples fail to build due to this issue now.

slog, 2020-04-18:
A simple solution can be to replace your esp-idf folder with previous stable version.
isn't it @igrr   
Ofcourse I got error & resolved with installing previous version & will wait till next stable version 
:+1: (Testing emoji)

mwiora, 2020-04-26:
solved with current master by deleting the sdkconfig - thanks

AxelLin, 2020-04-27:
> solved with current master by deleting the sdkconfig - thanks

Are you sure it's solved with current master?
I still got the build error when CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION is not set.

mwiora, 2020-04-27:
clone currrent master
```
rm sdkconfig
idf.py menuconfig
idf.py build
```
that's the way worked for me, without changing anything else

AxelLin, 2020-04-28:
> clone currrent master
> 
> ```
> rm sdkconfig
> idf.py menuconfig
> idf.py build
> ```
> 
> that's the way worked for me, without changing anything else

That is because CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION will be set by default if you re-generate sdkconfig.
The problem is still there if you disable CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION manually.

closed 2020-04-28 by espressif-bot