espressif/esp-idf #5047: sta的mac地址是蓝牙的mac地址-2的问题

fzboffice, 2020-04-02:
sta的mac地址是蓝牙的mac地址-2,那么会不会有进位的问题,打个比方 假如蓝牙mac地址是aa:aa:aa:aa:aa:01 那么sta的mac地址是aa:aa:aa:aa:a9:ff还是aa:aa:aa:aa:aa:ff？

MaxwellAlan, 2020-04-07:
@fzboffice ，您好！
蓝牙的mac地址根据不同的配置，有两种计算方式：
１）配置：4 universally administered：sta base mac addr + 2
２）配置：2 universally administered：sta base mac addr + 1
芯片在出厂时会在efuse中烧写mac地址，即sta base mac，为了保证mac地址的唯一性，因此烧写时会按照每隔４个mac地址进行烧写，即chip 1 mac: xx:xx:xx:00:00:00, chip 2:xx:xx:xx:00:00:04,chip 3: xx:xx:xx:00:00:08,chip 4: xx:xx:xx:00:00:0c, ... 
这样的烧写方式保证了不会出现您所担心的问题，即蓝牙的mac地址应该以02,06,0A,0E为最后两位。
参考链接：https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html?highlight=mac#mac-address
谢谢！

fzboffice, 2020-04-08:
> @fzboffice ，您好！
> 蓝牙的mac地址根据不同的配置，有两种计算方式：
> １）配置：4 universally administered：sta base mac addr + 2
> ２）配置：2 universally administered：sta base mac addr + 1
> 芯片在出厂时会在efuse中烧写mac地址，即sta base mac，为了保证mac地址的唯一性，因此烧写时会按照每隔４个mac地址进行烧写，即chip 1 mac: xx:xx:xx:00:00:00, chip 2:xx:xx:xx:00:00:04,chip 3: xx:xx:xx:00:00:08,chip 4: xx:xx:xx:00:00:0c, ...
> 这样的烧写方式保证了不会出现您所担心的问题，即当采用配置１）时，蓝牙的mac地址应该以02,06,0A,0E为最后两位，而当采用配置２）时，也不会出现进位相关的问题。
> 参考链接：https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html?highlight=mac#mac-address
> 谢谢！

感谢解释

closed 2020-04-08 by fzboffice