espressif/esp-idf #5048: BluFi handle GATT packet has buffer overflow

pokerfacett, 2020-04-02:
INSTRUCTIONS
============
In the process of BluFi's network configurationg through ble, there is an overflow when processing the write ATT command at characteristic = 0xFF01. The specific problem is in the btc_blufi_recv_handler function of blufi_prf.c. First ,parse the incoming ATT data data [0] and data [1] as the length of memery which apply from heap. And then copy the memory data starting from data [2] to the applyed memory, causing a buffer overflow

DETAILS
============
[image]

Here, if the fc parameter is 0x10, the overflow part can be entered. Total_len is parsed from data. Here, control data [0] = 1, data [1] = 0, so hdr-> data [0] | (((uint16_t) hdr-> data [1]) << 8) == 1, that is, total_len = 1, that is, aggr_buf allocates 1 byte of heap memory, and then copies aggr_buf, which can eventually overflow N bytes

An attacker could construct GATT payload to cause crash or remote code execution

The GATT payload as follows:
[image]

FIX SUGGESTIONG
============
```
if(blufi_env.offset + hdr->data_len  - 2 <= blufi_env.total_len){
    memcpy(blufi_env.aggr_buf + blufi_env.offset, hdr->data + 2, hdr->data_len  - 2)
}
```

pokerfacett, 2020-04-15:
any progress？

pokerfacett, 2020-05-21:
any progress?

WCCWCC, 2020-05-22:
Hi @pokerfacett ,
     Thank you for your suggestion, we will fix it as soon as possible.

pokerfacett, 2020-05-22:
Hi @WCCWCC ,
Thank you for relay ,if you fix it please put commit link here and i will close this issue

WCCWCC, 2020-06-08:
Hi @pokerfacett ，

I have submitted the MR and it will be modified as follows.

[image]

pokerfacett, 2020-06-09:
@WCCWCC hi，First of all, thank you for submitting the repair. Secondly, considering that ESPIDF has thousands of users and is mounted on tens of millions of devices, the vulnerability will remotely cause code execution and denial of service attacks on the device.In order to maintain the security of the entire ecology for development, we recommend that ESPIDF notify developers through github's security announcement to fix the problem on their own version to achieve risk avoidance and reduction. Specific reference: https://help.github.com/cn/github/managing-security -vulnerabilities/publishing-a-security-advisory, and apply for a CVE number for this to remind developers again of its importance and complete the repair as soon as possible

closed 2021-01-09 by pokerfacett

Leadrive, 2023-06-26:
````
[0;31mE (32342) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32402) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32462) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32522) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32582) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32642) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32702) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32762) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32852) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32912) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (32972) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (33062) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (33122) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (33182) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (33242) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (33302) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (33362) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
[0m
[0;31mE (33422) BT_BTC[0;32mI (32314) [handle_message, 154]
````
出现这个错误，如何定位？