espressif/esp-idf #5050: Poor uart performance

keelung-yang, 2020-04-02:
One of our products depends on Modbus-RTU heavenly. But after double uart baudrate, I only got about +12% performance totally. So I decided to benchmark uart performance by official uart echo example on master branch.

Here are my test results:
blocksize | baudrate setting | real speed
------------ | ------------- | -------------
1000 | 115200 * 10 | ~28% 
1000 | 115200 * 5 | ~40% 
1000 | 115200 * 2 | ~58% 
1000 | 115200 * 1 | ~69% 
100 | 115200 * 10 | ~4.6% 
100 | 115200 * 5 | ~9% 
100 | 115200 * 2 | ~17.5% 
100 | 115200 * 1 | ~28% 

And test script in Python 3.8+
```python3
import time

import serial

port = 'COM2'
baudrate = 115_200 * 1
timeout = 10

# Require pyserial version > 3.4, on master now
#    https://github.com/pyserial/pyserial/pull/371
with serial.Serial() as ser:
    ser.port = port
    ser.baudrate = baudrate
    ser.bytesize = serial.EIGHTBITS
    ser.parity = serial.PARITY_NONE
    ser.stopbits = serial.STOPBITS_ONE
    ser.xonxoff = False
    ser.rtscts = False
    ser.dsrdtr = False
    ser.timeout = timeout
    ser.write_timeout = timeout

    ser.open()

    i = 0
    blocksize = 100 # max 1024 in uart_echo example
    data = b'\x00' * blocksize
    seconds = []
    while True:
        i += 1
        begin = time.time()
        if ser.write(data) == len(data):
            if ser.read(len(data)) == data:
                end = time.time()
                seconds.append(end - begin)
                bps = 2 * 8 * len(data) * len(seconds) / sum(seconds)
                upto = bps / baudrate
                if i % 10 == 0:
                    print('\t'.join([
                        f'{i}',
                        f'{blocksize=}',
                        f'{baudrate=}',
                        f'{bps=:.2f}',
                        f'{upto=:.2%}',
                    ]))
                continue
        print(f'{i}\tError')
```

Output:
```shell
10	blocksize=100	baudrate=115200	bps=32514.36	upto=28.22%
20	blocksize=100	baudrate=115200	bps=32168.75	upto=27.92%
30	blocksize=100	baudrate=115200	bps=32027.35	upto=27.80%
40	blocksize=100	baudrate=115200	bps=32030.29	upto=27.80%
50	blocksize=100	baudrate=115200	bps=32013.12	upto=27.79%
60	blocksize=100	baudrate=115200	bps=32188.51	upto=27.94%
70	blocksize=100	baudrate=115200	bps=32174.41	upto=27.93%
80	blocksize=100	baudrate=115200	bps=32174.51	upto=27.93%
90	blocksize=100	baudrate=115200	bps=32271.55	upto=28.01%
100	blocksize=100	baudrate=115200	bps=32285.19	upto=28.03%
```

koobest, 2020-04-13:
Hi. @keelung-yang 
What is the difference between No 10~No100? I see that the baud rate is the same.

thanks !!

keelung-yang, 2020-04-13:
@koobest I test on a fixed baudrate on each compiling, since baudrate is hardcoded in uart_echo example and for keeping test code simple.

BTW, I've updated test script and results before or after your reading. 
Sorry If it's confused.  I'll only update it by replying from now on.

keelung-yang, 2020-04-15:
I've test read/write separately, on UART pins (Tx 18, Rx 23).

Read performance:
```shell
COM2    2020-04-15 18:29:19.997143
No.     BlockSize   Baudrate    RealSpeed   Percent     KB/s
10      100         115200      16314.43    14.16%      2.04
20      100         115200      16243.04    14.10%      2.03
30      100         115200      16205.99    14.07%      2.03
40      100         115200      16201.12    14.06%      2.03
50      100         115200      16199.05    14.06%      2.02
60      100         115200      16208.02    14.07%      2.03
70      100         115200      16209.13    14.07%      2.03
80      100         115200      16206.61    14.07%      2.03
90      100         115200      16203.96    14.07%      2.03
100     100         115200      16195.91    14.06%      2.02
```

Write performance:
```shell
COM2    2020-04-15 18:27:37.614797
No.     BlockSize   Baudrate    RealSpeed   Percent     KB/s
10      100         115200      103720.88   90.04%      12.97
20      100         115200      98401.10    85.42%      12.30
30      100         115200      96315.70    83.61%      12.04
40      100         115200      95435.30    82.84%      11.93
50      100         115200      94769.40    82.27%      11.85
60      100         115200      94453.37    81.99%      11.81
70      100         115200      94268.66    81.83%      11.78
80      100         115200      94078.62    81.67%      11.76
90      100         115200      93961.38    81.56%      11.75
100     100         115200      93823.83    81.44%      11.73
```

keelung-yang, 2020-04-15:
I've tested read/write separately, on UART pins (Tx 18, Rx 23).

Read performance:
```shell
COM2    2020-04-15 18:29:19.997143
No.     BlockSize   Baudrate    RealSpeed   Percent     KB/s
10      100         115200      16314.43    14.16%      2.04
20      100         115200      16243.04    14.10%      2.03
30      100         115200      16205.99    14.07%      2.03
40      100         115200      16201.12    14.06%      2.03
50      100         115200      16199.05    14.06%      2.02
60      100         115200      16208.02    14.07%      2.03
70      100         115200      16209.13    14.07%      2.03
80      100         115200      16206.61    14.07%      2.03
90      100         115200      16203.96    14.07%      2.03
100     100         115200      16195.91    14.06%      2.02
```

Write performance:
```shell
COM2    2020-04-15 18:27:37.614797
No.     BlockSize   Baudrate    RealSpeed   Percent     KB/s
10      100         115200      103720.88   90.04%      12.97
20      100         115200      98401.10    85.42%      12.30
30      100         115200      96315.70    83.61%      12.04
40      100         115200      95435.30    82.84%      11.93
50      100         115200      94769.40    82.27%      11.85
60      100         115200      94453.37    81.99%      11.81
70      100         115200      94268.66    81.83%      11.78
80      100         115200      94078.62    81.67%      11.76
90      100         115200      93961.38    81.56%      11.75
100     100         115200      93823.83    81.44%      11.73
```

koobest, 2020-04-15:
Hi,
Are you testing with uart_echo example, If I need to test, do I need to modify anything?

keelung-yang, 2020-04-15:
> Hi,
> Are you testing with uart_echo example, If I need to test, do I need to modify anything?

Yes. For reading:
[image]

For writing:
[image]

Sorry for updated test script again.

koobest, 2020-05-25:
Hi, @keelung-yang 
The performance of this UART test is not completely correct, from the pic, we can see:

[image]
 
The time interval between sending and receiving is very long.  This time interval comes from `uart_read_bytes` and PC. This is why you can get good performance in only reading tests.

Alvin1Zhang, 2020-06-02:
@keelung-yang Thanks for reporting, feel free to reopen with more details/updates. Thanks.

closed 2020-06-02 by Alvin1Zhang