espressif/esp-idf #5051: idf menuconfig not showing serial ports

luis-signalsecure, 2020-04-02:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v2.0.0.0_esp32-145-g408f0b3
- Build System:         [idf.py]
- Compiler version: 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

idf.py menuconfig -> serial flasher config not showing port serial selector, baud rate or compress

Using esp-at code obtained from github, also tested with hello_world code

make menuconfig is working as it should

### Expected Behavior
idf.py menuconfig -> serial flasher config shows the expected menu

### Actual Behavior
idf.py menuconfig -> serial flasher config not showing the first 3 options

### Steps to reproduce
install esp-idf
clone https://github.com/espressif/esp-at/
go to esp-at folder
export ESP_AT_PROJECT_PATH=$(pwd)
run idf.py menuconfig
go to serial flasher config
check serial port options not being there

### Code to reproduce this issue

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4422632/sdkconfig.txt)

igrr, 2020-04-02:
Hi @luis-signalsecure, in CMake build system (idf.py) serial port is no longer set in sdkconfig. This change was made mainly to avoid storing this computer-specific value inside sdkconfig file, when the project is shared by multiple developers and sdkconfig is checked into a git repository.

You can either pass the port using a `-p` argument of `idf.py flash`, or (to avoid typing it every time) set the port in the environment variable ESPPORT:

```
export ESPPORT=/dev/ttyUSB1
```

You may also automate the latter using tools such as direnv.

closed 2020-04-02 by igrr

sergioamr, 2020-11-29:
How do you make this thing work from VS Studio code?

ESPPORT on the environment doesn't seem to do anything. It gets the COM1 by default and I keep digging into the toolchain trying to find how to add the correct port.

I can flash by running manually the command.

`
python.exe esptool\esptool.py -p COM12 -b 115200 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
`

But the IDE option keeps running COM1.

Adding CONFIG_ESPTOOLPY_PORT="COM12" to the sdkconfig seems to be ignored too.