espressif/esp-idf #5053: Adding a Custom Partition Table in a Custom CMake Project

iqrohan, 2020-04-03:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (DevKitC): [v4]
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.2-dev-910-g4fe04f1
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I am trying to build a Custom CMake Project by following the documentation provided under `API Guides` -> `Build System` -> `Using ESP-IDF in Custom CMake Projects`. I am also following the example `idf_as_lib`

My requirement is to add a custom partition table using my `partitions.csv` file. To do this I have made a `sdkconfig` file by running the `menuconfig` tool in the `hello-world` example project. I was able to build the `hello-world` example project with the custom `partitions.csv` successfully.

Now I have copied the `sdkconfig` and `partitions.csv` files in the example `idf_as_lib`. I have edited the `CMakeLists.txt` file and added `SDKCONFIG_DEFAULTS` as below:

```
    idf_build_process(esp32
                    # try and trim the build; additional components
                    # will be included as needed based on dependency tree
                    #
                    # although esptool_py does not generate static library,
                    # processing the component is needed for flashing related
                    # targets and file generation
                    COMPONENTS esp32 freertos esptool_py
		    SDKCONFIG_DEFAULTS ${CMAKE_SOURCE_DIR}/sdkconfig
                    SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig
                    BUILD_DIR ${CMAKE_BINARY_DIR})
```
**When I compile the code I get the following error:**
```
ninja: error: 'esp-idf/partitions.csv', needed by 'partition_table/partition-table.bin', missing and no known rule to make it
```
**Note: If I do not use a custom partition table, the program compiles perfectly.**

### Expected Behavior

The compilation should be successful with a custom partition table.

When I use the esp-idf provided partition table `partitions_two_ota_coredump` the compilation is successful, so I assume there should not be any problem with custom partition table also.

### Actual Behavior

I get compilation error:

### Steps to reproduce
1. Generate a `sdkconfig` file from the `hello_world` example menuconfig which has a custom partition. My `sdkconfig` and `partitions.csv` files are attached
2. Copy these files into the example `idf_as_lib`
3. Edit the `idf_as_lib CMakeLists.txt` file to add the following line.

```
SDKCONFIG_DEFAULTS ${CMAKE_SOURCE_DIR}/sdkconfig
```
My `CMakeLists.txt` file is attached

4. Build the project by running the script `build-esp32.sh`

## Debug Logs

```
-- The C compiler identification is GNU 8.2.0
-- Check for working C compiler: /home/test1/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/test1/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- Checking Python dependencies...
Python requirements from /home/test1/esp/esp-idf/requirements.txt are satisfied.
Loading defaults file /home/test1/esp/esp-idf/examples/build_system/cmake/idf_as_lib/sdkconfig...
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/test1/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working CXX compiler: /home/test1/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/test1/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /home/test1/.espressif/python_env/idf4.2_py2.7_env/bin/python (found version "2.7.12") 
-- Found Perl: /usr/bin/perl (found version "5.22.1") 
-- App "" version: 
-- Adding linker script /home/test1/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs-time.ld
-- Adding linker script /home/test1/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/test1/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/test1/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/test1/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/test1/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/test1/esp/esp-idf/examples/build_system/cmake/idf_as_lib/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/test1/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/test1/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Configuring done
-- Generating done
-- Build files have been written to: /home/test1/esp/esp-idf/examples/build_system/cmake/idf_as_lib/build
ninja: error: 'esp-idf/partitions.csv', needed by 'partition_table/partition-table.bin', missing and no known rule to make it
```
## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

[idf_as_lib-res.zip](https://github.com/espressif/esp-idf/files/4428342/idf_as_lib-res.zip)

renzbagaporo, 2020-04-28:
Hi @iqrohan, sorry for the late reply. This was diagnosed to be a bug in `idf_build_process`. A fix has submitted internally. In the meantime, is it possible to use the existing partition table csv to contain the partition table you want?

iqrohan, 2020-04-28:
Hi @renzbagaporo, as of now I have edited one of the csv files inside the esp idf to fit my requirement. But this feature would be needed in the future.

Thanks

renzbagaporo, 2020-04-29:
Good to know that you have found a workaround. This issue will be closed once the fix has been submitted.

closed 2020-05-22 by espressif-bot

galah92, 2020-07-07:
I also have the same use-case. Can the fix be cherry-picked to the upcoming v4.1?

Alvin1Zhang, 2020-07-08:
@galah92  Thanks for reporting, the fix is being back ported to v4.1, but will take some time to appear on GitHub.

galah92, 2020-07-19:
@Alvin1Zhang Thanks! Is there an ETA for that? is it days/weeks/months?

galah92, 2020-09-23:
@Alvin1Zhang I see this hasn't been backported to v4.1 (nor it can even be found on v4.2-dev). Can something be done regarding this? I rely heavily on this feature in order to being able to build multiple images under one CMake invocation.

projectgus, 2020-09-28:
@galah92 Unfortunately the backport missed the v4.1 final release, but commit d101da4ca43f44716d91c8c09e891ff97a7282f0 is part of the release/v4.1 branch and will be included in the v4.1.1 release. You could consider [checking out the release branch](https://docs.espressif.com/projects/esp-idf/en/stable/versions.html#updating-to-a-release-branch) until then.

The v4.2 backport is unfortunately still pending, due to our internal testing requirements branches are frozen and unfrozen periodically.

renzbagaporo, 2020-09-28:
Hi, I've rechecked following @iqrohan's steps to reproduce and it now works with the fix on master (and on v4.1, v4.0 to which the fixes have been backported to). This might be a different use case @galah92?

galah92, 2020-10-01:
@projectgus I'll probably wait for v4.2 / v4.11, thanks
@renzbagaporo I've to get the exact same use case and is able to reproduce it. I'm afraid the fix @projectgus attached (both commits) is not part of v4.1.