espressif/esp-idf #5055: [BLE_MESH] How to create sensor network using external MCU as the sensor input?

dhorninger, 2020-04-03:
**Environment**

- 4 x ESP32-WROVER-B 
- 4 x STM32F051R8T6

**Issue**
I am unsure how to send data to the ESP32-WROVER-B using SPI so that it can be transmitted through a BLE Mesh Network. I am specifically looking for how to modify the input of each node to receive SPI data and communicate that data through the network.

**Desired Solution**

I would like to set up a BLE Mesh network using the ESP32-WROVER-B as an SPI slave. I am using an STM32F051R8T6 to receive sensor data and send it to the ESP32 using SPI. From there, the ESP32 must send the data to an end node to display the sensor data.

sheldon123z, 2020-04-18:
Hi @dhorninger ,

Have you built a sensor network already? I am basically doing exactly things as you do, but I am stuck at creating sensor models using ESP32. Could you please share how to create sensor models?

Best Regards

Campou, 2020-04-21:
Hi @sheldon123z ,

I have submitted a MR for sensor server/setup server model examples internally, waiting to be merged. When the MR is merged, it will be synchronized to Github soon.

Thanks for your patience

dhorninger, 2020-04-22:
Hi @Campou 
When do you expect the update to Github will appear?

Campou, 2020-07-27:
Hi @sheldon123z @dhorninger 

Please have a try with the latest idf, I have added sensor server related examples.

Thanks.

Alvin1Zhang, 2020-08-04:
Thanks for reporting, will close now, feel free to reopen with more updates. Thanks.

closed 2020-08-04 by Alvin1Zhang