espressif/esp-idf #5056: Guru Meditation Error in spi_bus_lock.c with CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y

no1seman, 2020-04-03:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-D0WD
- IDF version: v4.2-dev-1034-ge599b794b
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Power Supply:  USB|external 5V&Battery
## Problem Description
Guru Meditation Error with Default log verbosity == Debug

### Expected Behavior
No error
### Actual Behavior
Error
### Steps to reproduce

1. set CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y 
2. build && flash && monitor

### Code to reproduce this issue

## Debug Logs

```
I (591) heap_init: Initializing. RAM available for dynamic allocation:
D (598) heap_init: New heap initialised at 0x3ffae6e0
I (603) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (609) heap_init: New heap initialised at 0x3ffb3c10
I (614) heap_init: At 3FFB3C10 len 0002C3F0 (176 KiB): DRAM
I (621) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (627) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (633) heap_init: New heap initialised at 0x4008bedc
I (639) heap_init: At 4008BEDC len 00014124 (80 KiB): IRAM
I (645) cpu_start: Pro cpu start user code
D (657) clk: RTC_SLOW_CLK calibration value: 3357210
D (666) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (667) intr_alloc: Connected src 17 to int 3 (cpu 0)
D (668) intr_alloc: Connected src 24 to int 9 (cpu 0)
D (672) FLASH_HAL: extra_dummy: 1
Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40084564  PS      : 0x00060b30  A0      : 0x80083339  A1      : 0x3ffe3aa0
0x40084564: dev_lock_get_id at /home/user/esp/esp-idf/components/driver/spi_bus_lock.c:639
 (inlined by) spi_bus_lock_touch at /home/user/esp/esp-idf/components/driver/spi_bus_lock.c:659

A2      : 0x3ffb148c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0xb0000070
A6      : 0x00e0a000  A7      : 0xb0000000  A8      : 0x80084564  A9      : 0x3ffe3a80
A10     : 0x000002a3  A11     : 0xffffffff  A12     : 0x3ffb144c  A13     : 0x00060123
A14     : 0x00000001  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace:0x40084561:0x3ffe3aa0 0x40083336:0x3ffe3ac0 0x40084a85:0x3ffe3ae0 0x40083111:0x3ffe3b00 0x400831dd:0x3ffe3b30 0x400d6aed:0x3ffe3b60 0x400814c1:0x3ffe3ba0 0x400817a5:0x3ffe3bf0 0x400794db:0x3ffe3c30 |<-CORRUPTED
0x40084561: spi_bus_lock_touch at /home/user/esp/esp-idf/components/driver/spi_bus_lock.c:659

0x40083336: spi_start at /home/user/esp/esp-idf/components/spi_flash/spi_flash_os_func_app.c:70

0x40084a85: spiflash_start at /home/user/esp/esp-idf/components/spi_flash/esp_flash_api.c:75

0x40083111: esp_flash_read_chip_id at /home/user/esp/esp-idf/components/spi_flash/esp_flash_api.c:168

0x400831dd: esp_flash_init at /home/user/esp/esp-idf/components/spi_flash/esp_flash_api.c:123 (discriminator 2)

0x400d6aed: esp_flash_init_default_chip at /home/user/esp/esp-idf/components/spi_flash/esp_flash_spi_init.c:211

0x400814c1: start_cpu0_default at /home/user/esp/esp-idf/components/esp32/cpu_start.c:433

0x400817a5: call_start_cpu0 at /home/user/esp/esp-idf/components/esp32/cpu_start.c:265
```

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/4428587/sdkconfig.zip)

Alvin1Zhang, 2020-04-07:
@no1seman Thanks for reporting, we will look into.

ginkgm, 2020-04-08:
This is fixed by commit on latest master.

a0e66fef0880dba2caf93e9cb9c65cff7dc9b880 spi: fix the memory accessed while cache disabled issue in the bus lock when log level is verbose 

You may need to wait a few hours before github is sync-ed.

yorickvP, 2020-04-09:
It seems that this commit still isn't in master.

no1seman, 2020-04-09:
@ginkgm 
well done, Guru Mediation Error fixed, but on v4.2-dev-1097-g2e14149bf got the following log with 'Debug' log verbosity:
[log.zip](https://github.com/espressif/esp-idf/files/4456067/log.zip)

Why do I got 'D bus_lock: SPI dev changed from 0 to 0' on every SPI transaction? Seems there's some extralogging....

ginkgm, 2020-04-10:
@no1seman Because you are working under a debug log level, and that's the debug log for the bus lock. And unfortunately for that kind of message, it's not able to be disabled by `esp_log_level_set`...

For now you have to either:
1. add a `#define LOG_LOCAL_LEVEL ESP_LOG_INFO` in the spi_bus_lock.c, right before all the header files. 
2. select a higher global log level in the menuconfig..
3. Modify the driver, find the log line and make it verbose/comment it.

Sorry for your inconvenience.

no1seman, 2020-04-10:
@ginkgm,
if you will try earlier versions of master branch in 'debug' log verbosity you will see that there was no such overlogging on debug log verbosity, for example look here: https://github.com/espressif/esp-idf/issues/4747 
Seems that it would be better to make some changes in idf and more accurately divide messages that would be logged in 'debug' and 'verbose' log verbosities.

ginkgm, 2020-05-06:
Hi @no1seman , 

This is supposed to be fixed in commit: a0e66fef0880dba2caf93e9cb9c65cff7dc9b880. Could you have a try?

no1seman, 2020-05-06:
@ginkgm,
as I have already wrote main problem with Guru mediation error successfully solved, but seems to me that it will be more convenient to change verbosity of log output like "D bus_lock: SPI dev changed from 0 to 0" from 'debug' level to 'verbosity'.

closed 2020-05-15 by espressif-bot