espressif/esp-idf #5057: Unable to get data through NOTIFY_EVT in "gatt_client" example

ROYMURLI, 2020-04-03:
ESP_IDF = v3.1.5
OS- Ubuntu

I am using "gatt_client" example to connect to a device(server) and then receive data through it. 
(Server operation is to send data cotinuosly after the BLE connection)

I changed service and notify UUID according to the server and also name. The ESP32 and device is successfully connected but ESP is unable to recieve data.

Through logs, I can figure it out that the example not able call esp_ble_gattc_register_for_notify_event() function because "get_server" variable. "get_server = true" condition is in SEARCH_CMPL_EVT which is not called after connection. May be I am giving wrong UUID. Kindly check the UUID part because I used REMOTE_SERVICE_UUID as ESP_UUID_LED_16 and REMOTE_NOTIFY_CHAR_UUID as  ESP_UUID_LED_128.

I am attaching my code, ble scanner screenchot (nrF) and debug logs here.

[gattc_issue_code.zip](https://github.com/espressif/esp-idf/files/4428634/gattc_issue_code.zip)

Alvin1Zhang, 2020-04-07:
@ROYMURLI Thanks for reporting, we will look into.

younglee123, 2020-06-16:
You have entered the wrong service UUID and characteristic UUID,so you can't find the CUSTOM SERVICE on your phone.Here is my code,you can refer to it.
ps:Note that the CPU data is stored in little-endian storage mode.[https://blog.csdn.net/favory/article/details/4441361](url)
[gattc_demo.zip](https://github.com/espressif/esp-idf/files/4784224/gattc_demo.zip)

closed 2020-07-10 by espressif-bot