espressif/esp-idf #5058: SPI SLAVE assert error

chuledeco, 2020-04-03:
## Environment

- Development Kit:      ESP32-PICO-D4 in a custom board
- Module or chip used:  ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-318-g605da33c3
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp-2019r2-8.2.0 
- Operating System:     Windows
- Using an IDE?: Yes Eclipse
- Power Supply:     external 3.3V

## Problem Description

I found a problem with the SPI SLAVE driver. It seems that after:
ret=spi_slave_transmit(RCV_HOST, &t_slave, portMAX_DELAY);
the driver do not check if the CS line is idle before to check the assertion.
Example:
```cpp

//Configuration for the VSPI bus
    spi_bus_config_t buscfg2={
        .mosi_io_num=DATA_MOSI,
        .miso_io_num=DATA_MISO,
        .sclk_io_num=DATA_SCLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
		.max_transfer_sz=128
    };
    //Configuration for the SPI slave interface
    spi_slave_interface_config_t slvcfg={
        .mode=SPI_MODE,
        .spics_io_num=DATA_CS,
        .queue_size=3,
        .flags=0,
    };
    // Set apropiated pull-ups
    gpio_set_pull_mode(DATA_MOSI, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(DATA_SCLK, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(DATA_CS, GPIO_PULLUP_ONLY);
    //Initialize SPI slave interface
    ret=spi_slave_initialize(RCV_HOST, &buscfg2, &slvcfg, DMA_CHAN);
    assert(ret==ESP_OK);

WORD_ALIGNED_ATTR char recvbuf[20]="";
memset(recvbuf, 0, 20);
spi_slave_transaction_t t_slave;
memset(&t_slave, 0, sizeof(t_slave));
while (true) {
memset(recvbuf, 0, 20);
    	t_slave.length=128;
    	t_slave.tx_buffer=NULL;
    	t_slave.rx_buffer=recvbuf;
    	memset(recvbuf, 0xA5, 20);

    	ret=spi_slave_transmit(RCV_HOST, &t_slave, portMAX_DELAY);
	printf("rec: %s\n", recvbuf);
vTaskDelay(1000 / portTICK_PERIOD_MS);
}

```

In this typcal example if I hold DATA_CS down permanentlly, the output in the terminal is a a received message each one second. The driver seems to assert and start the transaction when CS is low with out checkig it was high before.

### Expected Behavior
I expected to the SPI wait until CS line goes from high to low to start the transaction. So if CS is hold low, the transaction never start and no output in the terminal.
### Actual Behavior
With CS low, the SPI start the transaction after “spi_slave_transmit(RCV_HOST, &t_slave, portMAX_DELAY);” so if there is something in the bus, the devide start recording in the recvbuf in the midle.I get a message in the terminal each 1 second
### Steps to reproduce

1. Copy the example above
2. Flash the firmware
3. Hold CS line down 
4. See terminal
### Code to reproduce this issue

<script src="https://gist.github.com/chuledeco/2848ad2cb1e06bae32dd60ea7d8d5b34.js"></script>

Alvin1Zhang, 2020-04-07:
@chuledeco Thanks for the detailed report, we will look into.

chuledeco, 2020-04-08:
Thanks @Alvin1Zhang 

I hope you can help me.

chuledeco, 2020-04-20:
Any update?
I'm still interested in getting this working

ginkgm, 2020-04-30:
Hi @chuledeco ,

It seems that our example is not written well. Suggest try either:

1. Change the timeout to be portMAX_DELAY, which will block the slave's task until transaction is actually done;
2. Use `queue_trans` and `get_trans_result` instead, and check the return value of them. The ESP_OK from `queue_trans` means a transaction is queued to send to master; the ESP_OK from `get_trans_result` means the transaction has completed. If ESP_ERR_TIMEOUT happen, means the call failed, and you should try again.

What's happening:

Actually the slave HW will not terminate the transaction until the CS is inactive, upon which the driver should get the interrupt. The return queue for the result will then unblock application code.

In your case, the `queue_trans` in the transmit has succeed, which makes the HW ready to communicate with the master; however the `get_trans_result` never returns successfully, because of master hasn't terminates the transaction. 

However because of the timeout set in the `get_trans_result` is not portMAX_DELAY, it will return with ESP_ERR_TIMEOUT when the transaction is not completed in time. If this return value is not handled properly, the application code will not know the transaction is not finished.

chuledeco, 2020-04-30:
Hello @ginkgm and thanks for your answer,

1. I actually have the timeout set to portMAX_DELAY (it is in line 185 of the example)
2. Actually `spi_slave_transmit` makes a `spi_slave_queue_trans` followed by a `spi_slave_get_trans_result`. The result is ESP_OK is both responses are ESP_OK so I get an ESP_OK because the driver finished the `spi_slave_get_trans_result`.

FYI I tried with `spi_slave_queue_trans` followed by a `spi_slave_get_trans_result` and both return ESP_OK.

`spi_slave_get_trans_result` always returns ESP_OK when the CS line is low even if the CS line is held low, it does not way to the falling edge so transactions start in the middle of the previous assertion.

Thanks, Julian

ginkgm, 2020-05-09:
Hi @chuledeco ,

We've reproduced this issue. Due to HW design, when the transaction length (determined by master) reaches the data amount specified by the SW driver, the HW will generate an interrupt, causing the issue you mentioned.

For now, the quickest workaround is that in the slave's application, it always prepare longer transaction than master is expect to send. The final bugfix will not come out very soon, you can try the workaround first. If you find any problem with the workaround, please feel free to post it here.

chuledeco, 2020-05-10:
Hi @ginkgm ,
Thanks for your answer. It does not seem to be the point.

Independently of the slave or master transaction length the problem occurs in the same way. I tested it with many configurations of that.

Now I'm using a third GPIO to allow the CS line to assert the slave only when the SPI peripherals registers are loaded. In that case, when I'm sure that the falling edge on the CS pad is after the SPI peripheral registers are loaded, the SPI slave works fine and the result is successful when the transaction in the slave is longer than in the master or beside (that is why I guess it is not the problem).

At this point, I'm sure the problem is because when the transaction is queued if the CS pad holds low, the RX buffer starts to be loaded immediately instead of waiting for a falling edge on CS pad. In other words, when the transaction is loaded when the master is asserting this slave, the RX starts to be loaded at the middle of a transaction => spurious data.

Thanks, Julian