espressif/esp-idf #5059: Time-of-Flight (TOF) measurements on ESP32-S2?

lubingxian, 2020-04-04:
**Is your feature request related to a problem? Please describe.**

I didn't find any document about the Time-of-Flight (TOF) measurements on ESP32-S2. Is it supported by the esp-idf now?

**Describe the solution you'd like**

Could you release a document or a demo?

krzychb, 2020-04-06:
Hi @lubingxian, 

ToF  is not supported yet. 

For updates please check https://esp32.com/viewtopic.php?f=10&t=14532 including the update history at the end of this post.

"March 29, 2020: added Digital Signature, I2C, I2S, RMT support. Added Wi-Fi ToF to the yet unsupported list."

MS71, 2020-11-16:
Hi,
is there any forecast when this feature will be available?

EpsilonZ, 2020-11-26:
Can we know an estimation at least? :) I've been interested since July, checking the mentioned forum each week, but from August does not seem like there's much progress.

Thanks!

sagb2015, 2020-11-28:
@EpsilonZ - We are planning a beta release for the feature in a week or two. Depending on the feedback, we will make it available on master.

sfambach, 2020-12-30:
@jack0c thanks for implementation. 
I tried to compile the ftm example, but i got some linker error. Checking the code implementations (c files) i cannot find the implementations of the below mentioned functions. I also checked your commit but can not find any implemetations.

I use Branch feature/FTM_SUPPORT (is this the correct one)
 
Menuconfig: I activated all wifi/ftm function including log.

Errors are as follows: 
ftm.elf
cmd.exe /C "cd . && C:\Users\spam\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe  -mlongcalls   @CMakeFiles\ftm.elf.rsp  -o ftm.elf  && cd ."
c:/users/spam/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(ftm_station_main.c.obj):(.literal.wifi_cmd_ftm+0xc): undefined reference to `esp_wifi_ftm_start_initiator'
c:/users/spam/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(ftm_station_main.c.obj): in function `wifi_cmd_ftm':
c:\devel\git\esp32-s2\src\ftm\build/../main/ftm_station_main.c:333: undefined reference to `esp_wifi_ftm_start_initiator'
c:/users/spam/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/esp_wifi/libesp_wifi.a(wifi_init.c.obj):(.literal.esp_wifi_init+0xc): undefined reference to `esp_wifi_set_ftm_report_log_level'
c:/users/spam/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/esp_wifi/libesp_wifi.a(wifi_init.c.obj): in function `esp_wifi_init':
C:/Users/spam/esp-idf/components/esp_wifi/src/wifi_init.c:234: undefined reference to `esp_wifi_set_ftm_report_log_level'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Is there something missing, do i use the wrong branch or miss something, Thanks in advance for your help and the implementation of ftm.

BR SteFam

sagb2015, 2020-12-31:
Hi @sfambach, The functions are in WiFi library. Most likely your submodules are not updated. You can do so using "git submodule update --init --recursive".

EpsilonZ, 2020-12-31:
Hi! For me it's working. There are some configs that cause a crash but I guess you haven't full tested it. Until it is on master I won't worry too much about it. 

What it would be cool from you guys is to have an in-depth study on estimated distance errors on an Espressif lab in which you guys know there is nothing that can cause intereferences or distance errors. I think that would be perfect for the developers. From what I tested, at 10 meters i get a consistent estimation of 5,3-5,7 meters and when moving to 20m it gives a consistent estimation of 6.3-6,7 meters. 

My intuition is that the chip does not calculate in a linear way as you get farther (y=mx+n where m is 1). In close distances (1,1 meters) the results are not really good. Now, I know that WiFi ToF errors can be up to 1,3 - 2m but it would be cool to also have estimated errors in the results we got. I'm sure that the chip shows a error distribution (i.e normal distribution) that can be mitigated by applying some sort of correction. It would be cool if you guys provided more info on that.

Huge thanks for the effort, it is a really cool feature that I'm sure a lot of developers will use.

sagb2015, 2020-12-31:
Hi @EpsilonZ, Thanks for your feedback! We have a second drop for the beta release that should fix some of the bugs and issues reported now. It will also include providing raw hardware values to the users as is, so they can have their own way of using the data. As you rightly pointed, the distance calculation still needs more research and having more inputs from developers like you always helps. We are already evaluating different metrics such as mean with standard deviation, tilt of correctness towards the shorted reported distance etc. Will keep you posted!

sfambach, 2020-12-31:
> Hi @sfambach, The functions are in WiFi library. Most likely your submodules are not updated. You can do so using "git submodule update --init --recursive".
@sagb2015
Thanks git did the job, I was not aware of the submodules thank you very much (what a newbie error ;) ) BR. SteFam

EpsilonZ, 2020-12-31:
Hi @sagb2015 thanks a lot for the response! I'm really looking for the second frop of the beta release :) I've worked a lot with Ultrawideband positioning, so as soon as you get that released I'll be conducting a study comparing both so you guys can check how are the differences and maybe make it available for the developers.

I'm curious if when doing the research will you guys test with NLOS (no line of sight) and LOS (line of sight). It would be cool to have both and see the impacts. When doing this type of estimations, you usually have a table with refraction indexes for each type of material and width of the material (i.e 30cm concrete wall does affect different than a 30cm wood wall). I've got a lot of experience with it so if you want to discuss it, we can maybe create another issue discussing tests that would be cool to have. 

I wish you a great new year!

Trudoysan, 2021-01-19:
Hi @sagb2015, I tested TOF on two ESP32-S2-DevKitM-1 and ftm example from git. I am using latest feature/ftm_support esp-idf. Basic settings (32readings, 200 ms) works great and I am getting reasonable ressults. But when I try parameter  -p 0  softAP always went down:

ftm> E (561255) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (561255) task_wdt:  - IDLE (CPU 0)
E (561255) task_wdt: Tasks currently running:
E (561255) task_wdt: CPU 0: wifi
E (561255) task_wdt: Print CPU 0 (current core) backtrace
Backtrace:0x400A942B:0x3FFC8260 0x4002448A:0x3FFC8280 0x4001A8D1:0x3FFD68D0  |<-CORRUPTED
0x400a942b: task_wdt_isr at C:/Users/Jirka/esp/esp-idf/components/esp_common/src/task_wdt.c:187
0x4002448a: _xt_lowint1 at C:/Users/Jirka/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

I need quick mesurement and I am not interested on fine precission. Thanks in advance

closed 2021-02-01 by igrr

nachiketkukade, 2021-02-03:
Instead of a new beta release, the FTM feature has now been merged with master. Apart from stability fixes, the new changes include ability to perform FTM without any connection and ability to receive raw data at the end from an event. Please refer the updated README from the ftm example for details, thanks!

FrGe2016, 2021-03-24:
All that seems interesting for my application. Before a go further. 
What is the maximum distance i can expect in an outdoor enviromenent with a minimum of obstacles ?
What would the typical accuracy at 20 meters from between 2 ESP32 ?

mikewen, 2021-05-05:
> Instead of a new beta release, the FTM feature has been now merged with master. Apart from stability fixes, the new changes include ability to perform FTM without any connection and ability to receive raw data at the end from an event. Please refer the updated README from the FTM example for details, thanks!

Do you have an example to perform FTM without any connection?
What's the API to receive raw data?

nachiketkukade, 2021-06-30:
It's given in the README.md in the FTM example folder, you need to scan first to find eligible AP's and issue `ftm -I -s <SSID>`.
Raw data is received from the FTM_REPORT event and stored in a global pointer in FTM example. You can check the example code for knowing how to parse it.