espressif/esp-idf #5060: esp_tls_conn_new does not timeout

kifantidis, 2020-04-04:
Hello there.
I'm using esp-idf v4.0. And I've noticed that some code is missing from esp_tls.c file. 

I'm not familiar of pull requests and other stuff and other things so i will post what I've found here and I hope it will help others too.

I've tried to use timeout config for "esp_tls_conn_new" command and I've noticed that the process was stuck and never timed out. So I did some digging and I've found that the timeout code was missing from the esp_tls.c file. In master version code was there and I've copied/pasted and everything worked. 

Old code starting at line 660:
esp_tls_t *esp_tls_conn_new(const char *hostname, int hostlen, int port, const esp_tls_cfg_t *cfg)
{
    esp_tls_t *tls = (esp_tls_t *)calloc(1, sizeof(esp_tls_t));
    if (!tls) {
        return NULL;
    }
    /* esp_tls_conn_new() API establishes connection in a blocking manner thus this loop ensures that esp_tls_conn_new()
       API returns only after connection is established unless there is an error*/
    while (1) {
        int ret = esp_tls_low_level_conn(hostname, hostlen, port, cfg, tls);
        if (ret == 1) {
            return tls;
        } else if (ret == -1) {
            esp_tls_conn_delete(tls);
            ESP_LOGE(TAG, "Failed to open new connection");
            return NULL;
        }
    }
    return NULL;
}

New code after changes: 

esp_tls_t *esp_tls_conn_new(const char *hostname, int hostlen, int port, const esp_tls_cfg_t *cfg)
{
    esp_tls_t *tls = (esp_tls_t *)calloc(1, sizeof(esp_tls_t));
    if (!tls) {
        return NULL;
    }
    /* esp_tls_conn_new() API establishes connection in a blocking manner thus this loop ensures that esp_tls_conn_new()
       API returns only after connection is established unless there is an error*/
    size_t start = xTaskGetTickCount();
    while (1) {
        int ret = esp_tls_low_level_conn(hostname, hostlen, port, cfg, tls);
        if (ret == 1) {
            return tls;
        } else if (ret == -1) {
            esp_tls_conn_delete(tls);
            ESP_LOGE(TAG, "Failed to open new connection");
            return NULL;
        }
        else if ((ret == 0) && (cfg->timeout_ms >= 0)) 
        {
            size_t timeout_ticks = pdMS_TO_TICKS(cfg->timeout_ms);
            uint32_t expired = xTaskGetTickCount() - start;
            if (expired >= timeout_ticks) {
                esp_tls_conn_delete(tls);
                ESP_LOGE(TAG, "Failed to open new connection in specified timeout");
                return NULL;
            }
        }
    }
    return NULL;

 }

I hope this will help others also. 

Best regards, Kostas.

Alvin1Zhang, 2020-04-07:
@kifantidis Thanks for reporting, we will look into.

AdityaHPatwardhan, 2020-04-07:
@kifantidis  thanks for the concern but if you check `esp_tls.c` in branch [release/v4.0](https://github.com/espressif/esp-idf/blob/release/v4.0/components/esp-tls/esp_tls.c) and also in branch  [release/v3.3](https://github.com/espressif/esp-idf/blob/release/v3.3/components/esp-tls/esp_tls.c) , the `config->timeout` option is still there, Before that in `release/v3.2`, it is not present no reason why it should be,  and since it is a very old branch we recommend you to shift to latest branches such as `release/v4.1`. 
If this solves your query you may please close the issue. Thanks you

kifantidis, 2020-04-07:
The reason why I created this issue is because I have release/v4.0. I've downloaded it 4-5 days after the release.

> pc_username@DESKTOP-97HD2QS MINGW64 /c/02_str
> $ cd c:/esp/esp-idf
> 
> pc_username@DESKTOP-97HD2QS MINGW64 /c/esp/esp-idf ((v4.0))
> $ git describe
> v4.0

@AdityaHPatwardhan  please close the case. Thank you for answering.

AdityaHPatwardhan, 2020-04-07:
@kifantidis Please download it again, maybe there was an error in downloading the files in your case but `release/v4.0` indeed has the `timeout` change.
@Alvin1Zhang , can you please close the issue, The issue is resolved.

Alvin1Zhang, 2020-04-07:
@kifantidis Thanks for reporting and @AdityaHPatwardhan thanks for supporting and follow up.

closed 2020-04-07 by Alvin1Zhang