espressif/esp-idf #5061: Wifi: Get rid of inefficient wifi packets

seppf, 2020-04-04:
Let's make it clear: 802.11b is dead - since long. I have to dig very deep in my sample device boxes to find a pure 802.11b device.
One of the major drawbacks of 802.11b is the minimum bitrate of 1 MBit/s that is used in some (but not all!) bgn systems for management frames, broadcasts and multicasts.
Just have a look at how much air time is wasted for a single 1 MBit beacon, ACK or a response to a wildcard (worst of all!) probe request when "normal" data packets rush along with 50, 60 or 70 MBits.
And why this all? To be backward compatible with wifi clients that lie six feet under since 10 years or more?
So please, let's have a gn mode in addition to b / bg / bgn. OFDM modulation is more efficient anyhow, So only a few users will be sad about not using CCK anymore.
With OFDM-only basic bitrate would be 6 MBits for both access points and stations. Obviously this boosts this set of packets as compared to 1 MBit. And in most cases the required ~2 dB more SNR don't really matter, as there is sufficient SNR for much higher bitrates.

Futhermore it would help to have an API function for (even dynamically) setting the bitrate for management frames, broadcasts and multicast even higher, for cases when throughput is more import than long reach.

Thanks for considering,
Best Regards, Josef

Alvin1Zhang, 2020-04-07:
@seppf Thanks for proposing this feature, we will evaluate. Thanks.

seppf, 2020-04-07:
Hi @Alvin1Zhang,
thanks for considering.
If I didn't make thgis clear enough, having 802.11g with ofdm-only would give the full performance with no 802.11b backward compatibility, but even using DSSS2M with short preamble for managment / broadcast / multicast would reduce air time considerably keeping the 802.11b compatibility.
Standard access points typically have two setting options: one bit flag for the ofdm-only mode and  one parameter for the tx-managment rate to use. Even without these setting many APs use 2MBits by default. Unfortunately I have no idea how difficult this would be to implement on esp32.  
Best Regards, Josef

liuzfesp, 2020-04-07:
Hi @seppf, thanks for your suggestions. We will consider to implement it. Hope we can implement it in June or July.

seppf, 2020-04-07:
Hi,
thanks again for considering and for giving information about time scale.
I'm aware that those performance issues may not be in the scope of implementation with a system like esp32 at all.
Maybe you could help me with this question: is this forum also intended for HW related issues or bug reports or should this better be placed in the Espressif forum?
Best Regards, Josef

liuzfesp, 2020-04-08:
Hi @seppf, this forum is also intended for HW related issues. You can raise any HW related issues or concerns here.

liuzfesp, 2020-07-08:
Hi @seppf we already made the minimum rate of data frame to 5.5M by default.
For 11g only mode, I don't think we will support it in short term since it's a big change for us.

closed 2022-01-06 by jack0c

HavingaThijs, 2023-10-19:
Using the ESP32-C6 in 802.11ax mode, I encountered that it is still using 802.11b packets for authentication. The `esp_wifi_set_protocol` function throws an error if `WIFI_PROTOCOL_11B` is not set in combination with `WIFI_PROTOCOL_11AX`. I was wondering whether this is only a limitation of the driver or if it is simply not possible to disable 802.11b for the Wi-Fi 6 chip.

Please let me know if I should open a new issue for this.

HavingaThijs, 2023-10-19:
For anyone encountering this: I worked around the issue by disabling 802.11b rates with `esp_wifi_config_11b_rate`.

xuxiao111, 2023-10-23:
Hi @HavingaThijs, the esp_wifi_set_protocol function throws an error, you can check this https://github.com/espressif/esp-idf/issues/702. The function of esp_wifi_config_11b_rate is used to enable/disable 802.11b rate.