espressif/esp-idf #5062: BLE-MESH generic level server model get response

sheldon123z, 2020-04-05:
#3443  Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [idf.py]

- Operating System:     [macOS]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [NO]
- Power Supply:         [USB]

## Problem Description

Hi, I wonder how to respond to the ``ESP_BLE_MESH_MODEL_OP_GEN_LEVEL_GET`` op-code. From the example, the ``esp_ble_mesh_model_publish`` and ``esp_ble_mesh_server_model_send_msg`` are used for the generic onoff models. The data type for the "data" parameter is uint8_t for the above two functions, but the data type of the "level" member in the ``esp_ble_mesh_gen_level_state_t`` is uint16_t, I tried to convert it to uint8_t but it doesn't work for the phone app. There is nothing shown on the app when I click the "Read" button on the model interface. 
Are there any other functions that we can use for sending messages?

Thanks.

Campou, 2020-08-10:
Hi @sheldon123z 

Sorry for the late reply.

For server model sending messages, only the `esp_ble_mesh_server_model_send_msg` & `esp_ble_mesh_model_publish` should be used, no other functions. And the `data` parameter for the two functions is just a `uint8_t *` pointer, for responding with a `uint16_t` data, you can provide the data length and covert the data to a `uint8_t *` pointer. 

BTW, does this issue still exist?

Thanks.

Alvin1Zhang, 2024-12-23:
Thanks for reporting, feel free to reopen with more updates.