espressif/esp-idf #5066: examples/protocols/http_server/restful_server doesn't flash correctly when following the instructions

NANASHI0X74, 2020-04-05:
## Environment

- Development Kit:     ESP32-DevKitC (v4)
- Module or chip used:  ESP32-WROOM-32D|
- IDF version v4.2-dev-1035-g9f0ef7f88
- Build System:      idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): command not found
- Operating System:     Linux (ubuntu)
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

when I follow the instructions exactly (exec npm run build in frontend dir, run idf.py flash monitor), flash errors because: no such file or directory. **Running idf.py build fixes this.**

suda-morris, 2020-04-06:
Sorry I didn't reproduce the issue you met. 
running `idf.py build flash monitor` and `npm run build` should from different dirs.

NANASHI0X74, 2020-04-06:
if you take a look at https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/restful_server/README.md
it says to run idf.py flash monitor, omitting the build step. That's the problem and it tripped me up for a while. I assumed that idf.py flash actually does the build too. And it does seem to do that actually which in most cases works. just in this case the flash step aparently doesn't produce www.bin, which I'm assuming is the bundled web application

suda-morris, 2020-04-07:
Ah I see now, thanks for reporting @NANASHI0X74 it seems like a regression, we will look into it.

renzbagaporo, 2020-04-07:
Hi @NANASHI0X74, this was a regression in our build system. A fix has been submitted internally. In the meantime, suggest using the executing `build`.

closed 2020-04-09 by espressif-bot