espressif/esp-idf #5068: mdns cannot resolve hostname by giving IP

AxelLin, 2020-04-06:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-1034-ge599b794bebe
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
Operating System: Linux
Power Supply: USB

Test using examples/protocols/mdns.

Verify resolve host name and address by below commands on Linux:

$ avahi-resolve-host-name esp32-mdns.local
esp32-mdns.local        192.168.0.104

$ avahi-resolve-address 192.168.0.104
Failed to resolve address '192.168.0.104': Timeout reached

Expected behavior: avahi-resolve-address should work.

AxelLin, 2020-04-15:
From the wireshark, it's clear esp32 mdns does not response to below: ( by avahi-resolve-address 192.168.0.104 )
19      4.649283002     192.168.0.103   224.0.0.251     MDNS    86      Standard query 0x0000 PTR 104.0.168.192.in-addr.arpa, "QM" question

Is this a bug?

david-cermak, 2020-04-20:
@AxelLin 
The esp mdns library doesn't support reverse lookups.

This could be implemented, if desired, preferably if more users vote for this functionality.

AxelLin, 2020-04-21:
Hi @david-cermak 
Then can I use the mdns responder from lwIP? (How to enable mdns from lwIP?)

david-cermak, 2020-04-21:
@AxelLin 
I'm not familiar with lwip mdns responder (seems to have some limitation in conflict resolution)
To get it compiled, you might need to provide some of those definitions as *PUBLIC*,  since they are used in public APIs in other components, too. Also adding a `apps/mdns/mdns.c` might be needed.

``` cmake
idf_component_get_property(lwip lwip COMPONENT_LIB)
target_compile_definitions(${lwip} PUBLIC -DLWIP_MDNS_RESPONDER=1 PUBLIC -DLWIP_NUM_NETIF_CLIENT_DATA=1 PRIVATE -DLWIP_NETIF_EXT_STATUS_CALLBACK=1)
target_sources(${lwip} PRIVATE "lwip/src/apps/mdns/mdns.c")
```

AxelLin, 2020-04-22:
Hi @david-cermak 

Follow above changes, I can include lwIP mdns to build.
However, it seems does not work well.
From wireshark log I can observe the device keep sending query but does not send response.
This is strange because exactly the same lwip mdns.c code works when I test on other platform(using different sdk but with similar lwip version).

Thanks,
Axel

closed 2020-06-05 by AxelLin