espressif/esp-idf #5069: ppp interface is UP after calling esp_netif_new()

AxelLin, 2020-04-06:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-1034-ge599b794bebe
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
Operating System: Linux
Power Supply: USB

With below code, the esp_netif_is_netif_up() returns true.

    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_PPP();
    esp_netif_t *esp_netif = esp_netif_new(&cfg);
    assert(esp_netif);

    bool is_up = esp_netif_is_netif_up(esp_netif);

This looks wrong to me because the modem is not yet dial-up and it even not yet init the modem.
For comparison, in STA and AP cases, after calling esp_netif_create_default_wifi_sta() or esp_netif_create_default_wifi_ap()
the interface is still DOWN.

The problem is when PPP is consider UP, the route algorithm will select it as default route.
But there could be some reason the application doesn't call dial-up immediately or never calls dial-up because sim card is not properly inserted.

AxelLin, 2020-04-06:
add @david-cermak

AxelLin, 2020-04-07:
If add checking netif_is_link_up() in esp_netif_is_netif_up(), the result looks fine.

see https://github.com/espressif/esp-lwip/commit/3ca5184998503dedec06adfad98ab604d137f5ff
the ppp netif now using netif_set_link_{up,down} instead of netif_set_{up,down} when PPP
reaches/leaves running state. PPP interface is now set to administratively
UP when created with link state down.

david-cermak, 2020-04-20:
Hi @AxelLin 

Thanks for reporting this issue and posting the hint on ppp netif status.
This needs to be fixed!

AxelLin, 2020-04-25:
I fixed it by below changes:
https://github.com/espressif/esp-idf/pull/5183

closed 2020-05-05 by espressif-bot