espressif/esp-idf #5070: esp_wifi_set_country shows "wifi:External AP's channel is out of channel range, disconnect"

AxelLin, 2020-04-06:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-1034-ge599b794bebe
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
Operating System: Linux
Power Supply: USB

First, I call esp_wifi_get_country and found the module's default country code is CN:
cc=CN, schan=1, nchan=13, policy=0

Then I call esp_wifi_set_country to change the country code to US, but got below warning:
I (10745) wifi:set country: cc=US schan=1 nchan=11 policy=0

W (10752) wifi:External AP's channel is out of channel range, disconnect

The expected behavior: It should work without any warnings.

The warning is not useful to me since it does not tell why AP's channel is out of channel range.
The warning is also quite strange because in my test code I got the warning *before*
calling esp_wifi_set_mode/esp_wifi_set_config/esp_wifi_start.

zhangyanjiaoesp, 2020-04-07:
@AxelLin 
If the ESP32 is in channel 12 or 13 before you call `esp_wifi_set_country` to change the country to [1,11,US], this log will be printed. 

Anyway, this warning may look strange to users, we will modify it or just delete it.

AxelLin, 2020-04-07:
@zhangyanjiaoesp 

But the point is wifi is not initialized yet.
I call esp_wifi_set_country first and got the message.

esp_wifi_set_mode/esp_wifi_set_config/esp_wifi_start are not yet called.
So I even don't know which channel was used when I call esp_wifi_set_country.
(And I don't think it's default is 12 or 13, something might be just wrong.)

zhangyanjiaoesp, 2020-04-07:
@AxelLin 

> First, I call esp_wifi_get_country and found the module's default country code is CN:
> cc=CN,` schan=1, nchan=13, policy=0
This means the `esp_wifi_init()` has been called. 

1. Which mode does your ESP32 in?
2. Please show me how you call the APIs. Like a demo how to reproduce your problem.

AxelLin, 2020-04-07:
@zhangyanjiaoesp 

I tested on ESP32-WROOM-32, Chip is ESP32D0WDQ6 (revision 1).

You can use examples/wifi/getting_started/softAP to reproduce this issue.
Attached diff of my modification and see the comments which shows my debug print output.

To reproduce the issue:
1. idf.py erase_flash
2. idf.py flash monitor

PS. The warning does not show up at the second time reboot.
    So make sure you do erase_flash first to see the symptom.

[softAP-example-diff.txt](https://github.com/espressif/esp-idf/files/4444904/softAP-example-diff.txt)

zhangyanjiaoesp, 2020-04-08:
@AxelLin
 Before calling `esp_wifi_start()`, the channel of ESP32 is 0, in this time you call `esp_wifi_set_country` to set channel range to [1,11], 0 is out of range, so the warning is printed.

Suggest you set country after `esp_wifi_start()` has been called.

AxelLin, 2020-04-08:
@zhangyanjiaoesp 

I'm surprised to call esp_wifi_start() before esp_wifi_set_country(),
it looks like a possibility of using wrong country code before applying new setting.

BTW, I don't find where the ESP-IDF set the default country code to CN.
It would be even better if I can override the default rather than calling esp_wifi_set_country again in my application code.
And from the fact I only see the warning for the first time boot, I'm wondering if the device saved the country code setting in NVS?

zhangyanjiaoesp, 2020-04-08:
@AxelLin 
Yes, when `esp_wifi_init()` is called, the device will get default country [1,13, CN] in NVS load.

AxelLin, 2020-04-08:
@zhangyanjiaoesp 

In wifi_init_config_t I only found nvs_enable flag set to WIFI_NVS_ENABLED.
I don't find where to change the default country code setting.
Could you provide an example to init different count code without calling esp_wifi_set_country()?

zhangyanjiaoesp, 2020-04-08:
@AxelLin We don't support this now. You need to call esp_wifi_set_country to set different country.

AxelLin, 2020-04-08:
@zhangyanjiaoesp 

I'm not convinced calling esp_wifi_set_country() after esp_wifi_start() is correct.
So I'm considering to keep my code as is, call esp_wifi_set_country() before
esp_wifi_start() and just ignored the warning.
Thanks for your prompt support.

closed 2020-04-08 by AxelLin

zhangyanjiaoesp, 2020-04-08:
ok, thanks. We will update the warning log.