espressif/esp-idf #5071: libmodbus components not works from C++ code

merlokk, 2020-04-06:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version: v2
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
   v4.1-beta1
- Build System:         idf.py
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:      USB

## Problem Description
libmodbus do not works from C++ code without:
```
extern "C" {
#include "mbcontroller.h"
}
```
### Expected Behavior
`#include "mbcontroller.h"` works without tricks

### Steps to reproduce

include `libmodbus` from c++ module

alisitsyn, 2020-04-07:
Hi  @merlokk,
Thanks for the issue. The CPP using of this library was not tested during development. What I can propose as a fix for this issue is to add `extern "C" { }` block into `mbcontroller.h` header file:

```
#ifndef _MODBUS_CONTROLLER_COMMON
#define _MODBUS_CONTROLLER_COMMON

#ifdef __cplusplus
extern "C" {
#endif

#include <stdint.h>                 // for standard int types definition
#include <stddef.h>                 // for NULL and std defines
#include "string.h"                 // for strerror()
#include "errno.h"                  // for errno
#include "esp_err.h"                // for error handling
#include "driver/uart.h"            // for uart port number defines
#include "sdkconfig.h"              // for KConfig options

#include "esp_modbus_master.h"
#include "esp_modbus_slave.h"

#ifdef __cplusplus
}
#endif

#endif
 ```
This will allow to avoid using "tricks" in your CPP code.

merlokk, 2020-04-07:
thank you)
yes, it works.
i just opened issue to add it to next version of sdk)

alisitsyn, 2020-04-07:
Ok. This solution will be added to following build of ESP-IDF. It will take some time. Thanks.

Alvin1Zhang, 2020-08-18:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/b732abec7f9082f6596c8a42976906a51bcac907. Feel free to reopen if the issue still happens. Thanks.

closed 2020-08-18 by Alvin1Zhang