espressif/esp-idf #5074: Add IDF version environment variable for use in CMakeLists.txt to assist with maintaining 3.x/4.x projects

DavidAntliff, 2020-04-07:
In version 4.1-beta1 of the IDF, some CMake variables have changed or been removed. For example, in v3.3 there is an `IDF_PROJECT_EXECUTABLE` variable, that in 4.1-beta1 has been removed, possibly replaced with `project_elf`.

This is a non-backwards compatible change (contrary to the docs which state that the v3.3 CMake Build System is still supported).

The issue I have is that I'd like to support my open-source projects for both v3.3 and future v4.x versions, but I need a way to identify which version of the IDF the user is using to build the project.

I haven't been able to find an existing variable (created by `export.sh`) in either v3.3 or v4.1-beta1 that I can reliable use to make the decision about which project ELF variable to use.

One alternative is to search variables such as `IDF_PATH` and attempt to extract the version number from this path, however this fails if the user has installed the IDF in the default `esp-idf` folder.

There is also no obvious `$IDF_PATH/version` file whose contents can be easily examined to determine the IDF version.

The best alternative I have at the moment is to run `cd $IDF_PATH && git tag --points-at HEAD` and grab the output, but this will fail if the user has moved off the released commit (e.g. when working on a PR, or cherry-picking a commit from elsewhere).

_Please consider adding an environment variable that a project's CMakeLists.txt can use to select which major version of the IDF to conform to._

projectgus, 2020-04-07:
Hi @DavidAntliff,

> In version 4.1-beta1 of the IDF, some CMake variables have changed or been removed. For example, in v3.3 there is an IDF_PROJECT_EXECUTABLE variable, that in 4.1-beta1 has been removed, possibly replaced with project_elf. 

I'm sorry for the difficulty caused by this change. Between v3.x and v4.0 we made some substantial changes to the internals of the CMake build system, and as a result some of these parts changed. 

There is a summary in the [Breaking Changes since v3.x](https://github.com/espressif/esp-idf/releases/v4.0) section of the v4.0 release notes, in this case this variable comes under the larger change of "*CMake "external project" integration API is different to ESP-IDF v3.x and not compatible.*"

We try not to make this kind of breaking change as we know it's a massive hassle, but sometimes it's inevitable as a result of some other requirements. Our policy is to only ever do it on [major version updates](https://docs.espressif.com/projects/esp-idf/en/stable/versions.html#versioning-scheme), and to list any breaking changes in the major version release notes.

> This is a non-backwards compatible change (contrary to the docs which state that the v3.3 CMake Build System is still supported).

If there's a part of the docs that you think is misleading on this point, please let me know and I'll make sure we update it to be clearer. We want to communicate clearly what can be expected when updating ESP-IDF.

> Please consider adding an environment variable that a project's CMakeLists.txt can use to select which major version of the IDF to conform to.

If I understand the requirement correctly, there is something you can use. 

The CMake build system since v4.0 defines a set of variables for the IDF version: https://docs.espressif.com/projects/esp-idf/en/v4.0/api-guides/build-system.html#build-project-variables

Will the `IDF_VERSION_MAJOR` variable meet your needs?

These aren't defined if using ESP-IDF v3.x versions, but you can assume that any ESP-IDF version that supports CMake but doesn't doesn't have this variable is v3.x, ie:

```
if(NOT IDF_VERSION_MAJOR)
  set(IDF_VERSION_MAJOR 3)
endif()
```

These project-wide variables should be defined in the project CMakeLists file after including either `idf.cmake` or `project.cmake`, whichever is used in the particular project CMakeLists file.

Does that meet the requirements of your project? In any case, please let me know if there is anything else we can add to make it easier to support both v3.x and v4.x in a single project.

DavidAntliff, 2020-04-07:
Hi @projectgus, thank you for your reply.

Re-reading the release notes I see that v3.x projects are intended to break. Sorry, for some reason I thought that the CMake system would continue but the old Makefile system would disappear entirely. Reading it again I see that even the CMake system is expected to break. Ok, I can live with that :)

`IDF_VERSION_MAJOR` should actually do the trick. I wasn't able to find mention of a variable such as this in the CMake docs but that's probably more the fault of my searching than the docs, as I see they are very clearly specified and documented.

So, sorry for taking up your time on this, I think I can make use of these variables. I'll close this issue.

closed 2020-04-07 by DavidAntliff