espressif/esp-idf #5075: IDF v3.3: idf.py --help reports `defconfig` action, but it is not a recognised option

DavidAntliff, 2020-04-07:
I am using ESP-IDF v3.3.

When displaying `idf.py --help`, the option `defconfig` is shown as an available action:

```
$ idf.py --help
usage: idf.py [-h] [-p PORT] [-b BAUD] [-C PROJECT_DIR] [-B BUILD_DIR]
              [-G {Ninja,Unix Makefiles}] [-n] [-v]
              [-D DEFINE_CACHE_ENTRY [DEFINE_CACHE_ENTRY ...]] [--no-ccache]
              {all,build,clean,fullclean,reconfigure,menuconfig,defconfig,confserver,size,size-components,size-files,bootloader,bootloader-clean,bootloader-flash,app,app-flash,efuse_common_table,efuse_custom_table,show_efuse_table,partition_table,partition_table-flash,flash,erase_flash,monitor,erase_otadata,read_otadata}
              [{all,build,clean,fullclean,reconfigure,menuconfig,defconfig,confserver,size,size-components,size-files,bootloader,bootloader-clean,bootloader-flash,app,app-flash,efuse_common_table,efuse_custom_table,show_efuse_table,partition_table,partition_table-flash,flash,erase_flash,monitor,erase_otadata,read_otadata} ...]

ESP-IDF build management tool

positional arguments:
  {all,build,clean,fullclean,reconfigure,menuconfig,defconfig,confserver,size,size-components,size-files,bootloader,bootloader-clean,bootloader-flash,app,app-flash,efuse_common_table,efuse_custom_table,show_efuse_table,partition_table,partition_table-flash,flash,erase_flash,monitor,erase_otadata,read_otadata}
                        Actions (build targets or other operations)
```

However when attempting to use this option:

```
 $ idf.py defconfig
...
Executing "ninja defconfig"...
ninja: error: unknown target 'defconfig', did you mean 'menuconfig'?
ninja failed with exit code 1
```

I would expect this option to generate `sdkconfig` and `build/sdkconfig/sdkconfig.h` using the default values specified in `Kconfig.projbuild` and `sdkconfig.defaults` (if it exists).

projectgus, 2020-04-07:
Hi @DavidAntliff,

Thanks for reporting this. The `defconfig` subcommand was removed in v4.0 (see #3965) but the change was not backported to v3.3.

The equivalent command to generate `sdkconfig.h` from CMake is `idf.py reconfigure`. This will work the same in ESP-IDF v3.x and v4.x.

We'll update `idf.py` to either remove this command (which never worked in the CMake Build System Preview) or alias it to `reconfigure`.

kumekay, 2021-05-03:
@DavidAntliff Since `idf.py defconfig` never worked for IDF v3.3 we removed it from the list. If you are ok using `reconfigure` instead could you please close this issue?

And thank you once again for reporting it!

DavidAntliff, 2021-05-04:
Happy to use `reconfigure` :)

closed 2021-05-04 by DavidAntliff