espressif/esp-idf #5077: esp_mqtt_client_publish blocking and causing task watchdog to fire

will-emmerson, 2020-04-07:
## Environment
- Development Kit: none
- Module or chip used: ESP32-WROOM-32
- IDF version: v4.2-dev-1034-ge599b794b
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: Clion
- Power Supply: external 3.3V

## Problem Description
`esp_mqtt_client_publish` appears to be blocking and this is causing the task watchdog to fire. This is only happening intermittently e.g. after several hours/days.

I am sending a fairly small amount of messages overall, probably 30-40 a minute but these tend to be grouped. I am replaying the same sequence every minute so it's not as if a sudden change in throughput is causing the problem. I am using less than 1% of cpu time according to `vTaskGetRunTimeStats`.

I am calling `esp_task_wdt_reset` before and after `esp_mqtt_client_publish` and `CONFIG_ESP_TASK_WDT_TIMEOUT_S` is at the default of 5.

Actually as usual as I'm just about to submit this issue I might just have seen the problem in the coredump below - I can see that `esp_transport_write` has a timeout of 10s, but where is this set?
`esp_transport_write (t=0x3ffe8710, buffer=0x3ffe9f06 <error: Cannot access memory at address 0x3ffe9f06>, len=324, timeout_ms=10000)`
Or should I just set the watchdog timeout to be longer than 10s?

All my tasks are running at priority=1. I am currently running a test with my mqtt task at priority 2 and the problem hasn't happened yet but it hasn't been running long, I don't know how much difference priority would make.

### Expected Behavior
`esp_mqtt_client_publish` shouldn't block for longer than `CONFIG_ESP_TASK_WDT_TIMEOUT_S`

### Actual Behavior
`esp_mqtt_client_publish` blocks for longer than `CONFIG_ESP_TASK_WDT_TIMEOUT_S`

## Other items if possible

sdkconfig.defaults:
```
# ethernet
CONFIG_ETH_ENABLED=y
CONFIG_ETH_USE_ESP32_EMAC=y
CONFIG_ETH_PHY_INTERFACE_RMII=y
CONFIG_ETH_RMII_CLK_OUTPUT=y
CONFIG_ETH_RMII_CLK_OUT_GPIO=17
CONFIG_ETH_DMA_BUFFER_SIZE=1600
CONFIG_ETH_USE_SPI_ETHERNET=n

# misc
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_ESP32_ENABLE_COREDUMP_TO_FLASH=y
CONFIG_ESP32_CORE_DUMP_STACK_SIZE=4096

# factory reset
CONFIG_BOOTLOADER_FACTORY_RESET=y
CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET=35
CONFIG_BOOTLOADER_OTA_DATA_ERASE=y
CONFIG_BOOTLOADER_DATA_FACTORY_RESET="nvs"

# debugging
CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT=y
CONFIG_HEAP_POISONING_COMPREHENSIVE=y
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS=y
CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y
CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM=y
CONFIG_COMPILER_STACK_CHECK=y
CONFIG_COMPILER_STACK_CHECK_MODE_NORM=y
CONFIG_COMPILER_WARN_WRITE_STRINGS=y

CONFIG_ESP_TASK_WDT_PANIC=y
```

coredump:
```
[... 265 lines cut ...]
.coredump.tasks.data 0x3ffc69e0 0x168 RW 
.coredump.tasks.data 0x3ffc6750 0x22c RW 
.coredump.tasks.data 0x3ffbd88c 0x168 RW 
.coredump.tasks.data 0x3ffbd6d0 0x1a8 RW 
.coredump.tasks.data 0x3ffbd22c 0x168 RW 
.coredump.tasks.data 0x3ffbd050 0x1c8 RW 
.coredump.tasks.data 0x3ffc7a40 0x168 RW 
.coredump.tasks.data 0x3ffc7820 0x20c RW 
.coredump.tasks.data 0x3ffc91e8 0x168 RW 
.coredump.tasks.data 0x3ffc9000 0x1d4 RW 
.coredump.tasks.data 0x3ffd2978 0x168 RW 
.coredump.tasks.data 0x3ffd27a0 0x1c4 RW 
.coredump.tasks.data 0x3ffbe840 0x168 RW 
.coredump.tasks.data 0x3ffec8d0 0x1f0 RW 
.coredump.tasks.data 0x3ffdb7c0 0x168 RW 
.coredump.tasks.data 0x3ffdb5b0 0x1fc RW 
.coredump.tasks.data 0x3ffafb34 0x168 RW 
.coredump.tasks.data 0x3ffaf960 0x1c0 RW 
.coredump.tasks.data 0x3ffe6540 0x168 RW 
.coredump.tasks.data 0x3ffe6210 0x31c RW 

===================== ESP32 CORE DUMP END =====================
===============================================================

```

Alvin1Zhang, 2020-04-08:
@will-emmerson Thanks for the detailed report, we will look into.

david-cermak, 2020-04-17:
Hi @will-emmerson 

As for the `esp_transport_write()` there's no need for setting WTD timeout to the transport timeout (10s) or higher, since it internally uses select, so even if it blocks for the whole 10s it still yields to lower prio tasks if socket not writable (and this is what's seen in the core dump, btw)

The transport timeout is not adjustable at the moment and is only set here https://github.com/espressif/esp-mqtt/blob/master/include/mqtt_config.h#L45

It might be possible that the `esp_mqtt_client_publish()` would "busy block" (i.e. causing the IDLE tasks to starve and the WDT to trigger), but that would practically happen only and when publishing very large data (typically MBs), as the payloads longer than the mqtt buffer would get fragmented and transmitted in a loop. What is the approximate size of your data to publish, after they accumulate?
But, this is not very likely to happen and, looking at your core dump, this is not what's happing here:
```
==================== THREAD 10 (TCB: 0x3ffe6d04, name: 'mqtt_publish') =====================
#0  0x40081a38 in esp_crosscore_int_send_yield (core_id=0) at /home/willemmerson/esp/esp-idf/components/esp32/crosscore_int.c:110
#1  0x40096920 in xQueueGenericReceive (xQueue=0x3fff367c, pvBuffer=0x0, xTicksToWait=<optimized out>, xJustPeeking=0) at /home/willemmerson/esp/esp-idf/components/soc/src/esp32/include/hal/cpu_ll.h:39
...
#3  0x401077a7 in lwip_select (maxfdp1=58, readset=0x0, writeset=0x3ffe7c80, exceptset=0x3ffe7c88, timeout=0x3ffe7c78) at /home/willemmerson/esp/esp-idf/components/lwip/lwip/src/api/sockets.c:2154
...
#13 0x400e39dc in esp_mqtt_client_publish (client=0x3ffe7ff0, topic=<optimized out>, data=<optimized out>, len=<optimized out>, qos=0, retain=<optimized out>) at /home/willemmerson/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1576
```
as you can see, the `esp_mqtt_client_publish()` in this case is blocked on select.

It is not very clear from this core dump what is the source of the abort() in IDLE task. I can see that all listed tasks are generally blocked on some semaphore/queue or similar. One potential issue is that the core dump is not complete, which might be the case if both cores are involved. What would help in this case is, if you can provide the actual backtrace from the board log, so something like:
```
Backtrace:0x400D1F04:0x3FFB0750 0x400821DD:0x3FFB0770 0x400D29A1:0x3FFB4810 0x400D149E:0x3FFB4840
```
with that information and knowing the elf file for decoding the addresses we could better understand the source of the issue.

will-emmerson, 2020-04-17:
Unfortunately I don't have the original coredump so will have to try and recreate the problem again. The coredump was saved to flash, sent to a server on the next boot and then was viewed with `espcoredump.py info_corefile build/8-channel.elf --core coredump.elf` so maybe something was lost in that process.
I set `CONFIG_TASK_WDT_TIMEOUT_S=15` and the problem hasn't happened since then (been running fine for 10 days or so) which supports the idea it's something to do with mqtt timeout but then again, could be anything.
I am not sending anything large, just tiny messages of a few hundred chars, and not really that many.

I am confused about the task watchdog, because even tasks which yield to lower priority tasks cause the watchdog to fire. For instance in this code, commenting out `esp_task_wdt_add` makes watchdog fire even though `vTaskDelay` should be yielding:

```
static void led_blink_task() {
    ESP_ERROR_CHECK(esp_task_wdt_add(NULL));
    while (true) {
//        ESP_ERROR_CHECK(esp_task_wdt_reset());
        pin_toggle(LED1);
        vTaskDelay(ONE_SECOND);
    }
}
```

will-emmerson, 2020-04-17:
Is there something weird going on with `emac_rx` task in the above coredump by the way? It's the only task not waiting.

david-cermak, 2020-04-17:
> because even tasks which yield to lower priority tasks cause the watchdog to fire

It should not, if you start from a clean project and setup a very high prio task to
``` c
while(1) {
    vTaskDelay(ONE_YEAR);
    }
```
the task WDT should *not* fire. Does the error message indicate which task/core caused the WDT expiry? was it the `led_blink_task()`?

> Is there something weird going on with emac_rx task in the above coredump

I don't think so, it's blocked on a `ulTaskNotifyTake()` which is also a kind of a semaphore...

It would be interesting to see what's happening on the other core, but unfortunately the core dump doesn't list tasks on both cores (there is a plan to support this soon in IDF)

There's been another issue https://github.com/espressif/esp-idf/issues/1981#issuecomment-516743396, which might indicate there could be something wrong with the task WDT triggering and consequent processing, but has to do with UART interrupts/not clearing flags...

will-emmerson, 2020-04-20:
I did as you said and created a clean project with the following code but the task watchdog still fired:

```c
#include <esp_task_wdt.h>

void loop_task() {
    ESP_ERROR_CHECK(esp_task_wdt_add(NULL));
    while (true) {
        vTaskDelay(portMAX_DELAY);
    }
}

void app_main(void) {
    xTaskCreate(loop_task, "loop_task", 2048, NULL, configMAX_PRIORITIES, NULL);
}
```

Output:

```
�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7148
load:0x40078000,len:13696
load:0x40080400,len:3984
0x40080400: _init at ??:?

entry 0x40080688
I (29) boot: ESP-IDF v4.2-dev-1034-ge599b794b-dirty 2nd stage bootloader
I (29) boot: compile time 09:25:35
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x053e8 ( 21480) map
I (122) esp_image: segment 1: paddr=0x00015410 vaddr=0x3ffb0000 size=0x0225c (  8796) load
I (126) esp_image: segment 2: paddr=0x00017674 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/willemmerson/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (132) esp_image: segment 3: paddr=0x00017a80 vaddr=0x40080404 size=0x08598 ( 34200) load
I (155) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x12878 ( 75896) map
0x400d0020: _stext at ??:?

I (184) esp_image: segment 5: paddr=0x000328a0 vaddr=0x4008899c size=0x016a0 (  5792) load
0x4008899c: cpu_ll_get_core_id at /home/willemmerson/esp/esp-idf/components/soc/src/esp32/include/hal/cpu_ll.h:35
 (inlined by) xPortGetCoreID at /home/willemmerson/esp/esp-idf/components/freertos/include/freertos/portable.h:205
 (inlined by) prvAddNewTaskToReadyList at /home/willemmerson/esp/esp-idf/components/freertos/tasks.c:1185

I (193) boot: Loaded app from partition at offset 0x10000
I (193) boot: Disabling RNG early entropy source...
I (194) cpu_start: Pro cpu up.
I (197) cpu_start: Application information:
I (202) cpu_start: Project name:     empty_esp32_project
I (208) cpu_start: App version:      1
I (212) cpu_start: Compile time:     Apr 20 2020 09:25:26
I (219) cpu_start: ELF file SHA256:  f8f5f51bbd9e2d21...
I (225) cpu_start: ESP-IDF:          v4.2-dev-1034-ge599b794b-dirty
I (231) cpu_start: Starting app cpu, entry point is 0x400811f8
0x400811f8: call_start_cpu1 at /home/willemmerson/esp/esp-idf/components/esp32/cpu_start.c:277

I (0) cpu_start: App cpu up.
I (242) heap_init: Initializing. RAM available for dynamic allocation:
I (249) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (255) heap_init: At 3FFB2AC0 len 0002D540 (181 KiB): DRAM
I (261) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (267) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (274) heap_init: At 4008A03C len 00015FC4 (87 KiB): IRAM
I (280) cpu_start: Pro cpu start user code
I (298) spi_flash: detected chip: generic
I (299) spi_flash: flash io: dio
W (299) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (309) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (10320) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10320) task_wdt:  - loop_task (CPU 0/1)
E (10320) task_wdt: Tasks currently running:
E (10320) task_wdt: CPU 0: IDLE0
E (10320) task_wdt: CPU 1: IDLE1

```

david-cermak, 2020-04-20:
@will-emmerson 
Ok, thanks for this test, now I think i'm slowly getting what you meant...
Sorry have missed from the beginning that you're actually watching your original `mqtt-publish` (thought when you were talking about WDT default settings, so i was under the impression that only IDLE tasks are being watched).

Is it correct? Are you watching the other tasks as well? In that case, yes, you would have to reset the task WDT at least every `CONFIG_ESP_TASK_WDT_TIMEOUT_S` period. So if the publish function is called with timeout of 10s, the watchdog timeout must be higher.

Now, to the example above: Running the
``` c
    ESP_ERROR_CHECK(esp_task_wdt_add(NULL));
``` 
would register this task to be watched, so the user must reset the WDT (no actual relation to yielding nor task priorities here).  By default (without calling the `esp_task_wdt_add()` function), only IDLE tasks are watched, so no need to reset anything. Just yielding to low-prio task would reset the watchdog indirectly, indicating the tasks with the lowest prio have chance on running.

will-emmerson, 2020-04-20:
Ok, it's starting to make sense now, sorry for the confusion. Yes I'm watching all tasks with the watchdog because I had a weird problem where one of them was getting 'stuck' and ultimately stopping the whole program. So what you're saying is that if I'm watching my own tasks then I need to be extra careful about what code I'm calling in idf because it might block and it's not going to reset the task watchdog itself?

david-cermak, 2020-04-20:
> So what you're saying is that if I'm watching my own tasks then I need to be extra careful about what code I'm calling in idf because it might block and it's not going to reset the task watchdog itself?

Exactly! see here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#task-watchdog-timer
> Each watched task must ‘reset’ the TWDT periodically to indicate that they have been allocated CPU time.

In contrast (with default settings) when watching the idle tasks only, you just have to be careful to yield so lower priorities.

will-emmerson, 2020-04-20:
Is it worth a note in the docs saying this a bit more obviously? Because with the default task watchdog (5 seconds) and default network timeout (10 seconds) there is going to be problem.

david-cermak, 2020-04-20:
> Is it worth a note in the docs

I don't think that all the details about network timeouts should be described in the ["task watchdog docs"](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#task-watchdog-timer) (since no task calling a network/blocking API is watched by default).
But agree, that it is not very clear from the MQTT docs, that some APIs might block with certain timeout: 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html?highlight=mqtt#_CPPv423esp_mqtt_client_publish24esp_mqtt_client_handle_tPKcPKciii

I suggest adding the notes on blocking APIs and timeouts to the MQTT docs. If you agree with this assessment, please keep this open and will close automatically on adding these docs.

closed 2020-05-05 by espressif-bot