espressif/esp-idf #5078: What does: "require flash encryption to be already enabled"

Nexus1212, 2020-04-07:
Hello,
in the security options of the esp-idf is the option available: "require flash encryption to be already enabled"

I found no dokumentation to that option!
What does that option?
I have found the Bootloader size becomes unter 8000bit if I activate this option...

Thank you!

igrr, 2020-04-07:
Quoting https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-secure-flash-require-already-enabled:

> If not set (default), and flash encryption is not yet enabled in eFuses, the 2nd stage bootloader will enable flash encryption: generate the flash encryption key and program eFuses. If this option is set, and flash encryption is not yet enabled, the bootloader will error out and reboot. If flash encryption is enabled in eFuses, this option does not change the bootloader behavior. Only use this option in testing environments, to avoid accidentally enabling flash encryption on the wrong device. The device needs to have flash encryption already enabled using espefuse.py.

Basically, it is an option which we have introduced to prevent accidentally enabling flash encryption on a development board, if flash encryption wasn't yet enabled. It should not be used in production.

Nexus1212, 2020-04-07:
Thank you very much. That is very helpfull!

closed 2020-04-07 by Nexus1212