espressif/esp-idf #5079: espcoredump.py: struct.error: bad char in struct format

szmodz, 2020-04-07:
UART coredumps basically don't work in the latest master. is this a known issue?

```
espcoredump.py info_corefile -t b64 -c xxx/core.txt xxx/build/xxx.elf
espcoredump.py v0.4-dev
Traceback (most recent call last):
  File "esp-idf/components/espcoredump/espcoredump.py", line 1810, in <module>
    main()
  File "esp-idf/components/espcoredump/espcoredump.py", line 1805, in main
    operation_func(args)
  File "esp-idf/components/espcoredump/espcoredump.py", line 1553, in info_corefile
    loader = ESPCoreDumpFileLoader(core_fname, args.core_format == 'b64')
  File "esp-idf/components/espcoredump/espcoredump.py", line 960, in __init__
    self.fcore = self._load_coredump(path, b64)
  File "esp-idf/components/espcoredump/espcoredump.py", line 983, in _load_coredump
    raise e
  File "esp-idf/components/espcoredump/espcoredump.py", line 976, in _load_coredump
    data = base64.standard_b64decode(line.rstrip(b'\r\n'))
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/base64.py", line 105, in standard_b64decode
    return b64decode(s)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/base64.py", line 87, in b64decode
    return binascii.a2b_base64(s)
binascii.Error: Incorrect padding
```

Assuming it's this problem:
https://gist.github.com/perrygeo/ee7c65bb1541ff6ac770

appending b'==='  to all input lines results in another problem, a bit further:

```
espcoredump.py v0.4-dev
Traceback (most recent call last):
  File "esp-idf/components/espcoredump/espcoredump.py", line 1810, in <module>
    main()
  File "esp-idf/components/espcoredump/espcoredump.py", line 1805, in main
    operation_func(args)
  File "esp-idf/components/espcoredump/espcoredump.py", line 1554, in info_corefile
    core_fname = loader.create_corefile(args.save_core, exe_name=args.prog, rom_elf=rom_elf)
  File "esp-idf/components/espcoredump/espcoredump.py", line 936, in create_corefile
    return self._extract_elf_corefile(core_fname, off + self.ESP_COREDUMP_HDR_SZ, exe_name)
  File "esp-idf/components/espcoredump/espcoredump.py", line 790, in _extract_elf_corefile
    note_read += note.read(ns.data[note_read:])
  File "esp-idf/components/espcoredump/espcoredump.py", line 207, in read
    self.name = struct.unpack("<%ds" % (nm_len - 1), data[hdr_sz:hdr_sz + nm_len - 1])[0].decode('ascii')
struct.error: bad char in struct format
```
This fails because nm_len_a is 0.

igrr, 2020-04-07:
Have you switched to a different Python version recently? If yes, could you please try `find $IDF_PATH -name '*.pyc' | xargs rm` to see if it solves the issue?

(Our CI tests for core dump to UART are passing at the moment, so if the above doesn't help, we will need more information to reproduce this)

szmodz, 2020-04-07:
`Failed to run espcoredump script: Command '['/opt/local/bin/python', '/Users/szmodz/xxx/esp-idf/tools/../components/espcoredump/espcoredump.py', 'info_corefile', '--core', '/var/folders/k6/3f_tslfx59n20rzyydxc_bjw0000gn/T/tmp38e5x579', '--core-format', 'b64', '/Users/szmodz/xxx/build/xxx.elf']' returned non-zero exit status 1.
`

I'll try to reproduce this basing on an esp-idf example.

szmodz, 2020-04-08:
Turns out it's the idf-monitor coredump decoding that's causing the problem.

When decoding manually with coredump decoding disabled, it works fine.

When coredump decoding is enabled, espcoredump.py fails, and the coredump (output between CORE DUMP START and END) is corrupted.

idf monitor coredump decoding works fine with small esp-idf examples, though. Perhaps it's related to the size of the dump? The project I'm working on is pretty big, so the coredump is also pretty big.

monitor output looks like this:
```
Core dump started (further output muted)
Received  18 kB...
Core dump finished!
Failed to run espcoredump script: Command '['/opt/local/bin/python', '/Users/szmodz/xxx/esp-idf/tools/../components/espcoredump/espcoredump.py', 'info_corefile', '--core', '/var/folders/k6/3f_tslfx59n20rzyydxc_bjw0000gn/T/tmpyiol234d', '--core-format', 'b64', '/Users/szmodz/xxx/build/xxx.elf']' returned non-zero exit status 1.

================= CORE DUMP START =================
<corrupted dump here, manual decoding fails>
================= CORE DUMP END =================

```

szmodz, 2020-04-08:
looks like some characters are missing from the output (!)

Broken dump:
```
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACPv6PwAAAAAAAAAAAAAAAAAAAAAAAAAA
ghAt74IQAAAAAAYAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
```

Working dump:
```
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACPv6PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA8F4IQCALBgCVvghAt74IQAAAAAAYAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
```