espressif/esp-idf #5080: ld.exe: BFD (crosstool-NG esp-2019r2) 2.31.1 internal error

vtunr, 2020-04-07:
## Environment

- Development Kit:      [Custom PCB]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
     `v4.0-2-g943a9b780 (Slightly modified to work with our code`
- Build System:         [CMake & idf.py]
- Compiler version:
    `xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0`
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt & regular CMD].
- Using an IDE?: [No]
- Power Supply:         [external 5V]

## Problem Description

Linking the binary is not working. It crashes with the following log : 
```
cmd.exe /C "cd . && C:\Users\usr\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\project.elf.rsp  -o project.elf  && cd ."
c:/users/usr/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: warning: size of symbol `status' changed from 22913405711024144 in c:/users/usr/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: BFD (crosstool-NG esp-2019r2) 2.31.1 internal error, aborting at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/binutils/bfd/bfd.c:884 in _bfd_doprnt

c:/users/usr/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: Please report this bug.
```

It's working on another project using the same toolchain
It's working on WSL2
It's working on Docker

I tried recloning my project, reinstalling the toolchain (using 2.3 install)

What's wrong ? Can I get more logs ?

gerekon, 2020-04-09:
Hi @vtunr
 
> It's working on another project using the same toolchain

I suppose it works with another project because linker does not produce such message in that case. Can you confirm?

vtunr, 2020-04-09:
@gerekon Yes, the other project has no linker issue on windows.
And the project where there is linker issue in windows, has no linker issue if building on WSL2/Docker (ubuntu for both)

antmak, 2020-04-09:
@vtunr Hi! Is it the C++ project where you the error get? I can only assume that the process don't have enough memory for linking (windows toolchain is a 32-bit one). I don't know about your project size and libs you use. Can you get clues about sizes? Or your thoughts on it.

vtunr, 2020-04-10:
Hi @antmak 
What size are you talking ? Binary size ? Number of object to link ? 

- Binary : 900 kB (vs 1.2MB with my second working project on windows)
- Number of object to link : Both project are quite similar, so i'm not sure this could be the cause ? Especially that as you can see, my not working project is smaller than my working project.

vtunr, 2020-04-10:
Just to give a bit more logs, i ran `xtensa-esp32-elf-g++.exe  -v -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\project.elf.rsp  -o project.elf` on both project.
You can find both verbose output attached :
[not_working.txt](https://github.com/espressif/esp-idf/files/4461259/not_working.txt)
[working.txt](https://github.com/espressif/esp-idf/files/4461260/working.txt)

vtunr, 2020-04-17:
Hi @gerekon, do you have any update or something I could do to help you understand the issue ?

Alvin1Zhang, 2020-04-20:
@vtunr Thanks for reporting, follow up and sharing the details. We will look into. Thanks.

vtunr, 2020-04-28:
Hi @Alvin1Zhang, any progress on this issue ?

MAVEREN, 2020-05-01:
@vtunr, I am having the same issue at the moment. 
I compared your working, and not working logs and noticed the line "warning: size of symbol `status' changed from 22913405711024144...". 
I have the same warning for "size of symbol " " changed". Unfortunately, I do not know if this is even relevant to the root issue.
Any progress on this?

MAVEREN, 2020-05-01:
I was able to successfully build and flash my project after having the same issue you are having. The warning I received was **"warning: size of symbol '_reserve_' changed from 229..."**. So I found all the instances in my code of the variable 'reserve' and noticed it was of type _int_ in some cases and type _uint8_t_ in others. After making all instances of this symbol the same type, the build was successful. I hope this is helpful.

vtunr, 2020-05-01:
@MAVEREN Thanks for your input ! if you managed to to fix the build by doing what you did, I will check if I can fix it the same way. I'll update the task once I'll test it.

vtunr, 2020-05-04:
@MAVEREN thanks for the help, it did worked ! 
@Alvin1Zhang I'm not sure I can close this issue as i'm not sure linux & windows toolchain should behave differently.

Alvin1Zhang, 2022-03-08:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/7cdd151612d3f63d857c2a69ec530466d24e185c, feel free to reopen.

closed 2022-03-08 by Alvin1Zhang