espressif/esp-idf #5081: generic onoff status parameters

AlpinProgrammer, 2020-04-07:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.0-beta2
- Build System:         Make
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Hello,
We are controlling an onoff server (ble_mesh_node example) with an onoff client (ble_mesh_client_model modified example).
We are currently having some issues to identify how the ACK (generic onoff status message) is sent by the server: advertising timings, number of times the message is sent through the 3 advertising channels...

Indeed, we would like to increase the performance of ACK reception on the client side as the radio performance is altered by using WiFi in our complete project (we already set the sdkconfig to privilege BLE over WiFi).

Thanks to some tests, we are sure that the ACK is sent by the server.
This is why the reliability of the ACK reception is of primary importance

### Expected Behavior
We are able to log the information regarding the ACK reception (timing, redundancy...)

### Actual Behavior
Despite the log (see section Debug Logs), we are unable to ensure how many ACK packets are received by the gateway and looking in esp-idf ble mesh core, we are unable to extract that information.

### Steps to reproduce

1. Flash ble_mesh_node example and ble_mesh_client_model example (modified to periodically send gen_onoff_set messages instead)
2. Provision the client then the server with nRF mesh
3. Observe the log output to determine how the ACK is transmitted

### Code to reproduce this issue

- ble_mesh_node example
- ble_mesh_client_model example modified:
replace app_main function by the following:

```
void app_main(void)
{
    int err;

    ESP_LOGI(TAG, "Initializing...");

    board_init();

    err = bluetooth_init();
    if (err) {
        ESP_LOGE(TAG, "esp32_bluetooth_init failed (err %d)", err);
        return;
    }

    /* Initialize the Bluetooth Mesh Subsystem */
    err = ble_mesh_init();
    if (err) {
        ESP_LOGE(TAG, "Bluetooth mesh init failed (err %d)", err);
    }

    uint8_t onoff = 0;
    uint8_t msg_tid = 0;
    while (1)
    {
        esp_ble_mesh_generic_client_set_state_t set = {0};
        esp_ble_mesh_client_common_param_t common = {0};
        esp_err_t err;

        common.opcode = ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET;
        common.ctx.net_idx = node_net_idx;
        common.ctx.app_idx = 0;
        common.ctx.addr = 4;
        common.ctx.send_ttl = 3;
        common.ctx.send_rel = false;
        common.msg_timeout = 0;
        common.msg_role = ROLE_NODE;

        common.model = onoff_client.model;
        set.onoff_set.op_en = false;
        set.onoff_set.onoff = onoff;
        set.onoff_set.tid = msg_tid++;

        ESP_LOGW(TAG, "In %s, sending data = %d (tid=%d)", __FUNCTION__, onoff, set.onoff_set.tid);

        err = esp_ble_mesh_generic_client_set_state(&common, &set);
        if (err) {
            ESP_LOGE(TAG, "%s: Generic OnOff Set failed with error %s", __func__, esp_err_to_name(err));
        }

        onoff = (onoff) ? 0:1;
        vTaskDelay(10000 / portTICK_PERIOD_MS);
    }

}
```

## Debug Logs

```
I (314798) NET: bt_mesh_net_recv: rssi -30 net_if 0
I (314798) TRANSPORT: trans_unseg: AFK 1 AID 0x33
I (314818) TRANSPORT: ASZMIC 0 AKF 1 AID 0x33
I (314818) TRANSPORT: len 7
I (314818) ACCESS: app_idx 0x0000 src 0x0003 dst 0x0002
I (314818) ACCESS: len 3: 820400
I (314828) ACCESS: OpCode 0x00008204
```

Thanks in advance for your help :smiley:

Alvin1Zhang, 2020-04-08:
@AlpinProgrammer Thanks for reporting the issue about ESP-BLE-MESH, we will look into.

Campou, 2020-04-08:
Hi @AlpinProgrammer 

> We are controlling an onoff server (ble_mesh_node example) with an onoff client (ble_mesh_client_model modified example).

The latest examples are [onoff_client](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_client) and [onoff_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server)

> We are currently having some issues to identify how the ACK (generic onoff status message) is sent by the server: advertising timings, number of times the message is sent through the 3 advertising channels...

The duration of each generic onoff status messages sent by the node is decided by the Network Transmit state, currently in the example, it's 90ms. And the BLE advertising interval is 20ms with a 0 - 10ms random delay.

> Indeed, we would like to increase the performance of ACK reception on the client side as the radio performance is altered by using WiFi in our complete project (we already set the sdkconfig to privilege BLE over WiFi).

On the client side, please make sure the full scan (component config -> bluetooth -> bluetooth controller -> ble full scan feature supported) is enabled, which will increase the rx performance of the client.
On the server side, you can enlarge the network transmit value for the node, which in fact will increase the number of ble advertising packets sent for each generic onoff status.

> Despite the log (see section Debug Logs), we are unable to ensure how many ACK packets are received by the gateway and looking in esp-idf ble mesh core, we are unable to extract that information.

Do you mean you want to know how many ack packets can be received by the client for each generic onoff set? Because we have specific duplicate filter policy in the controller, so each message (encrypted) will be posted to the host just once. Also in the mesh stack, network cache is used for the same purpose, which is to filter the same messages received by the device.

Thanks and Regards

AlpinProgrammer, 2020-04-09:
Hello @Campou ,
Thanks for your very fast answer !

> The latest examples are [onoff_client](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_client) and [onoff_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server)

I was unable to use the examples on v4.0-rc tag. That's why I downgraded to v4.0-beta2.

> The duration of each generic onoff status messages sent by the node is decided by the Network Transmit state, currently in the example, it's 90ms. And the BLE advertising interval is 20ms with a 0 - 10ms random delay.

This helps much! I see that those parameters are set in _esp_ble_mesh_cfg_srv_t_ struct. However, I don't see the 90ms are set.

> On the client side, please make sure the full scan (component config -> bluetooth -> bluetooth controller -> ble full scan feature supported) is enabled, which will increase the rx performance of the client.

Indeed, we set this parameter. However, we often get crashes, mostly after provisioning. I see [here](https://github.com/espressif/esp-idf/issues/2542) that the problem has been resolved on release/v4.0. Once there will be a stable version made, we can get this problem solved.

> On the server side, you can enlarge the network transmit value for the node, which in fact will increase the number of ble advertising packets sent for each generic onoff status.

By increasing TRANSMIT_COUNT in ```esp_ble_mesh_cfg_srv_t config_server{.net_transmit = ESP_BLE_MESH_TRANSMIT(TRANSMIT_COUNT, 20)}``` it seems.

> Do you mean you want to know how many ack packets can be received by the client for each generic onoff set? Because we have specific duplicate filter policy in the controller, so each message [...]

I was referring rather to the number of ACK adv packets sent by the server.
I was under the impression that the ACK message could be sent several times in case the radio is too busy or in case one of the packets gets lost.
It seems to be set in _esp_ble_mesh_cfg_srv_t_ struct.

Therefore, we won't be able to track identical packets for the ACK sent by the server unless we print low-level code in the stack ?

Thank you very much for taking the time to help !
Best regards

Campou, 2020-04-09:
Hi @AlpinProgrammer 

1. 90ms is decided by the `net_transmit` within the struct `esp_ble_mesh_cfg_srv_t`. In the example, the `net_transmit` is initialized with `ESP_BLE_MESH_TRANSMIT(2, 20)`, so the duration is (2 + 1) * (20 + 10) = 90ms.
2. Compared with the `v4.0-beta2` branch, the `release/v4.0` branch has a huge amount of updates (e.g. bug fixes, etc.). And using THE `release/v4.0` branch is recommended.
3. Currently on the client side, when the ack sent by the server is received, the packet will be posted to the application layer, this is the way recommended for the users to use. On the server side, only the duration of each ack sent can be printed. And the number of packets for the identical ack message needs to be calculated based on the ble adv interval, which currently is 20ms plus a 0 ~ 10ms random delay.

Thanks.

AlpinProgrammer, 2020-04-10:
Hi @Campou !
That's of great help, thank you very much.

According to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-to-a-release-branch the most stable option is to remain on the tags (v4.0 in our case). Do you advise to switch to release/v4.0 nonetheless ?

Besides, I tried again to run the ble_mesh_node onoff_server example on both v4.0 and release release/v4.0. With the latest nRF mesh app, it takes time to see the devices to provision, then the provisioning doesn't go till the end. It is then impossible to connect directly to that ESP considered as provisioned. I can open an issue for that if it's more suitable.

Thanks again

Campou, 2020-04-13:
Hi @AlpinProgrammer 

From a production side, please continue with the stable v4.0 version. When the stable v4.0.1 version is ready, you can switch to it.

Yes, for the issue with the nRF mesh app, please open another one for it.

Thanks.

AlpinProgrammer, 2020-04-14:
Thank you very much. I'll keep updated about the the new releases and stay on v4.0 meanwhile.
I can close this issue for now.

closed 2020-04-14 by AlpinProgrammer