espressif/esp-idf #5082: Log verbosity triggers Core0 Panic

hugoschaaf98, 2020-04-08:
## Environment

- Development Kit:  ESP32-DevKitC|
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]

- IDF version : ESP-IDF v4.2-dev-1034-ge599b794b
- Build System : idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System :  Linux (Ubuntu 18.04)
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

Hello everyone. I'm experiencing a strange issue. I was unable to run properly a bigger project so after hours I decided to build and flash the "hello world" example without any modification. 

### Expected Behavior

First build : 
Log output verbosity set to "Info", so it's the "default" config of the example.

Build -> Ok !
Flash -> Ok !
Run ->
Everything working as expected. The hello world example output in the terminal looks fine.

### Actual Behavior

Second build : 
Log output verbosity set to "Verbose". This is THE ONLY config modification i've done.

Build -> Ok !
Flash -> Ok !
Run ->
After many log output, I get a ```Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). ```

### Steps to reproduce

1. In the Hello world example, set the Log output verbosity to "Verbose" or even "Debug".
2. idf.py build flash monitor 

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7200
ho 0 tail 12 room 4
load:0x40078000,len:13696
load:0x40080400,len:3984
0x40080400: _init at ??:?

entry 0x40080688
I (31) boot: ESP-IDF v4.2-dev-1034-ge599b794b 2nd stage bootloader
I (31) boot: compile time 01:39:28
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
[... 215 lines cut ...]
Core  0 register dump:
PC      : 0x4008a0e9  PS      : 0x00060d34  A0      : 0x800d12b0  A1      : 0x3ffe3bc0  
0x4008a0e9: xt_ints_on at /media/linux_win7_exchange/esp/esp-idf/components/freertos/xtensa/xtensa_intr_asm.S:164

A2      : 0x0300000c  A3      : 0x0100000c  A4      : 0x00000019  A5      : 0x383fc000  
A6      : 0x00000016  A7      : 0x00000000  A8      : 0x15122500  A9      : 0x00060d20  
A10     : 0x3ffb24e4  A11     : 0x00008000  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00000018  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4008a0e6:0x3ffe3bc0 0x400d12ad:0x3ffe3bd0 0x400811d5:0x3ffe3bf0 0x400814b3:0x3ffe3c40 0x40079232:0x3ffe3c80 |<-CORRUPTED
0x4008a0e6: xt_ints_on at /media/linux_win7_exchange/esp/esp-idf/components/freertos/xtensa/xtensa_intr_asm.S:163

0x400d12ad: esp_cache_err_int_init at /media/linux_win7_exchange/esp/esp-idf/components/esp32/cache_err_int.c:70

0x400811d5: start_cpu0_default at /media/linux_win7_exchange/esp/esp-idf/components/esp32/cpu_start.c:397

0x400814b3: call_start_cpu0 at /media/linux_win7_exchange/esp/esp-idf/components/esp32/cpu_start.c:265

ELF file SHA256: 2218e24b4e8149ac
```

What could go wrong ? :(

sincerely, Hugo

igrr, 2020-04-08:
Could you please attach the log after power-on reset (reset by toggling RST line)?

hugoschaaf98, 2020-04-08:
Hello, here is the log output  after power-on reset.
[after_hard_reset.log](https://github.com/espressif/esp-idf/files/4450190/after_hard_reset.log)

igrr, 2020-04-08:
Ah okay, this seems to have been reported at https://github.com/espressif/esp-idf/issues/4499#issuecomment-607168967. The fix is merged but it hasn't yet appeared on Github, due to an issue found by an unrelated test. Will let you know when the fix is merged. For now you can use the suggestion from https://github.com/espressif/esp-idf/issues/4499#issuecomment-607732011 as a workaround.

hugoschaaf98, 2020-04-08:
Thank's this quick fix works great.

igrr, 2020-04-09:
Should be fixed in https://github.com/espressif/esp-idf/commit/a0e66fef0880dba2caf93e9cb9c65cff7dc9b880

Alvin1Zhang, 2020-04-09:
@hugoschaaf98 Thanks for reporting, would you please try the fix and help share feedback? Thanks.

hugoschaaf98, 2020-04-09:
Thanks for the job it works :)

Alvin1Zhang, 2020-04-10:
@hugoschaaf98 Glad to hear that your issue gets resolved. Feel free to reopen if the issue appears again. Thanks.

closed 2020-05-09 by Alvin1Zhang