espressif/esp-idf #5083: Wrong size in SSID and PASSWORD of smart config?

DiegoActionItem, 2020-04-08:
As of IDF 4.0 the SSID and PASSWORD are null terminated. If the supported sizes are 32, and 64 the buffers should be 33 and 65. This is what you do on the example: https://github.com/espressif/esp-idf/blob/v4.0/components/esp_wifi/src/smartconfig.c#L31

However, on the smartconfig headers, you use smaller buffers. Is seems like this is an "off by one bug", and the headers should be updated:
https://github.com/espressif/esp-idf/blob/v4.0/components/esp_wifi/include/esp_smartconfig.h#L46

Alvin1Zhang, 2020-04-10:
@DiegoActionItem Thanks for reporting, we will check. Thanks.

zhangyanjiaoesp, 2020-05-22:
hi @DiegoActionItem , I think it's not a problem, thanks all the same.

Alvin1Zhang, 2020-05-22:
@DiegoActionItem Thanks for reporting, feel free to reopen if you have any updates. Thanks.

closed 2020-05-22 by Alvin1Zhang