espressif/esp-idf #5084: idf.py menuconfig not working on windows (branch: release/v4.1)

volk-christian, 2020-04-08:
## Environment
- IDF version (run ``git describe --tags`` to find it): v4.1-beta1-106-g9a1c5a903
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No

## Problem Description
Calling **idf.py menuconfig** fails due to missing compiler ('CC') and so cannot be executed.

### Expected Behavior
Menuconfig shows up after executing **idf.py menuconfig**.

### Actual Behavior
idf.py throws the following error:
[1/4] Performing build step for 'mconf-idf'
FAILED: mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build kconfig_bin/mconf-idf
cmd.exe /C "cd /D C:\Development\esp-idf\examples\get-started\hello_world\build\kconfig_bin && rm -f C:/Development/esp-idf/tools/kconfig/zconf.lex.c C:/Development/esp-idf/tools/kconfig/zconf.hash.c && make -f C:/Development/esp-idf/tools/kconfig/Makefile mconf-idf && C:\Development\esp-idf-tools\tools\cmake\3.13.4\bin\cmake.exe -E touch C:/Development/esp-idf/examples/get-started/hello_world/build/mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build"
cc -c  -DCURSES_LOC="<curses.h>" -DLOCALE -MMD -MP -I "." -I "C:/Development/esp-idf/tools/kconfig"  C:/Development/esp-idf/tools/kconfig/mconf.c -o mconf.o
/usr/bin/sh: cc: command not found
C:/Development/esp-idf/tools/kconfig/Makefile:173: recipe for target 'mconf.o' failed
make: *** [mconf.o] Error 127
ninja: build stopped: subcommand failed.
Executing action: menuconfig
Running ninja in directory c:\development\esp-idf\examples\get-started\hello_world\build
Executing "ninja menuconfig"...
ninja failed with exit code 1

### Steps to reproduce
```bash
# Clone the ESP-IDF github repository
$ git clone --recursive --branch release/v4.1 https://github.com/espressif/esp-idf.git ${IDF_PATH}

# Install additional tools
$ exec ${IDF_PATH}/install.sh

# Activate ESP-IDF environment
$ source ${IDF_PATH}/export.sh

# Navigate into hello-world example and start menu config
$ cd ${IDF_PATH}/examples/get-started/hello_world
$ idf.py menuconfig
```

volk-christian, 2020-04-08:
mconf tool was removed with this commit: https://github.com/espressif/esp-idf/commit/01887f71e7fc8c514d90ab0e745b333d163e3f13
Before this commit, running **idf.py menuconfig** was possible without any issue.

Adding the tool manually to the esp-idf-tools and exporting it into the path is a workaround for the issue, but the problem of a missing tool (/usr/bin/sh: cc: command not found) still exists.

dobairoland, 2020-04-08:
@volk-christian What is `Plain Command Prompt`? Windows command line cmd.exe? The paths look like that. 

But the steps you are using to reproduce are for Unix. If you are in MSYS then you should not use the install and export scripts.

volk-christian, 2020-04-08:
@dobairoland We are running on windows and it's the very same behavior by using Git bash.exe or CMD.exe. Additionally, the behavior stays the same even if I do not install my environment using any of the install scripts but by the installer from you guidelines.

dobairoland, 2020-04-08:
In IDF v4.1, the main build system is `idf.py` (cmake) run from Windows command line. Here you use the the Windows tool installer (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer), and after that `install.BAT` and `export.BAT` files to setup the environment. Menuconfig is Python-based and it doesn't use `mconf-idf`. `mconf-idf` doesn't have to be and cannot be compiled in Windows command line. 

The deprecated build system is `make` which can be run from MSYS. For that there is an ZIP (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started-legacy/windows-setup.html#toolchain-setup) which you just unzip and everything is setup as it should be. Everything is pre-installed and you don't have to run `install.sh` and `export.sh`. You still can run `idf.py` from here but I would recommend to use `make` here. `mconf-idf` is used here with `make` and `idf.py` as well. It can be compiled here as everything was prepared in the ZIP-ped environment to do so.

Also if you are switching between git bash and cmd.exe then don't forget to remove the build directory because that could cause issues as well.

volk-christian, 2020-04-08:
Thanks for your further information, but as I can reproduce the issue using an example of IDF v4.1, I think we can discard the deprecated build system.
I removed some tools from my path (MinGW for example) and the behavior slightly changed:

### Prerequisites:
* Clone ESP-IDF branch v4.1
* Install tools using install.bat
_Remark: After installing the installer-setup no changes to the esp-tools have been noticed, so the benefit ist not clear --> skipped._

### Execution CMD.exe:
```batch
$ C:\Development\esp-idf> export.bat
$ C:\Development\esp-idf\examples\get-started\hello_world> idf.py menuconfig
```
--> works.

### Execution BASH.exe:
```bash
$ /c/Development/esp-idf > source export.sh
$ /c/Development/esp-idf/examples/get-started/hello_world > idf.py menuconfig
Warn about uninitialized values.
-- Found Git: C:/Development/Git/mingw64/bin/git.exe (found version "2.26.0.windows.1")
CMake Error at C:/Development/esp-idf/tools/cmake/kconfig.cmake:30 (message):
  Windows requires an MSYS2 version of gcc on the PATH to build mconf-idf.
  Consult the setup docs for ESP-IDF on Windows.
Call Stack (most recent call first):
  C:/Development/esp-idf/tools/cmake/build.cmake:161 (__kconfig_init)
  C:/Development/esp-idf/tools/cmake/idf.cmake:48 (__build_init)
  C:/Development/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
Executing action: menuconfig
Running cmake in directory c:\development\esp-idf\examples\get-started\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\development\esp-idf\examples\get-started\hello_world"...
cmake failed with exit code 1
```
--> fails.

I see the comment _Consult the setup docs for ESP-IDF on Windows._, but I cannot find anything about menuconfig within bash on Windows. Could you please point me to the concrete page?

dobairoland, 2020-04-08:
@volk-christian I'm happy that you got it working in Windows command line. Please keep using that because that is the best supported build system in IDF v4.x.

> but I cannot find anything about menuconfig within bash on Windows

Because it is not supported. And as I said earlier, you are not supposed to use `export.sh` anywhere on Windows because that script is for Unix platforms. If you setup a working MSYS environment then you should able to use it from GIT BASH as well but you will have to setup manually and don't look into the documentation regarding that because it is unsupported.

closed 2020-04-08 by dobairoland

volk-christian, 2020-04-08:
Okay, thanks for clarifying that git bash for windows is officially not supported by ESP-IDF.