espressif/esp-idf #5085: No rule to make target '/.../bootloader.bin'. Stop.

clbr, 2020-04-08:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version (run ``git describe --tags`` to find it): v4.0-280-ga3f3c7b
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-i686.tar.gz
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Using the stable 4.0 branch, trying to build the hello world project fails. Menuconfig works, but it fails in the bootloader.bin step.

```
$ make bootloader
Toolchain path: /usr/local/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp-2020r1
Expected to see version: esp-2019r2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 8.2.0
App "hello-world" version: v4.0-280-ga3f3c7bdc-dirty
WARNING: Toolchain version is not supported: esp-2020r1
Expected to see version: esp-2019r2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
make[1]: *** No rule to make target '/mnt/sda1/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin'.  Stop.
make: *** [/mnt/sda1/esp-idf/components/bootloader/Makefile.projbuild:41: /mnt/sda1/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin] Error 2
```

IDF_PATH=/mnt/sda1/esp-idf

### Expected Behavior

bootloader.bin make rule is found.

### Actual Behavior

Make fails to find the rule for bootloader.bin.

### Steps to reproduce

1. Prepare the stable 4.0 branch.
2. cd examples/get_started/hello_world
3. make menuconfig
4. make bootloader

### Code to reproduce this issue

## Debug Logs

## Other items if possible

- sdkconfig file: https://termbin.com/i9f4

clbr, 2020-04-08:
I can't find anywhere the pattern rule that would turn APP_ELF into APP_BIN. make/project.mk has a rule for APP_ELF, and lots of rules that depend on APP_BIN, but no rule that would creare APP_BIN.

Your makefiles are a huge mess... No wonder you started a rewrite in cmake. However cmake is even worse, I wish you'd just written cleaner makefiles.

clbr, 2020-04-08:
make --version
GNU make 4.2.1

igrr, 2020-04-08:
Hi @clbr, the rule is under esptool_py component: https://github.com/espressif/esp-idf/blob/e599b794bebebe66c6bb6d0af31a9451510ad3d4/components/esptool_py/Makefile.projbuild#L82

Unfortunately I wasn't able to reproduce the issue. Can you say, what is the filesystem under /mnt/sda1? Does the same issue happen if IDF is cloned into some other location (root filesystem or tmpfs)? Does the issue still happen if you delete the build directory and start from scratch? Did you start seeing this issue after some change (IDF update, something changed in the OS), or is this the first time you are trying to use IDF?

clbr, 2020-04-09:
sda1 is ext2. This is the first time I'm trying to get idf working. I've already tried to delete the build directory (under hello-world), it didn't help.

Will try the other things later.

clbr, 2020-04-09:
A fresh clone seems to help.

closed 2020-04-09 by clbr