espressif/esp-idf #5086: Impossible to compile BLE Mesh libraries with C++

faxik, 2020-04-08:
BLE Mesh component is written in C subset that is strictly not compatible with C++, which prevents using it in C++ projects.

## Environment
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-796-gbc56007aa
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
## Problem Description

//Detailed problem description goes here.
// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// Copyright 2017-2019 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include <stdio.h>
#include <string.h>

#include <esp_log.h>
#include <nvs_flash.h>

#include <freertos/FreeRTOS.h>
#include <freertos/timers.h>

#pragma GCC diagnostic ignored "-Wmissing-field-initializers"

#include <esp_bt.h>
#include <esp_bt_main.h>
#include <esp_bt_device.h>

#include <esp_ble_mesh_defs.h>
#include <esp_ble_mesh_common_api.h>
#include <esp_ble_mesh_networking_api.h>
#include <esp_ble_mesh_provisioning_api.h>
#include <esp_ble_mesh_config_model_api.h>
#include <esp_ble_mesh_generic_model_api.h>
#include <esp_ble_mesh_proxy_api.h>
#include <esp_ble_mesh_local_data_operation_api.h>

#include <esp_fast_prov_operation.h>
#include <esp_fast_prov_client_model.h>
#include <esp_fast_prov_server_model.h>
#include "provisioner_prov.h"
void app_main()
{

}

```

## Debug Logs
```
[...]/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h:49:22: note: to match this '('
     ((unsigned long) (IS_ARRAY(array) + \
                      ^
demo.cpp:115:21: note: in expansion of macro 'ARRAY_SIZE'
[... 193 lines cut ...]
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
demo.cpp: In lambda function:
[...]/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:254:65: error: expected '{' before '=' token
     .groups = { [0 ... (CONFIG_BLE_MESH_MODEL_GROUP_COUNT - 1)] =   \
                                                                 ^
[...]/esp-idf/components/bt/esp_ble_mesh/api/models/include/esp_ble_mesh_config_model_api.h:45:9: note: in expansion of macro 'ESP_BLE_MESH_SIG_MODEL'
         ESP_BLE_MESH_SIG_MODEL(ESP_BLE_MESH_MODEL_ID_CONFIG_CLI,          \
         ^~~~~~~~~~~~~~~~~~~~~~
demo.cpp:164:5: note: in expansion of macro 'ESP_BLE_MESH_MODEL_CFG_CLI'
     ESP_BLE_MESH_MODEL_CFG_CLI(&config_client),
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
demo.cpp: At global scope:
[...]/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:51:51: error: no match for 'operator=' (operand types are '<lambda()>' and 'int')
 #define ESP_BLE_MESH_ADDR_UNASSIGNED              0x0000
                                                   ^~~~~~
[...]/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:255:13: note: in expansion of macro 'ESP_BLE_MESH_ADDR_UNASSIGNED'
             ESP_BLE_MESH_ADDR_UNASSIGNED },                         \
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[...]/esp-idf/components/bt/esp_ble_mesh/api/models/include/esp_ble_mesh_config_model_api.h:45:9: note: in expansion of macro 'ESP_BLE_MESH_SIG_MODEL'
         ESP_BLE_MESH_SIG_MODEL(ESP_BLE_MESH_MODEL_ID_CONFIG_CLI,          \
         ^~~~~~~~~~~~~~~~~~~~~~
demo.cpp:164:5: note: in expansion of macro 'ESP_BLE_MESH_MODEL_CFG_CLI'
     ESP_BLE_MESH_MODEL_CFG_CLI(&config_client),

```

Alvin1Zhang, 2020-04-09:
@faxik Thanks for your interest in ESP-BLE-MESH, we will look into. Thanks.

Campou, 2020-04-09:
Hi @faxik 

I use the latest master branch and compile with the test code above, it works fine. The following is the screenshot of the detailed information.

[image]

Thanks.

faxik, 2020-04-09:
Hi @Campou 

You're using C compiler. C++ compiler cannot compile any of the following constructions:

/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h
```C
#define ESP_BLE_MESH_SIG_MODEL(_id, _op, _pub, _user_data)          \
{                                                                   \
    .model_id = (_id),                                              \
    .op = _op,                                                      \
    .keys = { [0 ... (CONFIG_BLE_MESH_MODEL_KEY_COUNT - 1)] =       \ <----------------
            ESP_BLE_MESH_KEY_UNUSED },                              \
    .pub = _pub,                                                    \
    .groups = { [0 ... (CONFIG_BLE_MESH_MODEL_GROUP_COUNT - 1)] =   \ <----------------
            ESP_BLE_MESH_ADDR_UNASSIGNED },                         \
    .user_data = _user_data,                                        \
}
```

/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h
```C
#define IS_ARRAY(array)                                     \
    ZERO_OR_COMPILE_ERROR(                                  \
        !__builtin_types_compatible_p(__typeof__(array),    \
                          __typeof__(&(array)[0])))
```

Such constructions (there are many) prevent any inclusion of the headers by a C++ compiler.
Additionally, the initialization order of structure members does not always follow the declaration order — this is allowed in C, but not C++.

Campou, 2020-04-10:
Hi @faxik 

Thanks for pointing the code out. Currently BLE Mesh does not fully support C++. I will mark this as a known issue and fix it later.

Thanks.

faxik, 2020-04-10:
Hi @Campou 

Thank you for looking into it. 

Please note, there are many places like I've pointed out. I just tried to be terse and descriptive to point out the nature of the incompatibilities.

Alvin1Zhang, 2020-04-12:
@faxik Many thanks for the feedback and suggestion, we would look into and improve.

zacwbond, 2020-05-15:
FWIW, I am able to build the BLE mesh code in C++.  All you have to do is ignore most of their initialization macros and instead define your own that are C++ compatible (that is, by ensuring the designated initializers are in the declared order and all fields are present).  

You might get more traction complaining at https://github.com/apache/mynewt-nimble.

faxik, 2020-05-15:
@zacwbond Thank you for sharing!

Did it take you a lot of time? 
Could you please suggest how to replace the offending construction such as:
```C
    .keys = { [0 ... (CONFIG_BLE_MESH_MODEL_KEY_COUNT - 1)] =
```
?

I will find time to file the corresponding bug to NimBLE, great suggestion!

zacwbond, 2020-05-16:
I simply dropped the array initialization syntax altogether and replaced it with explicit initialization, since for my build `CONFIG_BLE_MESH_MODEL_KEY_COUNT == 1` anyway.

For some reason, uniform initialization works for designated initializers of primitives only if they are default, hence the mixed style.  

```
bt_mesh_model vnd_models[]{
    {
        {
            .vnd{
                .company = COMPANY_ID,
                .id      = MODEL_ID,
            },
        },
        .elem_idx{},
        .mod_idx{},
        .flags{},
        .pub = &someElementPub,
        .keys   = {BT_MESH_KEY_UNUSED},
        .groups = {BT_MESH_ADDR_UNASSIGNED},
        .op = someElementOp,
        .user_data{},
    },
};

bt_mesh_elem elements[]{
    {
        .addr{},
        .loc{},
        .model_count     = std::size(root_models),
        .vnd_model_count = std::size(vnd_models),
        .models          = root_models,
        .vnd_models      = vnd_models,
    },
};
```

Alvin1Zhang, 2020-07-27:
Thanks for reporting, feel free to reopen if the issue still happens. Thanks.

closed 2020-07-27 by Alvin1Zhang

faxik, 2020-07-27:
@zacwbond 
Thank you!

@Alvin1Zhang 
The approach that was proposed above is a work-around, not a fix. 
Current code **as is** still does not support C++ compilation of any BLE Mesh examples.

When IDF-4 will be a part of Arduino, for example, there will be a lot of bug reports like that.