espressif/esp-idf #5087: Reading ADC2 using ULP

jgcjoec, 2020-04-08:
## Environment

- Development Kit:      Hiletgo ESP32
- Kit version 
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 4.0
- Build System:   idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:  USB

## Problem Description
I am able to read from ADC1 using the ULP no problem. However, when I adapt my code to read ADC2, it hangs on the ADC instruction - the "JUMP" is never reached. In the example below, level3_adc_channel is 4, which is GPIO 13. I confirmed that the main CPU is able to read from this same pad on GPIO 13, so my connections should be good.

### Expected Behavior
The ADC instruction should not hang when using ADC2
### Actual Behavior
The ADC instruction appears to hang when using ADC2
### Steps to reproduce
This **doesn't** wake the main CPU:
.global readLevel3
readLevel3:
	ADC r1, 1, level3_adc_channel + 1 // works out to ADC r1, 1, 4+ 1
	JUMP wakeUp // this is never reached. "wakeUp" tests to see if the main CPU can be woken up, waits until it can, then wakes it up.

This **does** wake the main CPU:
.global readLevel3
readLevel3:
	JUMP wakeUp
	ADC r1, 1, level3_adc_channel + 1 // works out to ADC r1, 1, 4+ 1

The above snippets tell me the ULP is hanging on the ADC instruction for some reason when using ADC2.

This similar code snippet, but for ADC1, **does** wake the main CPU:
.global readLevel2
readLevel2:
	ADC r1, 0, level2_adc_channel + 1
	JUMP wakeUp

### Code to reproduce this issue

See code in Steps to reproduce section

## Debug Logs
N/A
## Other items if possible
N/A

Alvin1Zhang, 2020-04-09:
@jgcjoec Thanks for reporting, we will look into.

boarchuz, 2020-04-09:
Include any relevant initialisation code for ADC too.

jgcjoec, 2020-04-09:
In case it matters, my program at a high-level works as follows:
1. Main CPU boots up
2. Load the ULP program
3. Take measurements using main CPU (I'm taking involved include ADC 1, ADC 2, I2C)
4. Anything measurements out of spec?
5. If no, start the ULP program and go to deep sleep
6. If yes, start Wi-Fi, send data, start ULP program, then go to deep sleep
7. While the main CPU is in deep sleep, the ULP monitors same peripherals in a low power state then wakes the main CPU if something falls out of line
8. There is also an EXT1 wakeup configured for GPIO25 using ESP_EXT1_WAKEUP_ANY_HIGH

To initialize the ADCs:
adc1_config_width(ADC_WIDTH_BIT_12);

adc1_config_channel_atten(7, ADC_ATTEN_DB_11);
adc1_config_channel_atten(4, ADC_ATTEN_DB_11);

adc2_config_channel_atten(0, ADC_ATTEN_DB_11);
adc2_config_channel_atten(6, ADC_ATTEN_DB_11);
adc2_config_channel_atten(3, ADC_ATTEN_DB_11);

adc1_characteristics = calloc(1, sizeof(esp_adc_cal_characteristics_t));
esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_ATTEN_DB_0, ADC_MAX_VOLTAGE, adc1_characteristics); // this is used later after calling adc1_get_raw()

adc2_characteristics = calloc(1, sizeof(esp_adc_cal_characteristics_t));
esp_adc_cal_characterize(ADC_UNIT_2, ADC_ATTEN_DB_11, ADC_ATTEN_DB_0, ADC_MAX_VOLTAGE, adc2_characteristics); // this is used later after calling adc2_get_raw()

adc1_ulp_enable(); // I noticed there is no adc2_ulp_enable()

jgcjoec, 2020-04-10:
I did some more testing - nothing has resulting in ADC2 being functional in the ULP. Now I'm at the point where even the presence of instructions to read ADC2 in the ULP cause it to hang - even if they are preceded by a WAKE instruction.

Things I've tried:
1. Commenting out the code to start Wi-Fi
2. Trying different channels on ADC2
3. Trying different target registers

If there's anything else anyone would like me to test to help solve this problem, please let me know.

boarchuz, 2020-04-10:
Can you try my ADC2 ULP init here?:
https://github.com/boarchuz/HULP/blob/7474ea78c6fb69b3f96a769897f7398c782f9899/hulp.cpp#L147

jgcjoec, 2020-04-11:
Are you referring to this section below, or the entire block?

        REG_CLR_BIT(SENS_SAR_MEAS_START2_REG, SENS_MEAS2_START_FORCE);
        REG_CLR_BIT(SENS_SAR_MEAS_START2_REG, SENS_SAR2_EN_PAD_FORCE);
        REG_CLR_BIT(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DIG_FORCE);
        REG_CLR_BIT(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_PWDET_FORCE);
        REG_SET_BIT(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_SAR2_MUX);

boarchuz, 2020-04-12:
Your initialisation code seems to handle the earlier stuff, the registers you quoted are the important part, specificially this one:
`REG_CLR_BIT(SENS_SAR_MEAS_START2_REG, SENS_MEAS2_START_FORCE);`
I think that alone will solve your problems.

I just tested it and ADC2 wakeup is working fine for me using that library.

jgcjoec, 2020-04-13:
Thanks, will give that a try!

Can you elaborate on what that does, or point me to the docs that discuss it?

boarchuz, 2020-04-13:
https://github.com/espressif/esp-idf/blob/c1d0daf36d0dca81c23c226001560edfa51c30ea/components/soc/soc/esp32/include/soc/sens_reg.h#L836-L838

jgcjoec, 2020-04-13:
That seemed to fix it - thank you! I'll do some more testing and report back if I have any additional issues.

closed 2020-04-15 by jgcjoec