espressif/esp-idf #5088: Security Boot unnecessary, if flash encryption active?

Nexus1212, 2020-04-08:
Hello,
in the menual of flash encryption is prefered to active always both, Security Boot and flash encryption.

But is Security Boot unnecessary, if flash encryption is active? 

[Manual](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html):
> Secure Boot is a feature for ensuring only your code can run on the chip. Data loaded from flash is verified on each reset.

If flash encryption is active, the bootloader, application and partiontable are encrypted. It is not possible to add a code (foreign code), beacause only the encrypted code will be running on the esp32.

What is the advantage, to activate to flash encryption in addation option security boot?

projectgus, 2020-04-09:
HI @Nexus1212,

You're right that if flash encryption is enabled (in Release mode for IDF v4.0 or newer, or enabled at all for IDF v3.x) then it's not possible for an attacker to easily program their code onto the device.

However, there are some situations where Secure Boot adds additional protections:

* If an attacker can exploit the firmware running on the device via a software bug in the application, then Secure Boot prevents an exploit from being stored in flash. Any exploit would have to be repeated by the attacker every time the device was powered on.
* If an attacker can exploit the OTA update mechanism (for example, compromise the server or because the app didn't use a secure transport protocol), then signature verification prevents an attacker's firmware from being written to flash by the OTA update system and then booted. (Note: it is possible to [enable only signature verification](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-secure-signed-on-boot-no-secure-boot) without enabling hardware secure boot, to have this protection without the first stage hardware check.)
* On other architectures (not related to ESP32), researchers have demonstrated that by corrupting a single block of encrypted code (16 bytes) repeatedly (essentially producing random values each time), they can eventually find a sequence which executes as a "noop" and therefore can bypass some important check in the code (for example, bypassing TLS certificate checking for an OTA update). Our preliminary investigations suggests that this technique is not easy to apply with the ESP32's instruction set, and we have no other evidence that the attack is possible against ESP32 at all. However, enabling Secure Boot makes it much more difficult for an attacker to carry out attacks of this kind at all.
* More generally, Secure Boot provides an additional layer of protection in the case that some other vulnerability is found in a system using Flash Encryption alone. We hope such a situation will not occur, but it's not impossible.

You may reasonably decide that flash encryption provides enough protection for your particular application. This is, of course, up to you. However we recommend enabling both together.

(Note: it's not recommended to *ever* use only Secure Boot without Flash Encryption.)

Angus

projectgus, 2020-04-30:
Closing the issue as it seems the question is answered.

(For future reference, we prefer questions like this to be asked via the https://esp32.com forum.)

closed 2020-04-30 by projectgus