espressif/esp-idf #5089: BLE Mesh callback functions should provide option for user data (void*)

faxik, 2020-04-08:
Currently, all BLE Mesh callback functions do not support custom user parameter (void*).

Such design decision prevents any incapsulation and managing of shared state is thus only possible using global variables, which is considered a poor practice. Additionally, this makes C++ integration nearly impossible, since there is no proper way to pass `this` pointer, implicitly or explicitly.

## Current state
```C

typedef void (* esp_ble_mesh_prov_cb_t)(esp_ble_mesh_prov_cb_event_t event,
                                        esp_ble_mesh_prov_cb_param_t *param);

typedef void (* esp_ble_mesh_model_cb_t)(esp_ble_mesh_model_cb_event_t event,
        esp_ble_mesh_model_cb_param_t *param);

typedef void (* esp_ble_mesh_cfg_client_cb_t)(esp_ble_mesh_cfg_client_cb_event_t event,
        esp_ble_mesh_cfg_client_cb_param_t *param);

typedef void (* esp_ble_mesh_cfg_server_cb_t)(esp_ble_mesh_cfg_server_cb_event_t event,
        esp_ble_mesh_cfg_server_cb_param_t *param);

typedef void (* esp_ble_mesh_generic_server_cb_t)(esp_ble_mesh_generic_server_cb_event_t event, esp_ble_mesh_generic_server_cb_param_t *param);
```

## Proposed signatures

```C

typedef void (* esp_ble_mesh_prov_cb_t)(esp_ble_mesh_prov_cb_event_t event,
                                        esp_ble_mesh_prov_cb_param_t *param, void* user_param);

typedef void (* esp_ble_mesh_model_cb_t)(esp_ble_mesh_model_cb_event_t event,
        esp_ble_mesh_model_cb_param_t *param, void* user_param);

typedef void (* esp_ble_mesh_cfg_client_cb_t)(esp_ble_mesh_cfg_client_cb_event_t event,
        esp_ble_mesh_cfg_client_cb_param_t *param, void* user_param);

typedef void (* esp_ble_mesh_cfg_server_cb_t)(esp_ble_mesh_cfg_server_cb_event_t event,
        esp_ble_mesh_cfg_server_cb_param_t *param, void* user_param);

typedef void (* esp_ble_mesh_generic_server_cb_t)(esp_ble_mesh_generic_server_cb_event_t event, esp_ble_mesh_generic_server_cb_param_t *param, void* user_param);
```

Alvin1Zhang, 2020-04-09:
@faxik Thanks for reporting, we will look into and evaluate. Thanks.

ankit-thealchemist, 2020-05-15:
> Currently, all BLE Mesh callback functions do not support custom user parameter (void*).
>
> Such design decision prevents any incapsulation and managing of shared state is thus only possible using [...]

Yes you pointed out the right thing. Yesterday I was working on the server example. I just declared the variable in the main function then I just halt there. I do not understand what to d. In the end I just declare the global variable and start from there.

By the way did you tried the oob functionality in provisioning  in the ble mesh. Because I when I tried to do that, I just found it just do not work. All the node, whatever is the oob value always provisioned with no oob functionality. if you know something, please help me on that.

Campou, 2020-05-19:
Hi @faxik 

Thanks for the advice.

For sharing states between different models, this is something which needs a better approach.
And If the `user_param` is needed for a specific model for sharing states with other models, may be we can add the `void *user_param` in the structure of the model, without modifying the public functions.

Thanks.

Alvin1Zhang, 2020-08-18:
Thanks for reporting, will close now, feel free to reopen with more details/updates, thanks.

closed 2020-08-18 by Alvin1Zhang