espressif/esp-idf #5091: Make --system-site-packages available to user while creating virtuanenv

keelung-yang, 2020-04-10:
For some reasons, someone want to use packages in site-packages, instead of installing a fixed verson package sometimes.

Such as:
- gevent v1.4 not support python 3.8 so no wheel available currently. _esp-idf/intall.bat_ will failed if no vc14(vs2015) build tool installed on windows. Even if I've installed gevent v1.5 wheel from https://www.lfd.uci.edu/~gohlke/pythonlibs/#gevent
- Some python packages aren't pip installable, by @danyeaw in https://github.com/pypa/pipenv/issues/419
- Test some packages which not released but used by esp-idf, such as _pyserial_

Alvin1Zhang, 2020-04-10:
@keelung-yang Thanks for sending the feature request, we have noted this and would evaluate. Will provide feedback once evaluation is finished. Thanks.

igrr, 2020-04-10:
Hi @keelung-yang, which package is requiring the compiler to get built? The packages in requirements.txt normally distribute binary wheels, so they should be installable on Windows without requiring a compiler. Sometimes if a new version of the package (or Python) comes out, it may take some number of hours or days for the wheel to be uploaded to pypi.

keelung-yang, 2020-04-10:
@igrr It's gevent, required by gdbgui. The latest version is [v1.5a5](http://www.gevent.org/changelog.html) which supports Python 3.8 but unreleased yet.

igrr, 2020-04-10:
I see, maybe we need to pin gdbgui to an older version which doesn't have this issue.

igrr, 2020-04-10:
Looks like gevent 1.5.0 has been tagged today: https://github.com/gevent/gevent/releases/tag/1.5.0. The release is [not yet on PyPI](https://pypi.org/project/gevent/#files), but hopefully the wheels will get uploaded soon.

igrr, 2020-04-11:
gevent 1.5.0 wheels are is now on Pypi, including Python 3.8 ones. I think this can be closed.

closed 2020-04-11 by igrr

keelung-yang, 2020-04-11:
@igrr gevent is just one of reasons of needinesses of --system-site-packages. This feature requestion is not just for gevent's compiling.

igrr, 2020-04-11:
Regarding

> Test some packages which not released but used by esp-idf, such as pyserial

I think you can still run `pip install --upgrade pyserial==x.y.z` inside the virtual environment if you need to test a different version of pyserial.

Regarding

> Some python packages aren't pip installable

can you please give a practical example when you required such package?

The reason I'm not in favor of the --system-site-packages option is that it introduces more ways in which Python packages installation can go wrong, e.g. due to incompatible versions of packages installed globally. And it means more installation-specific issues for us to troubleshoot.

Finally, you can always create the virtual environment manually at the location where IDF expects to see it (i.e. under %IDF_TOOLS_PATH%/python_env/). `install.bat` will not re-create the existing virtual environment, and will simply ensure that IDF required packages are installed. Creating the virtual environment manually requires only a single command. So seems like it is already simple enough?

keelung-yang, 2020-04-11:
@igrr Great! Thanks!!